lbbrhzn/ocpp #1078: ZJ BENY - changing current gives invalid JSON error

ToastySefac, 2024-02-06:
## Version of the custom_component
OCPP: v0.5.1
ZJ BENY: v1.0.31  BCP-A2N-L

## Describe the bug

When I change the charging current it starts giving errors and the device goes offline for about 25 seconds before coming back online, then repeating.

I can see in the logs that it is reporting that the message is not valid JSON and ChatGPT suggests the issue is the leading comma before the first object in the 'sampledValue' array. Would that be causing the unit to become unresponsive. If so, does this mean there is an issue with the firmware, or is there a possible work around to get the unit stable?

## Debug log

```text

2024-02-06 19:06:21.580 INFO (MainThread) [ocpp] charger: send [2,"f2b57d3b-1d2d-47ec-ab9d-d3778b0fc8e7","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2024-02-06 19:06:21.977 INFO (MainThread) [ocpp] charger: receive message [3,"f2b57d3b-1d2d-47ec-ab9d-d3778b0fc8e7",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"}]}]
2024-02-06 19:06:21.978 INFO (MainThread) [ocpp] charger: send [2,"0e75fc9a-1906-4d7a-a1cf-00a1d47bcecf","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2024-02-06 19:06:22.084 INFO (MainThread) [ocpp] charger: receive message [3,"0e75fc9a-1906-4d7a-a1cf-00a1d47bcecf",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"10"}]}]
2024-02-06 19:06:22.090 INFO (MainThread) [ocpp] charger: send [2,"768714ab-5d43-4da1-96d2-4e9ada87eaa6","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":10,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":19.0}]}}}]
2024-02-06 19:06:22.235 INFO (MainThread) [ocpp] charger: receive message [3,"768714ab-5d43-4da1-96d2-4e9ada87eaa6",{"status":"Accepted"}]
2024-02-06 19:06:33.831 INFO (MainThread) [ocpp] charger: send [2,"9ccb5185-bc1a-4329-8c26-41caa55f457e","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2024-02-06 19:07:03.834 INFO (MainThread) [ocpp] charger: send [2,"9315d242-e8ee-49a9-87bd-9d957bcd6e86","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2024-02-06 19:07:03.835 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140343519912896] Waited 30s for response on [2,"9ccb5185-bc1a-4329-8c26-41caa55f457e","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}].
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 500, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 327, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 893, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 361, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 502, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 238, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2149, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2186, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 272, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 882, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 952, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 111, in async_set_value
    await entity.async_set_native_value(native_value)
[... 334 lines cut ...]
    return await self.charge_points[cp_id].set_charge_rate(limit_amps=value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 605, in set_charge_rate
    resp = await self.get_configuration(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 828, in get_configuration
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/ocpp/charge_point.py", line 331, in call
    raise asyncio.TimeoutError(
TimeoutError: Waited 30s for response on [2,"706efad7-87ba-4350-a864-6572dd94159b","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}].
2024-02-06 19:10:05.971 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2024-02-06 19:10:05.971 INFO (MainThread) [custom_components.ocpp] Charger websocket path=///charger
2024-02-06 19:10:05.971 INFO (MainThread) [custom_components.ocpp] Charger charger reconnected to 192.168.1.106:9000.
2024-02-06 19:10:05.971 DEBUG (MainThread) [custom_components.ocpp] Reconnect websocket to charger
2024-02-06 19:10:05.971 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to 'charger'
2024-02-06 19:10:08.826 INFO (MainThread) [ocpp] charger: receive message [2,"8ihps4pd1460etv1yj2ra5nb","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-02-06T08:10:06Z"}]
2024-02-06 19:10:08.827 INFO (MainThread) [ocpp] charger: send [3,"8ihps4pd1460etv1yj2ra5nb",{}]
2024-02-06 19:10:08.828 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to 'charger'
2024-02-06 19:10:09.197 INFO (MainThread) [ocpp] charger: receive message [2,"zfx0uomszb7rw369wbkz7zoc","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2024-02-06T08:10:06Z"}]
2024-02-06 19:10:09.198 INFO (MainThread) [ocpp] charger: send [3,"zfx0uomszb7rw369wbkz7zoc",{}]
2024-02-06 19:10:19.072 DEBUG (MainThread) [custom_components.ocpp] Connection closed to 'charger': sent 1000 (OK); no close frame received
2024-02-06 19:10:19.073 INFO (MainThread) [custom_components.ocpp] Charger charger disconnected from 192.168.1.106:9000.

```

Conradical314, 2024-02-10:
I finally got my ZJ Beny connected to OCPP today. I get this issue as well.  BCP-AT2N-L firmware 1.0.31. 
Though I am stoked to be able to use the on/off via HA instead of the tedious Bluetooth control!

PeterBorfai, 2024-02-18:
I appear to have the same issue. During charging, the charger becomes unavailable for 25 seconds or so. Then recovers after 40 seconds. This cycle repeats throughout the charging session.

BCP-AT2N-P by Benyi Firmware: 1.0.31

ToastySefac, 2024-02-18:
On the Charge HQ site, it mentions a similar issue: https://chargehq.net/kb/zj-beny-ev-charger-configuration#:~:text=There%20is%20a%20known%20issue,caused%20by%20the%20charger%20itself.

> There is a known issue where ZJ Beny chargers will frequently disconnect and reconnect to Charge HQ. This can happen tens or even hundreds of times per day. You may see a "charger offline" error message in Charge HQ which appears and disappears frequently.

> This issue is caused by the charger itself. It doesn't generally lead to any problems, as the charger should reconnect quickly each time.

> We are hoping that this issue will be resolved by a firmware update from ZJ Beny.

Although, when I've used Charge HQ, it seems to work fine. Maybe they have worked around any bug... Have either of you had any success in Charge HQ?

Conradical314, 2024-02-18:
Thanks for covering the Charge HQ side, I'd been wondering about that. I assumed similar to above - that Charge HQ users must get the same issue, but that they must have worked around the bad data in their OCPP implementation.

I got in touch with the company that sold me the ZJ Beny, Ultipower in Australia. I wasn't expecting much but they've got in contact with ZJ Beny and at least by appearances we're getting engagement.

A few days ago I tried to dig into the error and the code to see how one might workaround the charger bug when using this HA project. Other issue threads has people mentioning that it's possible, I think I saw for the Grizzl-E. But I couldn't figure out how to edit (permanently or in a nice way) the messages.py in the underlying OCPP Python project that this HA project uses. I'd appreciate any tips, though I suspect any workaround like that could be truly nasty. Best to get ZJ Beny to fix. How long that will take, who knows.

I got this response from ZJ Beny:

```
I have sent your information to our engineer this morning.

But he want to get below informations of Home Assistant, so that we can solve this issue.

1) URL address:
2) WEB or APP Account:
3) Pile Number: 

So could you please provide your account of Home Assistant for our reference? 
```

I think they're asking to access my server >.<. Which is not something I'll be doing as I haven't setup any remote access yet. I honestly don't know how to respond and assist them just yet. If anyone can safely provide them their server I can connect you.

PeterBorfai, 2024-02-18:
I just connected my Benyi to ChargeHQ. Connection State = Online. Status = Charging.  Looks stable over several minutes.
Chargehq says they've tested Benyi firmware 1.0.14 and 1.0.23. But my 1.0.31 works ok.

CP-AT2N-P by Benyi Firmware: 1.0.31

ToastySefac, 2024-02-19:
> I think they're asking to access my server >.<. Which is not something I'll be doing as I haven't setup any remote access yet. I honestly don't know how to respond and assist them just yet. If anyone can safely provide them their server I can connect you.

I bought from Ultipower as well. Their contact at ZJ Beny has sent me a similar request.

I've setup a sandboxed home assistant install with the charger connected and sent them the details to access it. So, hopefully that helps them work out the problem.

ToastySefac, 2024-02-27:
Here's the response I got back from ZJ Beny after I followed them up. Pretty disappointing.

> Sorry for the late reply, due to the customer visited us last week.
> 
> Our engineer have login and test the Home Assistant.
> This program is a free charge software, and will be often offline after connected.
> Their Open Source Server has the problem, will not respone the message from EV Charger.
> 
> It is better to use our EV-BOX (EVB) App, free charge to use, more stable than Home Assistant.
> 
> Please note and feel free to let me know if any question, thank you.

Conradical314, 2024-02-27:
wow that's quite disappointing. They should do better, and present a much better argument than "HA is free open source therefore it's bad". 

How to reply and continue.... hmm. So it's not that I think that the JSON error is an absolute smoking gun. We can acknowledge the ambiguity coming from an interface between the ZJ Beny and a server.

I'll email them in my own thread as well, but I think the **pressure to apply is in OCPP _capability_ and _compliance_.** While we cannot assess compliance with our level of knowledge, we an argue that their charger is not demonstrating OCPP capability. Their suggestion to use their App does not address the fact their charger is sold as OCPP capable.

**Home Assistant is showing offline because ZJ Beny is dumping what seems to be invalid input messages.** The engineer on the ZJ Beny side is possibly having an even worse time of it on a remote connection.

ToastySefac, 2024-02-27:
I've sent a similar response.

ToastySefac, 2024-02-28:
I got another message back from ZJ BENY and they have released a beta firmware update 1.0.33.

I updated my unit and it's now working properly with Home Assistant! Might need some time to check that there are no errors, but I tried a few different settings and current levels and it responded as expected.

Here's their email with the right firmware link for BCP-A2N-L OCPP.

> Thank you for the information, helpful for us and we have found the problem.
> 
> Please upgrade the firmware as per the instruction as attached.
> 
> Your charger model number is BCP-A2N-L OCPP, so please select first URL address for updating.
> Household EV charger, Signel phase with cable
> http://106.15.78.131/V_DEBUG/BCP-A2N-P_SW1_0_33_HW1_DEBUG.bin
> 
> It is Beta Version, we will launch a official version around May.

PeterBorfai, 2024-02-28:
Thank you for the update. The new version 1.0.33 BETA appears to work correctly on my BCP-AT2N-P! I will monitor it and report back any issues. Thanks again!

Conradical314, 2024-02-28:
Yeah outstanding, I was surprised by the same reply this morning. I only had time for a quick look in the docs for how to update the firmware over OCPP. I know I've seen it talked about in other threads but couldn't find any documentation.
Sorry if I've missed something obvious, would one of you mind pointing me at the right process to update via OCPP? Or just the best way to update? Hopefully without factory resetting to use the Bluetooth connection

ToastySefac, 2024-02-28:
I changed the server over to use my Charge hq account. In there you can tap on your charger and update the firmware. Just need to paste in the url from the post above 

Just make sure you can reach the firmware server first with your computer. For some reason my router/isp was blocking it, but Optus didn't.

PeterBorfai, 2024-02-28:
To update the firmware:
Check the list of available files at  http://106.15.78.131   The new BETA firmware is in the V_DEBUG folder.
In Home Assistant, go to  Developer Tools / Services
Scroll through the list of Services to  "Open Charge Point Protocol (OCPP): Update charger firmware"
Go to YAML mode and modify the example provided to match your URL for your model. My YAML looked like this:
------
service: ocpp.update_firmware
data:
  firmware_url: http://106.15.78.131/V_DEBUG/BCP-AT2N-P_SW1_0_33_HW1_DEBUG.bin
  delay_hours: 0
-------
Then press "Call Service"
I got timestamped notifications to show: 
-------
Ocpp integration
Firmware upload status: Downloading
-------
Ocpp integration
Firmware upload status: Downloaded
-------
Ocpp integration
Firmware upload status: Installing
-------
Ocpp integration
Firmware upload status: Installed

It worked first time!

GuzziDoug, 2024-04-02:
I have been running the OCPP version of the ZJBeny charger for about 18 months. The only issue I have is that after a firmware update, one needs to perform a ´Clear Profile' otherwise one cannot change the current. On the ver 31 update, I also needed to power cycle the charger. Personally, I think it is a well priced Charger (also sourced from Ultipower). Currently running Ver 32 firmware.

Conradical314, 2024-04-21:
Thanks PeterBofai for the details of firmware update. 
And @GuzziDoug for mentioning the current change issue. If you might be able to continue to help, I seem to have that issue, however I can't Clear Profile. 
I've tried power cycling the charger (turn break off and on)

```
Y_CHARGER: send [2,"c54169c6-42fa-46fd-891f-177dbe3c0d3a","ClearChargingProfile",{}]
Y_CHARGER: receive message [3,"c54169c6-42fa-46fd-891f-177dbe3c0d3a",{"status":"Unknown"}]
WARNING (MainThread) [custom_components.ocpp] Failed with response: Unknown
```

Calling the Clear Profile Service from Home Assistant. I looked at the YAML but I can't figure out what data {} I might want to put in from a crude dig of the code https://github.com/mobilityhouse/ocpp/blob/d96d243c8fb3e042f6837aa4ccecda5356598064/ocpp/v201/call.py#L47

GuzziDoug, 2024-04-21:
fyi, I had trouble with the Clear Profile. But after I power cycled the charger, I found initially it ran full tilt (ie max rate), but if I moved the current setting on the Overview page, the current changed to the setting. So, the secret may be to try changing the current on the overview page,

regards, Doug

closed 2024-06-21 by ToastySefac

Idan37S, 2025-05-01:
> Thanks PeterBofai for the details of firmware update. And [@GuzziDoug](https://github.com/GuzziDoug) for mentioning the current change issue. If you might be able to continue to help, I seem to have that issue, however I can't Clear Profile. I've tried power cycling the charger (turn break off and on)
> 
> ```
> Y_CHARGER: send [2,"[key]","ClearChargingProfile",{}]
> Y_CHARGER: receive message [3,"[key]",{"status":"Unknown"}]
> WARNING (MainThread) [custom_components.ocpp] Failed with response: Unknown
> ```
> 
> Calling the Clear Profile Service from Home Assistant. I looked at the YAML but I can't figure out what data {} I might want to put in from a crude dig of the code https://github.com/mobilityhouse/ocpp/blob/d96d243c8fb3e042f6837aa4ccecda5356598064/ocpp/v201/call.py#L47

@Conradical314 The exact same thing happens to me,
I'm using FW v1.0.39 (model BCP-AT2N-P) and when I clear charging profiles i get this error:
```
2025-04-30 09:49:04.551 INFO (MainThread) [ocpp] 240300155: send [2,"73986dd1-09cc-4713-8028-cd3c769f03be","ClearChargingProfile",{}]
2025-04-30 09:49:04.648 INFO (MainThread) [ocpp] 240300155: receive message [3,"73986dd1-09cc-4713-8028-cd3c769f03be",{"status":"Unknown"}]
2025-04-30 09:49:04.649 WARNING (MainThread) [custom_components.ocpp] Failed with response: Unknown
```

Were you able to resolve your issue?

Idan37S, 2025-05-08:
For anyone who will encounter my issue, 
Disabling **OCA Mode** in the charger has fixed all my issues and everything is working great now.

GuzziDoug, 2025-05-08:
Idan,
 where is the OCA mode? (My charger is an early one, so it might be a later option??)
 Cannot see it in the list. (But my charger working fine, so not worried!)

Thx Doug

BrentAU, 2026-11-10:
I got similar errors.  Seemed like it might be a particular measurand failing to read properly and creating a bad JSON.  I fixed it by configuring less metervalues to be sampled. Couldn't tell you which one was causing it to fail,  but restricting to a smaller set worked for me. I fixed it in the config once, but managed to overwrite it somehow, so this time I just made an automation that reconfigures to this smaller set every hour... stupid but works, and will keep even if I accidentally reconfigure.  Seems to load these values reliably now.

alias: Set OCPP Config Every Hour
description: ''
triggers:
  - trigger: time_pattern
    minutes: '55'
conditions: []
actions:
  - action: ocpp.configure
    data:
      devid: charger6
      ocpp_key: MeterValuesSampledData
      value: Energy.Active.Import.Register,Power.Active.Import
mode: single