lbbrhzn/ocpp #1096: After HA reboot, MG / Starcharge Charger gets stuck in a reconnect loop

callifo, 2024-02-28:
**Describe the bug**
My MG (rebadged Starcharge) Charger, seems to not cope well with rebooting HA. Once HA starts backup, it seems to just reconnect every minute (reconnect count keeps going up), and you can't send commands to it reliably during this time. 

I've setup a handler, that watches the reconnect count, and forces a Reboot of the charger (via the OCPP button). This fixes it about 50-60% of the time. 

If the reboot doesn't fix it, I usually got to the wallcharger and pop the breaker. That always fixes it. 

Is there anything in the log that would suggest why it does this?

I can grab more log as an attached file, but there seems to be a limit on the amount of text I can include inline.

**To Reproduce**
Steps to reproduce the behavior:
Restart HA (e.g. Developer Tools -> Restart

**Expected behavior**
OCPP connection recover, and Charger usable after reboot.

**Screenshots**
[image]

```
2024-02-29 07:34:28.577 INFO (MainThread) [ocpp] 1000: send [2,"7b93654e-2e95-4cef-bf7d-9fa6f5ba794e","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-02-29 07:34:29.602 INFO (MainThread) [ocpp] 1000: receive message [3,"7b93654e-2e95-4cef-bf7d-9fa6f5ba794e",{ "configurationKey": [ { "key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger" } ], "unknownKey": [ ] }]
2024-02-29 07:34:29.603 INFO (MainThread) [ocpp] 1000: send [2,"45a625fb-1035-4f8d-84a5-5eaed3eaa927","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-02-29 07:34:30.620 INFO (MainThread) [ocpp] 1000: receive message [3,"45a625fb-1035-4f8d-84a5-5eaed3eaa927",{ "configurationKey": [ { "key": "NumberOfConnectors", "readonly": true, "value": "1" } ], "unknownKey": [ ] }]
2024-02-29 07:34:30.621 INFO (MainThread) [ocpp] 1000: send [2,"9ec71b8b-3cbe-4978-a07e-9091a940268a","GetConfiguration",{"key":["HeartbeatInterval"]}]
2024-02-29 07:34:31.631 INFO (MainThread) [ocpp] 1000: receive message [3,"9ec71b8b-3cbe-4978-a07e-9091a940268a",{ "configurationKey": [ { "key": "HeartbeatInterval", "readonly": false, "value": "3600" } ], "unknownKey": [ ] }]
2024-02-29 07:34:31.635 INFO (MainThread) [ocpp] 1000: send [2,"a50493cd-1da4-4b80-a55f-42fe91562efa","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export"}]
2024-02-29 07:34:32.633 INFO (MainThread) [ocpp] 1000: receive message [2,"bPDn0rDjcPjUNBqeDXXXExNGI9mMnIcl18tI","MeterValues",{ "connectorId": 1, "meterValue": [ { "timestamp": "2024-02-28T21:04:00.000Z", "sampledValue": [ {"value": "238.4", "context": "Sample.Clock", "measurand": "Voltage", "location": "Outlet", "unit": "V", "phase": "L1"},{"value": "0.01", "context": "Sample.Clock", "measurand": "Current.Import", "location": "Outlet", "unit": "A", "phase": "L1"},{ "value": "2", "context": "Sample.Clock", "measurand": "Power.Active.Import","location": "Outlet", "unit": "W" },{ "value": "0", "context": "Sample.Clock", "measurand": "Power.Offered","location": "Outlet", "unit": "W" },{ "value": "49700", "context": "Sample.Clock", "measurand": "Energy.Active.Import.Register","location": "Outlet", "unit": "Wh" } ] } ] }]
2024-02-29 07:34:32.636 INFO (MainThread) [ocpp] 1000: send [3,"bPDn0rDjcPjUNBqeDXXXExNGI9mMnIcl18tI",{}]
2024-02-29 07:34:33.633 INFO (MainThread) [ocpp] 1000: receive message [3,"a50493cd-1da4-4b80-a55f-42fe91562efa",{"status":"Rejected"}]
2024-02-29 07:34:33.634 INFO (MainThread) [ocpp] 1000: send [2,"d9cccea0-0c79-457f-a676-ce8b742e15ed","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
2024-02-29 07:34:34.719 INFO (MainThread) [ocpp] 1000: receive message [3,"d9cccea0-0c79-457f-a676-ce8b742e15ed",{"status":"Accepted"}]
2024-02-29 07:34:34.720 INFO (MainThread) [ocpp] 1000: send [2,"29ff9707-df43-4ef4-841e-dc919aff1e3d","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
2024-02-29 07:34:35.840 INFO (MainThread) [ocpp] 1000: receive message [3,"29ff9707-df43-4ef4-841e-dc919aff1e3d",{"status":"Accepted"}]
2024-02-29 07:34:35.842 INFO (MainThread) [ocpp] 1000: send [2,"549120bd-fc96-4361-a464-5f019db3ba56","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Export.Interval"}]
2024-02-29 07:34:36.580 INFO (MainThread) [ocpp] 1000: receive message [3,"549120bd-fc96-4361-a464-5f019db3ba56",{"status":"Rejected"}]
2024-02-29 07:34:36.581 INFO (MainThread) [ocpp] 1000: send [2,"60ec4829-6da5-45a8-bfde-31043173cd0f","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Export.Register"}]
2024-02-29 07:34:36.927 INFO (MainThread) [ocpp] 1000: receive message [3,"60ec4829-6da5-45a8-bfde-31043173cd0f",{"status":"Rejected"}]
2024-02-29 07:34:36.928 INFO (MainThread) [ocpp] 1000: send [2,"599947be-e91d-48c9-89a8-21b2c09b8890","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Interval"}]
2024-02-29 07:34:37.928 INFO (MainThread) [ocpp] 1000: receive message [3,"599947be-e91d-48c9-89a8-21b2c09b8890",{"status":"Rejected"}]
2024-02-29 07:34:37.929 INFO (MainThread) [ocpp] 1000: send [2,"1e20808b-63a5-4613-809b-3c93d7b4cba1","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
2024-02-29 07:34:38.928 INFO (MainThread) [ocpp] 1000: receive message [3,"1e20808b-63a5-4613-809b-3c93d7b4cba1",{"status":"Accepted"}]
2024-02-29 07:34:38.929 INFO (MainThread) [ocpp] 1000: send [2,"213b0fac-2481-4541-9079-ff6523a4601e","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Export.Interval"}]
2024-02-29 07:34:39.929 INFO (MainThread) [ocpp] 1000: receive message [3,"213b0fac-2481-4541-9079-ff6523a4601e",{"status":"Rejected"}]
2024-02-29 07:34:39.930 INFO (MainThread) [ocpp] 1000: send [2,"820e15ff-3722-487c-b49d-2b9ae2ac53cf","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Export.Register"}]
2024-02-29 07:34:40.930 INFO (MainThread) [ocpp] 1000: receive message [3,"820e15ff-3722-487c-b49d-2b9ae2ac53cf",{"status":"Rejected"}]
2024-02-29 07:34:40.931 INFO (MainThread) [ocpp] 1000: send [2,"c0cc86d2-1195-4424-b9cd-20975b3470f7","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Import.Interval"}]
2024-02-29 07:34:41.930 INFO (MainThread) [ocpp] 1000: receive message [3,"c0cc86d2-1195-4424-b9cd-20975b3470f7",{"status":"Rejected"}]
2024-02-29 07:34:41.932 INFO (MainThread) [ocpp] 1000: send [2,"9ac28a69-83a3-4f92-ace6-9fa98d5d9e37","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Reactive.Import.Register"}]
2024-02-29 07:34:42.930 INFO (MainThread) [ocpp] 1000: receive message [3,"9ac28a69-83a3-4f92-ace6-9fa98d5d9e37",{"status":"Rejected"}]
2024-02-29 07:34:42.934 INFO (MainThread) [ocpp] 1000: send [2,"caed0c82-f658-4fe9-a3d9-91f1baa23edb","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Frequency"}]
2024-02-29 07:34:43.931 INFO (MainThread) [ocpp] 1000: receive message [3,"caed0c82-f658-4fe9-a3d9-91f1baa23edb",{"status":"Rejected"}]
2024-02-29 07:34:43.932 INFO (MainThread) [ocpp] 1000: send [2,"fb7703f4-5769-4167-bd1f-0a5044d1fa24","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Export"}]
2024-02-29 07:34:44.931 INFO (MainThread) [ocpp] 1000: receive message [3,"fb7703f4-5769-4167-bd1f-0a5044d1fa24",{"status":"Rejected"}]
2024-02-29 07:34:44.932 INFO (MainThread) [ocpp] 1000: send [2,"9863fd56-75f6-4f16-9e1e-d67f51b0c74c","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
2024-02-29 07:34:45.933 INFO (MainThread) [ocpp] 1000: receive message [3,"9863fd56-75f6-4f16-9e1e-d67f51b0c74c",{"status":"Accepted"}]
2024-02-29 07:34:45.935 INFO (MainThread) [ocpp] 1000: send [2,"59ac358f-2031-48b5-9bb6-9e0f31d37515","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Factor"}]
2024-02-29 07:34:46.933 INFO (MainThread) [ocpp] 1000: receive message [3,"59ac358f-2031-48b5-9bb6-9e0f31d37515",{"status":"Rejected"}]
2024-02-29 07:34:46.934 INFO (MainThread) [ocpp] 1000: send [2,"45234d38-7881-4445-9269-300746955dfc","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
2024-02-29 07:34:47.933 INFO (MainThread) [ocpp] 1000: receive message [3,"45234d38-7881-4445-9269-300746955dfc",{"status":"Accepted"}]
[... cut ...] ", "unit": "A", "phase": "L1"},{ "value": "2", "context": "Sample.Clock", "measurand": "Power.Active.Import","location": "Outlet", "unit": "W" },{ "value": "0", "context": "Sample.Clock", "measurand": "Power.Offered","location": "Outlet", "unit": "W" },{ "value": "49700", "context": "Sample.Clock", "measurand": "Energy.Active.Import.Register","location": "Outlet", "unit": "Wh" } ] } ] }]
2024-02-29 07:42:31.930 INFO (MainThread) [ocpp] 1000: send [3,"e6ymtz1vyLo0rB5v5PZ8cwlI9vsvGCxFaEIk",{}]
```

drc38, 2024-03-25:
@callifo any improvement with 0.5.2?

callifo, 2024-03-26:
Will test today, however getting this at boot

```
2024-03-27 09:42:18.795 ERROR (Recorder) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/ocpp/charge_point.py", line 327, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 958, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/ocpp/charge_point.py", line 361, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 459, in post_connect
    await self.get_supported_features()
  File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 571, in get_supported_features
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/ocpp/charge_point.py", line 331, in call
    raise asyncio.TimeoutError(
TimeoutError: Waited 30s for response on [2,"c260adbb-b0f6-4211-808b-7712d9a6d2bd","GetConfiguration",{"key":["SupportedFeatureProfiles"]}].

```

callifo, 2024-03-27:
It charged OK yesterday, however this morning I wanted to test how it handled HA rebooting without me forcing a reboot before shutdown (previously to reduce the issue with the charger going into a reconnect loop, I would do a hard reboot of the charger via OCPP when HA shuts down). 

This time it seemed to be OK after HA started back up, however the OCPP interface just stopped working a few minutes later. Charger OCPP info in HA was good, however logging shows this,
```
2024-03-28 07:32:10.437 INFO (MainThread) [ocpp] 1000: receive message [3,"58d97dc9-d1bd-4879-97cd-55c166a8a1f6",{ "configurationKey": [ { "key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger" } ], "unknownKey": [ ] }]
2024-03-28 07:32:10.461 INFO (MainThread) [ocpp] 1000: send [2,"323cdc2a-ecac-4578-9366-0cef50f956e9","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-03-28 07:32:12.900 INFO (MainThread) [ocpp] 1000: receive message [3,"323cdc2a-ecac-4578-9366-0cef50f956e9",{ "configurationKey": [ { "key": "NumberOfConnectors", "readonly": true, "value": "1" } ], "unknownKey": [ ] }]
2024-03-28 07:32:12.903 INFO (MainThread) [ocpp] 1000: send [2,"f1761767-bb89-41e9-830a-54fc83abb1f1","GetConfiguration",{"key":["HeartbeatInterval"]}]
2024-03-28 07:32:15.500 INFO (MainThread) [ocpp] 1000: receive message [3,"f1761767-bb89-41e9-830a-54fc83abb1f1",{ "configurationKey": [ { "key": "HeartbeatInterval", "readonly": false, "value": "3600" } ], "unknownKey": [ ] }]
2024-03-28 07:32:15.517 INFO (MainThread) [ocpp] 1000: send [2,"f0444b03-0101-47a8-aaf8-f3c3aa601753","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
2024-03-28 07:32:16.840 INFO (MainThread) [ocpp] 1000: receive message [3,"f0444b03-0101-47a8-aaf8-f3c3aa601753",{"status":"Accepted"}]
2024-03-28 07:32:16.866 INFO (MainThread) [ocpp] 1000: send [2,"3118d0fd-eb18-4a5d-97e6-ba754b206456","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
2024-03-28 07:32:18.247 INFO (MainThread) [ocpp] 1000: receive message [3,"3118d0fd-eb18-4a5d-97e6-ba754b206456",{"status":"Accepted"}]
2024-03-28 07:32:18.254 INFO (MainThread) [ocpp] 1000: send [2,"5322ffcf-67f4-4dc5-8a2b-e5919b9ae18a","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
2024-03-28 07:32:19.458 INFO (MainThread) [ocpp] 1000: receive message [3,"5322ffcf-67f4-4dc5-8a2b-e5919b9ae18a",{"status":"Accepted"}]
2024-03-28 07:32:19.483 INFO (MainThread) [ocpp] 1000: send [2,"d284e609-b56d-4699-a726-b0615d316250","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
2024-03-28 07:32:21.041 INFO (MainThread) [ocpp] 1000: receive message [3,"d284e609-b56d-4699-a726-b0615d316250",{"status":"Accepted"}]
2024-03-28 07:32:21.056 INFO (MainThread) [ocpp] 1000: send [2,"2c044604-e2ad-4640-a3cf-f0c6f18d14c1","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
2024-03-28 07:32:23.640 INFO (MainThread) [ocpp] 1000: receive message [3,"2c044604-e2ad-4640-a3cf-f0c6f18d14c1",{"status":"Accepted"}]
2024-03-28 07:32:23.691 INFO (MainThread) [ocpp] 1000: send [2,"d1277859-ca00-43b1-94bb-e111f24e6610","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Temperature"}]
2024-03-28 07:32:24.983 INFO (MainThread) [ocpp] 1000: receive message [3,"d1277859-ca00-43b1-94bb-e111f24e6610",{"status":"Accepted"}]
2024-03-28 07:32:25.083 INFO (MainThread) [ocpp] 1000: send [2,"760fe9a5-3150-4f1c-93c5-1971f92cfe40","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
2024-03-28 07:32:28.631 INFO (MainThread) [ocpp] 1000: receive message [3,"760fe9a5-3150-4f1c-93c5-1971f92cfe40",{"status":"Accepted"}]
2024-03-28 07:32:28.649 INFO (MainThread) [ocpp] 1000: send [2,"081479e7-3ab9-4328-a31f-fa8c1263d543","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2024-03-28 07:32:30.030 INFO (MainThread) [ocpp] 1000: receive message [3,"081479e7-3ab9-4328-a31f-fa8c1263d543",{ "configurationKey": [ { "key": "MeterValuesSampledData", "readonly": false, "value": "Voltage" } ], "unknownKey": [ ] }]
2024-03-28 07:32:30.047 INFO (MainThread) [ocpp] 1000: send [2,"9af6a3e8-5fae-4f5e-be41-07532ce58bf5","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Power.Offered,Temperature,Voltage"}]
2024-03-28 07:32:32.894 INFO (MainThread) [ocpp] 1000: receive message [3,"9af6a3e8-5fae-4f5e-be41-07532ce58bf5",{"status":"Accepted"}]
2024-03-28 07:32:33.400 INFO (MainThread) [ocpp] 1000: send [2,"6e4c0e47-34b8-42ee-b86c-24231a2a3b83","GetConfiguration",{"key":["MeterValueSampleInterval"]}]
2024-03-28 07:32:34.372 INFO (MainThread) [ocpp] 1000: receive message [2,"2raAwZNGs9HIw7p3zbiXLzXp6YiR4KJLR6hY","MeterValues",{ "connectorId": 1, "meterValue": [ { "timestamp": "2024-03-27T21:02:00.000Z", "sampledValue": [ {"value": "239.0", "context": "Sample.Clock", "measurand": "Voltage", "location": "Outlet", "unit": "V", "phase": "L1"},{"value": "0.01", "context": "Sample.Clock", "measurand": "Current.Import", "location": "Outlet", "unit": "A", "phase": "L1"},{ "value": "2", "context": "Sample.Clock", "measurand": "Power.Active.Import","location": "Outlet", "unit": "W" },{ "value": "3300", "context": "Sample.Clock", "measurand": "Power.Offered","location": "Outlet", "unit": "W" },{ "value": "197940", "context": "Sample.C [... cut ...]
2024-03-28 07:32:40.310 INFO (MainThread) [ocpp] 1000: receive message [2,"rwufWQyhpSfP1SzNamzemiJLSe3bk0li2jSZ","StatusNotification",{ "connectorId": 1, "errorCode": "NoError", "status": "Available", "timestamp": "2024-03-27T21:02:36.000Z", "vendorId": "StarCharge" }]
2024-03-28 07:32:40.311 INFO (MainThread) [ocpp] 1000: send [3,"rwufWQyhpSfP1SzNamzemiJLSe3bk0li2jSZ",{}]
...
Nothing for 2.5 hours
....
I rebooted it when I noticed
...
2024-03-28 09:58:35.881 INFO (MainThread) [ocpp] 1000: send [2,"0f1d3995-7c3a-43f3-a519-c6817dbf95b4","Reset",{"type":"Hard"}]
```

I forced a OCPP reboot of the charger almost 2.5 hours after it (or HA?) stopped talking OCPP. Once it came back up it was ok.

drc38, 2024-03-30:
You can try using one of the ocpp services and see if the charger is still responsive. 

Also check what is happening at the websocket level.

callifo, 2024-04-03:
Debugging + websockets
```
2024-04-03 17:05:07.108 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2024-04-03 17:05:07.108 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/1000
2024-04-03 17:05:07.108 INFO (MainThread) [custom_components.ocpp] Charger 1000 connected to 0.0.0.0:8887.
2024-04-03 17:05:17.688 INFO (MainThread) [ocpp] 1000: send [2,"313d85bc-4d01-4b2b-aa24-146a6af98eaf","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-04-03 17:05:18.350 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=0.0 ms, pong=605.0 ms
2024-04-03 17:05:18.351 INFO (MainThread) [ocpp] 1000: receive message [3,"313d85bc-4d01-4b2b-aa24-146a6af98eaf",{ "configurationKey": [ { "key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger" } ], "unknownKey": [ ] }]
2024-04-03 17:05:18.457 DEBUG (MainThread) [custom_components.ocpp] Feature profiles returned: Profiles.CORE|FW|SMART|RES|REM|AUTH
2024-04-03 17:05:18.459 INFO (MainThread) [ocpp] 1000: send [2,"3328987b-9807-46c0-896f-da9dc965d5cb","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-04-03 17:05:19.025 INFO (MainThread) [ocpp] 1000: receive message [3,"3328987b-9807-46c0-896f-da9dc965d5cb",{ "configurationKey": [ { "key": "NumberOfConnectors", "readonly": true, "value": "1" } ], "unknownKey": [ ] }]
2024-04-03 17:05:19.035 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for NumberOfConnectors: 1
2024-04-03 17:05:19.036 INFO (MainThread) [ocpp] 1000: send [2,"d8ba51d7-04f4-46df-89b3-7aed59695bf2","GetConfiguration",{"key":["HeartbeatInterval"]}]
2024-04-03 17:05:20.372 INFO (MainThread) [ocpp] 1000: receive message [3,"d8ba51d7-04f4-46df-89b3-7aed59695bf2",{ "configurationKey": [ { "key": "HeartbeatInterval", "readonly": false, "value": "3600" } ], "unknownKey": [ ] }]
2024-04-03 17:05:20.375 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for HeartbeatInterval: 3600
2024-04-03 17:05:20.375 DEBUG (MainThread) [custom_components.ocpp] '1000' trying measurand 'Current.Import'
2024-04-03 17:05:20.380 INFO (MainThread) [ocpp] 1000: send [2,"dab745fa-0673-4c77-9142-6e498618b04b","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
2024-04-03 17:05:20.857 INFO (MainThread) [ocpp] 1000: receive message [3,"dab745fa-0673-4c77-9142-6e498618b04b",{"status":"Accepted"}]
2024-04-03 17:05:20.860 DEBUG (MainThread) [custom_components.ocpp] '1000' adding measurand 'Current.Import'
2024-04-03 17:05:20.861 DEBUG (MainThread) [custom_components.ocpp] '1000' trying measurand 'Current.Offered'
2024-04-03 17:05:20.862 INFO (MainThread) [ocpp] 1000: send [2,"4c1a5081-b3d1-4b3a-a703-a648454d9dc8","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
2024-04-03 17:05:21.715 INFO (MainThread) [ocpp] 1000: receive message [3,"4c1a5081-b3d1-4b3a-a703-a648454d9dc8",{"status":"Accepted"}]
2024-04-03 17:05:21.716 DEBUG (MainThread) [custom_components.ocpp] '1000' adding measurand 'Current.Offered'
2024-04-03 17:05:21.716 DEBUG (MainThread) [custom_components.ocpp] '1000' trying measurand 'Energy.Active.Import.Register'
2024-04-03 17:05:21.716 INFO (MainThread) [ocpp] 1000: send [2,"239068aa-1a49-4250-8305-969bfd287988","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
2024-04-03 17:05:23.347 INFO (MainThread) [ocpp] 1000: receive message [3,"239068aa-1a49-4250-8305-969bfd287988",{"status":"Accepted"}]
2024-04-03 17:05:23.350 DEBUG (MainThread) [custom_components.ocpp] '1000' adding measurand 'Energy.Active.Import.Register'
2024-04-03 17:05:23.350 DEBUG (MainThread) [custom_components.ocpp] '1000' trying measurand 'Power.Active.Import'
2024-04-03 17:05:23.351 INFO (MainThread) [ocpp] 1000: send [2,"7dc365da-5009-4d8d-8ee3-5f91ca3da1a6","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
2024-04-03 17:05:24.044 INFO (MainThread) [ocpp] 1000: receive message [3,"7dc365da-5009-4d8d-8ee3-5f91ca3da1a6",{"status":"Accepted"}]
2024-04-03 17:05:24.136 DEBUG (MainThread) [custom_components.ocpp] '1000' adding measurand 'Power.Active.Import'
2024-04-03 17:05:24.136 DEBUG (MainThread) [custom_components.ocpp] '1000' trying measurand 'Power.Offered'
2024-04-03 17:05:24.137 INFO (MainThread) [ocpp] 1000: send [2,"a051d911-3f25-4db7-994b-c698ff99a5e5","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
2024-04-03 17:05:25.390 INFO (MainThread) [ocpp] 1000: receive message [3,"a051d911-3f25-4db7-994b-c698ff99a5e5",{"status":"Accepted"}]
2024-04-03 17:05:25.417 DEBUG (MainThread) [custom_components.ocpp] '1000' adding measurand 'Power.Offered'
2024-04-03 17:05:25.417 DEBUG (MainThread) [custom_components.ocpp] '1000' trying measurand 'Temperature'
2024-04-03 17:05:25.417 INFO (MainThread) [ocpp] 1000: send [2,"0df2f4ce-24c7-483f-9763-db5685758d81","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Temperature"}]
2024-04-03 17:05:26.200 INFO (MainThread) [ocpp] 1000: receive message [3,"0df2f4ce-24c7-483f-9763-db5685758d81",{"status":"Accepted"}]
2024-04-03 17:05:26.251 DEBUG (MainThread) [custom_components.ocpp] '1000' adding measurand 'Temperature'
2024-04-03 17:05:26.251 DEBUG (MainThread) [custom_components.ocpp] '1000' trying measurand 'Voltage'
[... cut ...]
2024-04-03 17:14:07.912 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=0.0 ms, pong=2.0 ms
2024-04-03 17:14:17.916 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=0.0 ms, pong=2.0 ms
2024-04-03 17:14:27.919 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=0.0 ms, pong=2.0 ms
2024-04-03 17:14:37.922 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=0.0 ms, pong=2.0 ms
```

callifo, 2024-04-03:
Seems like it dies right after
```
2024-04-03 17:06:24.714 INFO (MainThread) [ocpp] 1000: send [2,"dc581a66-730e-4884-98d1-b5c831d50fd0","TriggerMessage",{"requestedMessage":"BootNotification"}]
2024-04-03 17:06:25.709 INFO (MainThread) [ocpp] 1000: receive message [2,"SLPsY0eiqwkVs5XResWERsqicZ0xrKi0JlXU","StatusNotification",{ "connectorId": 0, "errorCode": "NoError", "status": "Available", "timestamp": "2024-04-03T06:36:23.000Z", "vendorId": "StarCharge" }]
2024-04-03 17:06:25.711 INFO (MainThread) [ocpp] 1000: send [3,"SLPsY0eiqwkVs5XResWERsqicZ0xrKi0JlXU",{}]
2024-04-03 17:06:27.157 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=1.0 ms, pong=3.0 ms
2024-04-03 17:06:37.161 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=0.0 ms, pong=2.0 ms
2024-04-03 17:06:47.164 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=0.0 ms, pong=2.0 ms
2024-04-03 17:06:54.715 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to '1000'
2024-04-03 17:06:54.721 INFO (MainThread) [custom_components.ocpp] Charger 1000 disconnected from 0.0.0.0:8887.
```
Although ping pong's are still going.

Sent this through Services,
```
service: ocpp.get_configuration
data:
  ocpp_key: WebSocketPingInterval
```
No response,
```
2024-04-03 17:17:45.938 INFO (MainThread) [ocpp] 1000: send [2,"77a4625c-9d14-4475-9beb-38344583b57e","GetConfiguration",{"key":["WebSocketPingInterval"]}]
2024-04-03 17:18:15.942 ERROR (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Waited 30s for response on [2,"77a4625c-9d14-4475-9beb-38344583b57e","GetConfiguration",{"key":["WebSocketPingInterval"]}].
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/ocpp/charge_point.py", line 327, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 958, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/ocpp/charge_point.py", line 361, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/homeassistant/helpers/script.py", line 485, in _async_step
    await getattr(self, handler)()
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/homeassistant/helpers/script.py", line 723, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/homeassistant/helpers/script.py", line 685, in _async_run_long_action
    return long_task.result()
           ^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/homeassistant/core.py", line 2279, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/homeassistant/core.py", line 2316, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 417, in handle_get_configuration
    await self.get_configuration(key)
  File "/home/homeassistant/.homeassistant/custom_components/ocpp/api.py", line 893, in get_configuration
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant_p3_12/lib/python3.12/site-packages/ocpp/charge_point.py", line 331, in call
    raise asyncio.TimeoutError(
TimeoutError: Waited 30s for response on [2,"77a4625c-9d14-4475-9beb-38344583b57e","GetConfiguration",{"key":["WebSocketPingInterval"]}].
2024-04-03 17:18:15.969 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140109469919424] Error handling message: Timeout (timeout) Nick from 192.168.10.100 (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/123.0.0.0 Safari/537.36 Edg/123.0.0.0)
```

drc38, 2024-04-03:
Very strange, looks like an issue at the charger end. You can try the diagnostic service to see if the charger is logging information that may help understand what's going on.

callifo, 2024-04-03:
This appears to be the valid section, though I dont find it particularly readable. Note the charger does not support 1/2hr timezones so its 30 minutes behind HA logs.

```
2024-04-03T16:35:36+0900 [ocpp] net_interface nums: 4
Interface:    eth0	 IP:  192.168.100.79	 
Interface:  eth0:1	 IP:  192.168.88.206	 
Interface:   wlan0	 IP:   192.168.1.136	 
ICCID=+CME ERROR: SIM failure
AutoConnect=TRUE
Online=FALSE
Carrier=
APN=
RetryTimes=0

2024-04-03T16:35:38+0900 [comproc] ##########_HJ CardReader_##########: CardReader ComErr!
2024-04-03T16:35:38+0900 [comproc] =>pwrboard:AA FF FF FF 02 00 FF 85 C0 
2024-04-03T16:35:46+0900 [comproc] ##########_HJ CardReader_##########: CardReader ComErr!
2024-04-03T16:35:46+0900 [comproc] =>pwrboard:AA FF FF FF 02 00 FF 85 C0 
2024-04-03T16:35:55+0900 [comproc] ##########_HJ CardReader_##########: CardReader ComErr!
2024-04-03T16:35:55+0900 [comproc] =>pwrboard:AA FF FF FF 02 00 FF 85 C0 
2024-04-03T16:36:00+0900 [comproc] =>pwrboard:AA EE EE 04 01 01 BF 44 
2024-04-03T16:36:00+0900 [ocpp] =>Dbus:FF 6A A6 EE EE 00 07 [00 04] 00 01 01 47 C0 
2024-04-03T16:36:00+0900 [comproc] <=pwrboard:BB EE EE 04 1E 01 63 57 00 00 5D 09 01 00 00 00 00 00 2E 00 01 5D 09 01 00 00 00 00 00 00 00 00 00 00 00 3B 9B 
2024-04-03T16:36:01+0900 [ocpp] <=Dbus:FF 6A A6 EE EE 00 24 [00 04] 00 01 01 00 00 57 63 09 5D 00 01 00 00 00 00 00 00 00 00 00 09 5D 00 01 00 00 00 00 2E 01 00 00 D6 84 
2024-04-03T16:36:01+0900 [ocpp] connId:1,connectorType:1,meterval:223710,AC_voltage_A:239.7,AC_current_A:0.01,MaxPowerOffered:0,temperature:46,soc:0,AC_Power:2
2024-04-03T16:36:01+0900 [ocpp] =>CSMS:[2,"mez4CeiYQDgCsjt9DzZXGbJx9uSVWj4XcGn1","MeterValues",{ "connectorId": 1, "meterValue": [ { "timestamp": "2024-04-03T06:35:30.000Z", "sampledValue": [ {"value": "239.7", "context": "Sample.Clock", "measurand": "Voltage", "location": "Outlet", "unit": "V", "phase": "L1"},{"value": "0.01", "context": "Sample.Clock", "measurand": "Current.Import", "location": "Outlet", "unit": "A", "phase": "L1"},{ "value": "2", "context": "Sample.Clock", "measurand": "Power.Active.Import","location": "Outlet", "unit": "W" },{ "value": "0", "context": "Sample.Clock", "measurand": "Power.Offered","location": "Outlet", "unit": "W" },{ "value": "223710", "context": "Sample.Clock", "measurand": "Energy.Active.Import.Register","location": "Outlet", "unit": "Wh" } ] } ] }]
2024-04-03T16:36:01+0900 [comproc] <=pwrboard:BB 19 20 01 0B 01 01 06 00 00 00 00 00 09 2E 25 23 E1 
2024-04-03T16:36:01+0900 [comproc] icomCode = 0,index=0
2024-04-03T16:36:01+0900 [comproc]  inside error test : board warnning RelayAdhesion error 

2024-04-03T16:36:01+0900 [comproc] <=pwrboard:BB EE EE 2D 01 01 36 96 
2024-04-03T16:36:01+0900 [comproc]  boardtemp = 46, conntemp = 37 ;

2024-04-03T16:36:01+0900 [comproc] pwrCom error Code vol:0,1,9
2024-04-03T16:36:01+0900 [comproc] ocpp error Code :6,0
2024-04-03T16:36:01+0900 [ocpp] <=Dbus:FF 6A A6 19 20 00 0F [00 01] 00 01 01 01 06 00 00 00 00 00 00 6A BB 
2024-04-03T16:36:01+0900 [ocpp] <=Dbus:FF 6A A6 EE EE 00 05 [00 2D] 01 1C DE 
2024-04-03T16:36:01+0900 [comproc] =>pwrboard:AA 96 D8 2C 04 00 00 00 00 70 AB 
2024-04-03T16:36:01+0900 [ocpp] =>Dbus:FF 6A A6 96 D8 00 06 [00 2D] 00 00 49 89 
2024-04-03T16:36:01+0900 [comproc] =>pwrboard:AA 19 20 01 04 01 03 E7 01 2E E7 
2024-04-03T16:36:01+0900 [ocpp] =>Dbus:FF 6A A6 19 20 00 04 [00 01] D4 03 
2024-04-03T16:36:01+0900 [ocpp] <=Dbus:FF 6A A6 17 52 00 04 [50 09] 37 07 
2024-04-03T16:36:02+0900 [ocpp] <=CSMS:[3,"mez4CeiYQDgCsjt9DzZXGbJx9uSVWj4XcGn1",{}]
2024-04-03T16:36:03+0900 [comproc] ##########_HJ CardReader_##########: CardReader ComErr!
2024-04-03T16:36:03+0900 [comproc] =>pwrboard:AA FF FF FF 02 00 FF 85 C0 
2024-04-03T16:36:03+0900 [ocpp] <=CSMS:[99,"Proprietary Protocol","Connection",{"connectionStatus":0}]
2024-04-03T16:36:03+0900 [ocpp] -------------Disconnect from the server----------------
2024-04-03T16:36:06+0900 [ocpp] <=CSMS:[99,"Proprietary Protocol","Connection",{"connectionStatus":1}]
2024-04-03T16:36:06+0900 [ocpp] -------------Connect with server success----------------
2024-04-03T16:36:08+0900 [ocpp] <=CSMS:[2,"2dffff8d-2b73-4f90-9593-f8f6bbe4f128","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-04-03T16:36:08+0900 [ocpp] =>CSMS:[3,"2dffff8d-2b73-4f90-9593-f8f6bbe4f128",{ "configurationKey": [ { "key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger" } ], "unknownKey": [ ] }]
2024-04-03T16:36:09+0900 [ocpp] <=CSMS:[2,"97def3ae-781b-4464-ba4a-8cd1998af242","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-04-03T16:36:09+0900 [ocpp] =>CSMS:[3,"97def3ae-781b-4464-ba4a-8cd1998af242",{ "configurationKey": [ { "key": "NumberOfConnectors", "readonly": true, "value": "1" } ], "unknownKey": [ ] }]
2024-04-03T16:36:10+0900 [ocpp] <=CSMS:[2,"ceb9b107-1120-4be5-b668-11a0bd3faab5","GetConfiguration",{"key":["HeartbeatInterval"]}]
2024-04-03T16:36:10+0900 [ocpp] =>CSMS:[3,"ceb9b107-1120-4be5-b668-11a0bd3faab5",{ "configurationKey": [ { "key": "HeartbeatInterval", "readonly": false, "value": "3600" } ], "unknownKey": [ ] }]
2024-04-03T16:36:11+0900 [comproc] ##########_HJ CardReader_##########: CardReader ComErr!
2024-04-03T16:36:11+0900 [comproc] =>pwrboard:AA FF FF FF 02 00 FF 85 C0 
2024-04-03T16:36:11+0900 [ocpp] <=CSMS:[2,"ab674f37-87c7-49e6-aa13-0b59a2ba740f","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
[... cut ...] "Outlet", "unit": "W" },{ "value": "223710", "context": "Sample.Clock", "measurand": "Energy.Active.Import.Register","location": "Outlet", "unit": "Wh" } ] } ] }]
2024-04-03T16:55:32+0900 [ocpp] <=CSMS:[3,"NPZZ1K3Htzya7PD6mkGodUkUtC6z1LBTnn2p",{}]
2024-04-03T16:55:37+0900 [comproc] <=pwrboard:BB 1E 32 01 0B 01 01 06 00 00 00 00 00 09 2D 23 C1 34 
2024-04-03T16:55:37+0900 [comproc] icomCode = 0,index=0
2024-04-03T16:55:37+0900 [comproc]  inside error test : board warnning RelayAdhesion error 
```

callifo, 2024-04-03:
Unfortunately not, MG dont respond to multiple attempts at emails. 

This bit
```
2024-04-03 17:06:24.714 INFO (MainThread) [ocpp] 1000: send [2,"dc581a66-730e-4884-98d1-b5c831d50fd0","TriggerMessage",{"requestedMessage":"BootNotification"}]
2024-04-03 17:06:25.709 INFO (MainThread) [ocpp] 1000: receive message [2,"SLPsY0eiqwkVs5XResWERsqicZ0xrKi0JlXU","StatusNotification",{ "connectorId": 0, "errorCode": "NoError", "status": "Available", "timestamp": "2024-04-03T06:36:23.000Z", "vendorId": "StarCharge" }]
2024-04-03 17:06:25.711 INFO (MainThread) [ocpp] 1000: send [3,"SLPsY0eiqwkVs5XResWERsqicZ0xrKi0JlXU",{}]
2024-04-03 17:06:27.157 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=1.0 ms, pong=3.0 ms
2024-04-03 17:06:37.161 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=0.0 ms, pong=2.0 ms
2024-04-03 17:06:47.164 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=0.0 ms, pong=2.0 ms
2024-04-03 17:06:54.715 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to '1000'
2024-04-03 17:06:54.721 INFO (MainThread) [custom_components.ocpp] Charger 1000 disconnected from 0.0.0.0:8887.
2024-04-03 17:06:57.729 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2024-04-03 17:06:57.729 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/1000
2024-04-03 17:06:57.729 INFO (MainThread) [custom_components.ocpp] Charger 1000 reconnected to 0.0.0.0:8887.
2024-04-03 17:06:57.729 DEBUG (MainThread) [custom_components.ocpp] Reconnect websocket to 1000
```
Correlates to this bit
```
2024-04-03T16:36:44+0900 [comproc] ##########_HJ CardReader_##########: CardReader ComErr!
2024-04-03T16:36:44+0900 [comproc] =>pwrboard:AA FF FF FF 02 00 FF 85 C0 
2024-04-03T16:36:49+0900 [ocpp] [2024-04-03T06:35:48.000Z]Network status is:[Connected]!
2024-04-03T16:36:52+0900 [comproc] ##########_HJ CardReader_##########: CardReader ComErr!
2024-04-03T16:36:52+0900 [comproc] =>pwrboard:AA FF FF FF 02 00 FF 85 C0 
2024-04-03T16:36:54+0900 [ocpp] <=CSMS:[99,"Proprietary Protocol","Connection",{"connectionStatus":0}]
2024-04-03T16:36:54+0900 [ocpp] -------------Disconnect from the server----------------
2024-04-03T16:36:57+0900 [ocpp] <=CSMS:[99,"Proprietary Protocol","Connection",{"connectionStatus":1}]
2024-04-03T16:36:57+0900 [ocpp] -------------Connect with server success----------------
2024-04-03T16:36:59+0900 [comproc] icomCode = 0,index=0
2024-04-03T16:36:59+0900 [comproc]  inside error test : board warnning RelayAdhesion error 
```

Is this suggesting HA is closing the connection?

drc38, 2024-04-03:
You need to debug the websocket messages to see why the connection closes.

callifo, 2024-04-03:
How do I do that? I've already got this in the config,
```
    custom_components.ocpp: debug
    websocket: debug
```

drc38, 2024-04-03:
It is `websockets`

callifo, 2024-04-03:
Here is the debug level for websockets + ocpp

```
2024-04-03 18:49:36.318 INFO (MainThread) [ocpp] 1000: send [2,"ed53da3d-f0f3-4020-869e-301c1acba5de","TriggerMessage",{"requestedMessage":"BootNotification"}]
2024-04-03 18:49:36.318 DEBUG (MainThread) [websockets.server] > TEXT '[2,"ed53da3d-f0f3-4020-869e-301c1acba5de","Trig...e":"BootNotification"}]' [99 bytes]
2024-04-03 18:49:36.319 DEBUG (MainThread) [websockets.server] < PONG 0d 27 82 ce [binary, 4 bytes]
2024-04-03 18:49:36.320 DEBUG (MainThread) [websockets.server] < TEXT '[2,"Ino6nMdv51SQpJkE6bOex4nWQs8YhfmmEMdk","Stat...dorId": "StarCharge" }]' [198 bytes]
2024-04-03 18:49:36.321 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=1.0 ms, pong=7.0 ms
2024-04-03 18:49:36.321 INFO (MainThread) [ocpp] 1000: receive message [2,"Ino6nMdv51SQpJkE6bOex4nWQs8YhfmmEMdk","StatusNotification",{ "connectorId": 0, "errorCode": "NoError", "status": "Available", "timestamp": "2024-04-03T08:19:35.000Z", "vendorId": "StarCharge" }]
2024-04-03 18:49:36.322 INFO (MainThread) [ocpp] 1000: send [3,"Ino6nMdv51SQpJkE6bOex4nWQs8YhfmmEMdk",{}]
2024-04-03 18:49:36.322 DEBUG (MainThread) [websockets.server] > TEXT '[3,"Ino6nMdv51SQpJkE6bOex4nWQs8YhfmmEMdk",{}]' [45 bytes]
2024-04-03 18:49:37.315 DEBUG (MainThread) [websockets.server] < TEXT '[2,"sNOgNwDVfb20PtKogByEXeN60Si8HFMMowDK","Stat...dorId": "StarCharge" }]' [198 bytes]
2024-04-03 18:49:37.315 INFO (MainThread) [ocpp] 1000: receive message [2,"sNOgNwDVfb20PtKogByEXeN60Si8HFMMowDK","StatusNotification",{ "connectorId": 1, "errorCode": "NoError", "status": "Preparing", "timestamp": "2024-04-03T08:19:35.000Z", "vendorId": "StarCharge" }]
2024-04-03 18:49:37.318 INFO (MainThread) [ocpp] 1000: send [3,"sNOgNwDVfb20PtKogByEXeN60Si8HFMMowDK",{}]
2024-04-03 18:49:37.318 DEBUG (MainThread) [websockets.server] > TEXT '[3,"sNOgNwDVfb20PtKogByEXeN60Si8HFMMowDK",{}]' [45 bytes]
2024-04-03 18:49:46.322 DEBUG (MainThread) [websockets.server] > PING 9c 1c 89 bd [binary, 4 bytes]
2024-04-03 18:49:46.324 DEBUG (MainThread) [websockets.server] < PONG 9c 1c 89 bd [binary, 4 bytes]
2024-04-03 18:49:46.325 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=2.0 ms, pong=2.0 ms
2024-04-03 18:49:56.326 DEBUG (MainThread) [websockets.server] > PING 'w<x)' [text, 4 bytes]
2024-04-03 18:49:56.329 DEBUG (MainThread) [websockets.server] < PONG 'w<x)' [text, 4 bytes]
2024-04-03 18:49:56.330 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=1.0 ms, pong=3.0 ms
2024-04-03 18:50:06.322 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to '1000'
2024-04-03 18:50:06.322 DEBUG (MainThread) [websockets.server] = connection is CLOSING
2024-04-03 18:50:06.322 DEBUG (MainThread) [websockets.server] > CLOSE 1000 (OK) [2 bytes]
2024-04-03 18:50:06.325 DEBUG (MainThread) [websockets.server] < CLOSE 1000 (OK) [2 bytes]
2024-04-03 18:50:06.325 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2024-04-03 18:50:06.325 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-04-03 18:50:06.326 INFO (MainThread) [custom_components.ocpp] Charger 1000 disconnected from 0.0.0.0:8887.
2024-04-03 18:50:06.327 INFO (MainThread) [websockets.server] connection closed
2024-04-03 18:50:09.333 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-04-03 18:50:09.337 DEBUG (MainThread) [websockets.server] < GET /1000 HTTP/1.1
2024-04-03 18:50:09.338 DEBUG (MainThread) [websockets.server] < Pragma: no-cache
2024-04-03 18:50:09.338 DEBUG (MainThread) [websockets.server] < Cache-Control: no-cache
2024-04-03 18:50:09.338 DEBUG (MainThread) [websockets.server] < Host: 192.168.100.7
2024-04-03 18:50:09.338 DEBUG (MainThread) [websockets.server] < Origin: http://192.168.100.7
2024-04-03 18:50:09.338 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2024-04-03 18:50:09.338 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2024-04-03 18:50:09.338 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: FkDQWpC7YHEW7IbwvNU5ZA==
2024-04-03 18:50:09.338 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2024-04-03 18:50:09.338 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2024-04-03 18:50:09.339 DEBUG (MainThread) [websockets.server] < authorization: Basic MTAwMDoxMjM=
2024-04-03 18:50:09.339 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2024-04-03 18:50:09.339 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2024-04-03 18:50:09.339 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2024-04-03 18:50:09.339 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: DGVu2hrJEz+yfrNcibew2fyuz8A=
2024-04-03 18:50:09.340 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
2024-04-03 18:50:09.340 DEBUG (MainThread) [websockets.server] > Date: Wed, 03 Apr 2024 08:20:09 GMT
2024-04-03 18:50:09.340 DEBUG (MainThread) [websockets.server] > Server: Python/3.12 websockets/12.0
2024-04-03 18:50:09.342 INFO (MainThread) [websockets.server] connection open
2024-04-03 18:50:09.343 DEBUG (MainThread) [websockets.server] = connection is OPEN
2024-04-03 18:50:09.343 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2024-04-03 18:50:09.343 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/1000
[... cut ...]
2024-04-03 18:50:19.349 DEBUG (MainThread) [websockets.server] < PONG c9 bd 7e ab [binary, 4 bytes]
2024-04-03 18:50:19.349 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=2.0 ms, pong=2.0 ms
2024-04-03 18:50:29.350 DEBUG (MainThread) [websockets.server] > PING '\x14\x1c\x06b' [text, 4 bytes]
2024-04-03 18:50:29.354 DEBUG (MainThread) [websockets.server] < PONG '\x14\x1c\x06b' [text, 4 bytes]
2024-04-03 18:50:29.355 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=2.0 ms, pong=3.0 ms
```

callifo, 2024-04-03:
Here is after sending ocpp reboot to charger to fix it,

```
2024-04-03 20:50:54.957 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-04-03 20:50:54.965 DEBUG (MainThread) [websockets.server] < GET /1000 HTTP/1.1
2024-04-03 20:50:54.965 DEBUG (MainThread) [websockets.server] < Pragma: no-cache
2024-04-03 20:50:54.966 DEBUG (MainThread) [websockets.server] < Cache-Control: no-cache
2024-04-03 20:50:54.966 DEBUG (MainThread) [websockets.server] < Host: 192.168.100.7
2024-04-03 20:50:54.966 DEBUG (MainThread) [websockets.server] < Origin: http://192.168.100.7
2024-04-03 20:50:54.966 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2024-04-03 20:50:54.966 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2024-04-03 20:50:54.966 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: XTH1OKWvFYV2xmipWc2l9A==
2024-04-03 20:50:54.966 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2024-04-03 20:50:54.966 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2024-04-03 20:50:54.966 DEBUG (MainThread) [websockets.server] < authorization: Basic MTAwMDoxMjM=
2024-04-03 20:50:54.967 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2024-04-03 20:50:54.967 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2024-04-03 20:50:54.967 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2024-04-03 20:50:54.967 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: DzsnTlwBpzPdDG5kGxONLpL7xko=
2024-04-03 20:50:54.967 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
2024-04-03 20:50:54.967 DEBUG (MainThread) [websockets.server] > Date: Wed, 03 Apr 2024 10:20:54 GMT
2024-04-03 20:50:54.967 DEBUG (MainThread) [websockets.server] > Server: Python/3.12 websockets/12.0
2024-04-03 20:50:54.970 INFO (MainThread) [websockets.server] connection open
2024-04-03 20:50:54.970 DEBUG (MainThread) [websockets.server] = connection is OPEN
2024-04-03 20:50:54.971 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2024-04-03 20:50:54.971 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/1000
2024-04-03 20:50:54.971 INFO (MainThread) [custom_components.ocpp] Charger 1000 reconnected to 0.0.0.0:8887.
2024-04-03 20:50:54.971 DEBUG (MainThread) [custom_components.ocpp] Reconnect websocket to 1000
2024-04-03 20:50:54.971 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to '1000'
2024-04-03 20:50:54.971 DEBUG (MainThread) [websockets.server] = connection is CLOSING
2024-04-03 20:50:54.971 DEBUG (MainThread) [websockets.server] > CLOSE 1000 (OK) [2 bytes]
2024-04-03 20:50:55.003 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '1000': ping=2.0 ms, pong=10000 ms
2024-04-03 20:50:59.954 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-04-03 20:50:59.955 DEBUG (MainThread) [websockets.server] ! failing connection with code 1006
2024-04-03 20:50:59.960 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2024-04-03 20:50:59.962 DEBUG (MainThread) [custom_components.ocpp] Closing websocket to '1000'
2024-04-03 20:50:59.962 DEBUG (MainThread) [websockets.server] = connection is CLOSING
2024-04-03 20:50:59.962 DEBUG (MainThread) [websockets.server] > CLOSE 1000 (OK) [2 bytes]
2024-04-03 20:50:59.964 DEBUG (MainThread) [websockets.server] < CLOSE 1000 (OK) [2 bytes]
2024-04-03 20:50:59.965 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2024-04-03 20:50:59.965 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-04-03 20:50:59.966 INFO (MainThread) [websockets.server] connection closed
2024-04-03 20:50:59.967 INFO (MainThread) [websockets.server] connection closed
2024-04-03 20:51:02.970 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-04-03 20:51:02.974 DEBUG (MainThread) [websockets.server] < GET /1000 HTTP/1.1
2024-04-03 20:51:02.974 DEBUG (MainThread) [websockets.server] < Pragma: no-cache
2024-04-03 20:51:02.974 DEBUG (MainThread) [websockets.server] < Cache-Control: no-cache
2024-04-03 20:51:02.974 DEBUG (MainThread) [websockets.server] < Host: 192.168.100.7
2024-04-03 20:51:02.974 DEBUG (MainThread) [websockets.server] < Origin: http://192.168.100.7
2024-04-03 20:51:02.974 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2024-04-03 20:51:02.975 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2024-04-03 20:51:02.975 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: vknSLfSiwHA7kprZQ9PJfQ==
2024-04-03 20:51:02.975 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2024-04-03 20:51:02.975 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2024-04-03 20:51:02.975 DEBUG (MainThread) [websockets.server] < authorization: Basic MTAwMDoxMjM=
2024-04-03 20:51:02.975 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2024-04-03 20:51:02.976 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2024-04-03 20:51:02.976 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2024-04-03 20:51:02.976 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: +xb5OuORhbouTTmtyFEELJFcHUs=
2024-04-03 20:51:02.976 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
2024-04-03 20:51:02.976 DEBUG (MainThread) [websockets.server] > Date: Wed, 03 Apr 2024 10:21:02 GMT
[... cut ...]
2024-04-03 20:51:33.371 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Voltage, phase_info: {'unit': 'V', 'L1': 240.5} value: 240.500000 unit :V
2024-04-03 20:51:33.372 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Import, phase_info: {'unit': 'A', 'L1': 0.01} value: 0.010000 unit :A
2024-04-03 20:51:33.372 INFO (MainThread) [ocpp] 1000: send [3,"swa1aGFX6Rv5ums7akRE8nLe4VX5xwQ24DFp",{}]
2024-04-03 20:51:33.372 DEBUG (MainThread) [websockets.server] > TEXT '[3,"swa1aGFX6Rv5ums7akRE8nLe4VX5xwQ24DFp",{}]' [45 bytes]
```

drc38, 2024-04-03:
Looks like the charger is not responding to the TriggerMessage you can try commenting it out on line 561 of `api.py`

callifo, 2024-04-03:
Yes that seems to fix the boot issue. Can that be included as a feature request, otherwise I'll just make that change when new updates come out?

drc38, 2024-04-03:
As it is a charger specific issue (and should be fixed in the firmware) it's not something that would get merged.

callifo, 2024-04-03:
Ok

Last night I tried emailing the charger's OEM to get support, mentioning that MG (whom rebadge the charger) don't respond to requests for support. Co-incidentally this morning I get an email from MG saying they had missed the email (I had sent multiple...), and they plan some updates to the charger software mid year. Hopefully this might offer some improvements.

closed 2024-04-14 by callifo