lbbrhzn/ocpp #1099: Charger continually reboots

carlosandtheo, 2024-02-29:
I'm using the latest V0.5.1 OCPP integration. 
The charger reboots approximately every 60 seconds. This wasnt occuring in earlier version.
The log file with debugging enabled is shown below.
Any suggestions  whether i have a setup issue would be appreciated.

websockets.exceptions.InvalidMessage: did not receive a valid HTTP request
2024-03-01 09:59:57.989 DEBUG (MainThread) [websockets.server] > HTTP/1.1 400 Bad Request
2024-03-01 09:59:57.989 DEBUG (MainThread) [websockets.server] > Date: Thu, 29 Feb 2024 22:59:57 GMT
2024-03-01 09:59:57.989 DEBUG (MainThread) [websockets.server] > Server: Python/3.12 websockets/11.0.3
2024-03-01 09:59:57.989 DEBUG (MainThread) [websockets.server] > Content-Length: 77
2024-03-01 09:59:57.990 DEBUG (MainThread) [websockets.server] > Content-Type: text/plain
2024-03-01 09:59:57.990 DEBUG (MainThread) [websockets.server] > Connection: close
2024-03-01 09:59:57.990 DEBUG (MainThread) [websockets.server] > [body] (77 bytes)
2024-03-01 09:59:57.991 INFO (MainThread) [websockets.server] connection failed (400 Bad Request)
2024-03-01 09:59:57.991 DEBUG (MainThread) [websockets.server] x closing TCP connection
2024-03-01 09:59:57.991 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-03-01 09:59:57.993 INFO (MainThread) [websockets.server] connection closed
2024-03-01 10:00:01.015 WARNING (MainThread) [homeassistant.components.media_player] Updating webostv media_player took longer than the scheduled update interval 0:00:10
2024-03-01 10:00:04.066 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-03-01 10:00:05.046 DEBUG (MainThread) [websockets.server] < GET / HTTP/1.1
2024-03-01 10:00:05.047 DEBUG (MainThread) [websockets.server] < Host: 0:9000
2024-03-01 10:00:05.047 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2024-03-01 10:00:05.047 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2024-03-01 10:00:05.048 DEBUG (MainThread) [websockets.server] < Origin: ws://0
2024-03-01 10:00:05.048 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: dGhlIHNhbXBsZSBub25jZQ==
2024-03-01 10:00:05.048 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2024-03-01 10:00:05.048 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2024-03-01 10:00:05.049 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2024-03-01 10:00:05.050 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2024-03-01 10:00:05.050 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2024-03-01 10:00:05.050 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: s3pPLMBiTxaQ9kYGzzhZRbK+xOo=
2024-03-01 10:00:05.050 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
2024-03-01 10:00:05.051 DEBUG (MainThread) [websockets.server] > Date: Thu, 29 Feb 2024 23:00:05 GMT
2024-03-01 10:00:05.051 DEBUG (MainThread) [websockets.server] > Server: Python/3.12 websockets/11.0.3
2024-03-01 10:00:05.053 INFO (MainThread) [websockets.server] connection open
2024-03-01 10:00:05.053 DEBUG (MainThread) [websockets.server] = connection is OPEN
2024-03-01 10:00:05.685 DEBUG (MainThread) [websockets.server] < TEXT '[2,"108361","BootNotification",{"chargePointVen...meterType":"AC meter"}]' [284 bytes]
2024-03-01 10:00:05.687 INFO (MainThread) [ocpp] : receive message [2,"108361","BootNotification",{"chargePointVendor":"SINO","chargePointModel":"PEVC2107E","chargeBoxSerialNumber":"123050003018","chargePointSerialNumber":"123050003018","firmwareVersion":"V2.05","iccid":"NULL","imsi":"NULL","meterSerialNumber":"1232132132sn","meterType":"AC meter"}]
2024-03-01 10:00:05.692 INFO (MainThread) [ocpp] : send [3,"108361",{"currentTime":"2024-02-29T23:00:05Z","interval":3600,"status":"Accepted"}]
2024-03-01 10:00:05.692 DEBUG (MainThread) [websockets.server] > TEXT '[3,"108361",{"currentTime":"2024-02-29T23:00:05...0,"status":"Accepted"}]' [87 bytes]
2024-03-01 10:00:06.580 DEBUG (MainThread) [websockets.server] < TEXT '[2,"108362","StatusNotification",{"connectorId"...2024-02-29T23:00:05Z"}]' [131 bytes]
2024-03-01 10:00:06.581 INFO (MainThread) [ocpp] : receive message [2,"108362","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEV","timestamp":"2024-02-29T23:00:05Z"}]
2024-03-01 10:00:06.583 INFO (MainThread) [ocpp] : send [3,"108362",{}]
2024-03-01 10:00:06.583 DEBUG (MainThread) [websockets.server] > TEXT '[3,"108362",{}]' [15 bytes]
2024-03-01 10:00:06.887 DEBUG (MainThread) [websockets.server] < TEXT '[2,"108363","StatusNotification",{"connectorId"...2024-02-29T23:00:05Z"}]' [129 bytes]
2024-03-01 10:00:06.890 INFO (MainThread) [ocpp] : receive message [2,"108363","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-02-29T23:00:05Z"}]
2024-03-01 10:00:06.892 INFO (MainThread) [ocpp] : send [3,"108363",{}]
2024-03-01 10:00:06.892 DEBUG (MainThread) [websockets.server] > TEXT '[3,"108363",{}]' [15 bytes]
2024-03-01 10:00:07.057 INFO (MainThread) [ocpp] : send [2,"2e485df4-6744-4610-b1b2-11f511ec609f","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-03-01 10:00:07.058 DEBUG (MainThread) [websockets.server] > TEXT '[2,"2e485df4-6744-4610-b1b2-11f511ec609f","GetC...rtedFeatureProfiles"]}]' [98 bytes]
2024-03-01 10:00:07.141 DEBUG (MainThread) [websockets.server] < TEXT '[2,"108364","Heartbeat",{}]' [27 bytes]
2024-03-01 10:00:07.144 INFO (MainThread) [ocpp] : receive message [2,"108364","Heartbeat",{}]
2024-03-01 10:00:07.147 INFO (MainThread) [ocpp] : send [3,"108364",{"currentTime":"2024-02-29T23:00:07Z"}]
2024-03-01 10:00:07.147 DEBUG (MainThread) [websockets.server] > TEXT '[3,"108364",{"currentTime":"2024-02-29T23:00:07Z"}]' [51 bytes]
2024-03-01 10:00:07.808 DEBUG (MainThread) [websockets.server] < TEXT '[3,"2e485df4-6744-4610-b1b2-11f511ec609f",{"con...ging,RemoteTrigger"}]}]' [214 bytes]
2024-03-01 10:00:07.809 INFO (MainThread) [ocpp] : receive message [3,"2e485df4-6744-4610-b1b2-11f511ec609f",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,Reservation,LocalAuthListManagement,SmartCharging,RemoteTrigger"}]}]
2024-03-01 10:00:07.813 INFO (MainThread) [ocpp] : send [2,"a15a9a6f-d8b3-4ca0-9cf7-1058d5d5f7c8","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-03-01 10:00:07.813 DEBUG (MainThread) [websockets.server] > TEXT '[2,"a15a9a6f-d8b3-4ca0-9cf7-1058d5d5f7c8","GetC...rtedFeatureProfiles"]}]' [98 bytes]
2024-03-01 10:00:08.423 DEBUG (MainThread) [websockets.server] < TEXT '[3,"a15a9a6f-d8b3-4ca0-9cf7-1058d5d5f7c8",{"con...ging,RemoteTrigger"}]}]' [214 bytes]
2024-03-01 10:00:08.423 INFO (MainThread) [ocpp] : receive message [3,"a15a9a6f-d8b3-4ca0-9cf7-1058d5d5f7c8",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,Reservation,LocalAuthListManagement,SmartCharging,RemoteTrigger"}]}]
2024-03-01 10:00:08.425 INFO (MainThread) [ocpp] : send [2,"4ae3d9e3-ecac-45b1-97aa-e5c6efbc805f","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-03-01 10:00:08.425 DEBUG (MainThread) [websockets.server] > TEXT '[2,"4ae3d9e3-ecac-45b1-97aa-e5c6efbc805f","GetC..."NumberOfConnectors"]}]' [92 bytes]
2024-03-01 10:00:09.036 DEBUG (MainThread) [websockets.server] < TEXT '[3,"4ae3d9e3-ecac-45b1-97aa-e5c6efbc805f",{"con...y":true,"value":"1"}]}]' [122 bytes]
2024-03-01 10:00:09.037 INFO (MainThread) [ocpp] : receive message [3,"4ae3d9e3-ecac-45b1-97aa-e5c6efbc805f",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2024-03-01 10:00:09.040 INFO (MainThread) [ocpp] : send [2,"8cd5761f-451e-4512-969f-da0dcef3a43b","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-03-01 10:00:09.040 DEBUG (MainThread) [websockets.server] > TEXT '[2,"8cd5761f-451e-4512-969f-da0dcef3a43b","GetC..."NumberOfConnectors"]}]' [92 bytes]
2024-03-01 10:00:09.650 DEBUG (MainThread) [websockets.server] < TEXT '[3,"8cd5761f-451e-4512-969f-da0dcef3a43b",{"con...y":true,"value":"1"}]}]' [122 bytes]
2024-03-01 10:00:09.651 INFO (MainThread) [ocpp] : receive message [3,"8cd5761f-451e-4512-969f-da0dcef3a43b",{"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2024-03-01 10:00:09.654 INFO (MainThread) [ocpp] : send [2,"82dfdd79-8475-4671-9253-262dfe8c290b","GetConfiguration",{"key":["HeartbeatInterval"]}]
[... cut ...]
2024-03-01 10:00:47.630 DEBUG (MainThread) [websockets.server] ! failing connection with code 1002
2024-03-01 10:00:47.633 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2024-03-01 10:00:47.642 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-03-01 10:00:47.643 INFO (MainThread) [websockets.server] connection closed
2024-03-01 10:00:47.735 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-03-01 10:00:48.639 DEBUG (MainThread) [websockets.server] ! invalid handshake

timderspieler, 2024-03-14:
@carlosandtheo / @drc38 any news on this? The issue is really annoying :D

drc38, 2024-03-14:
Have you tried the applying the changes in #1101 to confirm it fixes the issue?

carlosandtheo, 2024-03-14:
Yes the changes in #1101 fixed my installation thanks. I also needed to remove ChangeConfiguration because it isnt accepted by my charger and causes it to reboot.

PR #1101 (2024-03-23): Extra logic if measurand not accepted
Potentially fixes #1099 #1096
closed 2024-03-14 by carlosandtheo