lbbrhzn/ocpp #1117: EVEC VEC01 support

b3uk, 2024-03-22:
I've successfully managed to connect my charger, an EVEC VEC01 (which is using OCPP 1.6J), using v0.5.1 of the custom component, and various diagnostic values are shown in the entities, however if I toggle the charge control, I get the error

'list index out of range'

The charger is connected via ethernet (it doesn't support OCPP over Wi-Fi).

The log details are as follows:

```yaml
Logger: custom_components.ocpp
Source: custom_components/ocpp/api.py:961
integration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [issues](https://github.com/lbbrhzn/ocpp/issues))
First occurred: 22:53:14 (630 occurrences)
Last logged: 23:25:23

Unexpected exception in connection to '': 'list index out of range'
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 961, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 432, in post_connect
    resp = await self.get_configuration(ckey.number_of_connectors.value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 830, in get_configuration
    value = resp.configuration_key[0][om.value.value]
            ~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

lbbrhzn, 2024-03-24:
can you add a log of the OCPP messages between charger and server?

b3uk, 2024-03-25:
Sure thing. I turned on debug logging for the websockets server see below. This is just 9 seconds, but it does this constantly. If I leave it on for very long then the log gets so big it won't render in the web browser any more.

I triggered the charge control at 19:29:44, just before the final message included here.

`2024-03-25 19:29:41.722 DEBUG (MainThread) [websockets.server] = connection is OPEN
2024-03-25 19:29:43.728 DEBUG (MainThread) [websockets.server] > TEXT '[2,"01a4b8b1-86ed-4f50-9d04-36fd63b6c213","GetC...rtedFeatureProfiles"]}]' [98 bytes]
2024-03-25 19:29:43.960 DEBUG (MainThread) [websockets.server] < TEXT '[3,"01a4b8b1-86ed-4f50-9d04-36fd63b6c213",{"con...ing, RemoteTrigger"}]}]' [206 bytes]
2024-03-25 19:29:43.964 DEBUG (MainThread) [websockets.server] > TEXT '[2,"f0d1bf8d-dee7-42dd-bcab-02af59789b46","GetC..."NumberOfConnectors"]}]' [92 bytes]
2024-03-25 19:29:44.426 DEBUG (MainThread) [websockets.server] < TEXT '[3,"f0d1bf8d-dee7-42dd-bcab-02af59789b46",{"configurationKey":[]}]' [66 bytes]
2024-03-25 19:29:44.438 DEBUG (MainThread) [websockets.server] = connection is CLOSING
2024-03-25 19:29:44.439 DEBUG (MainThread) [websockets.server] > CLOSE 1000 (OK) [2 bytes]
2024-03-25 19:29:44.617 DEBUG (MainThread) [websockets.server] ! failing connection with code 1006
2024-03-25 19:29:44.618 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-03-25 19:29:44.620 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2024-03-25 19:29:44.621 INFO (MainThread) [websockets.server] connection closed
2024-03-25 19:29:44.735 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-03-25 19:29:44.739 DEBUG (MainThread) [websockets.server] < GET / HTTP/1.1
2024-03-25 19:29:44.740 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2024-03-25 19:29:44.740 DEBUG (MainThread) [websockets.server] < Host: 10.0.1.62
2024-03-25 19:29:44.740 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2024-03-25 19:29:44.741 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2024-03-25 19:29:44.741 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2024-03-25 19:29:44.741 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: JATeGott6Se1fYM67pryfg==
2024-03-25 19:29:44.742 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2024-03-25 19:29:44.743 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2024-03-25 19:29:44.743 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2024-03-25 19:29:44.743 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: r+OeyfD+dePRJm87SwhoaeIkAVE=
2024-03-25 19:29:44.743 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
2024-03-25 19:29:44.744 DEBUG (MainThread) [websockets.server] > Date: Mon, 25 Mar 2024 19:29:44 GMT
2024-03-25 19:29:44.744 DEBUG (MainThread) [websockets.server] > Server: Python/3.12 websockets/11.0.3
2024-03-25 19:29:44.746 INFO (MainThread) [websockets.server] connection open
2024-03-25 19:29:44.746 DEBUG (MainThread) [websockets.server] = connection is OPEN
2024-03-25 19:29:46.751 DEBUG (MainThread) [websockets.server] > TEXT '[2,"e66d0719-bba6-4543-a62d-5df95d27c39a","GetC...rtedFeatureProfiles"]}]' [98 bytes]
2024-03-25 19:29:46.980 DEBUG (MainThread) [websockets.server] < TEXT '[3,"e66d0719-bba6-4543-a62d-5df95d27c39a",{"con...ing, RemoteTrigger"}]}]' [206 bytes]
2024-03-25 19:29:46.983 DEBUG (MainThread) [websockets.server] > TEXT '[2,"7c2936eb-b00a-4643-9987-0871a0548131","GetC..."NumberOfConnectors"]}]' [92 bytes]
2024-03-25 19:29:47.448 DEBUG (MainThread) [websockets.server] < TEXT '[3,"7c2936eb-b00a-4643-9987-0871a0548131",{"configurationKey":[]}]' [66 bytes]
2024-03-25 19:29:47.462 DEBUG (MainThread) [websockets.server] = connection is CLOSING
2024-03-25 19:29:47.462 DEBUG (MainThread) [websockets.server] > CLOSE 1000 (OK) [2 bytes]
2024-03-25 19:29:47.643 DEBUG (MainThread) [websockets.server] ! failing connection with code 1006
2024-03-25 19:29:47.643 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-03-25 19:29:47.646 DEBUG (MainThread) [websockets.server] x half-closing TCP connection
2024-03-25 19:29:47.647 INFO (MainThread) [websockets.server] connection closed
2024-03-25 19:29:47.760 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-03-25 19:29:47.764 DEBUG (MainThread) [websockets.server] < GET / HTTP/1.1
2024-03-25 19:29:47.765 DEBUG (MainThread) [websockets.server] < Upgrade: websocket
2024-03-25 19:29:47.765 DEBUG (MainThread) [websockets.server] < Host: 10.0.1.62
2024-03-25 19:29:47.765 DEBUG (MainThread) [websockets.server] < Connection: Upgrade
2024-03-25 19:29:47.766 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Protocol: ocpp1.6
2024-03-25 19:29:47.766 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Version: 13
2024-03-25 19:29:47.766 DEBUG (MainThread) [websockets.server] < Sec-WebSocket-Key: RHdH1FU67X4Im+NY2nUD1Q==
2024-03-25 19:29:47.767 DEBUG (MainThread) [websockets.server] > HTTP/1.1 101 Switching Protocols
2024-03-25 19:29:47.767 DEBUG (MainThread) [websockets.server] > Upgrade: websocket
2024-03-25 19:29:47.768 DEBUG (MainThread) [websockets.server] > Connection: Upgrade
2024-03-25 19:29:47.768 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Accept: 8GSjzMU4gYjuVvfBSLI/cHfxsYI=
2024-03-25 19:29:47.768 DEBUG (MainThread) [websockets.server] > Sec-WebSocket-Protocol: ocpp1.6
[... cut ...]
2024-03-25 19:29:50.316 INFO (MainThread) [ocpp] : receive message [3,"57d37a8f-69ec-4ccd-bda9-763427f4143e",{"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core, FirmwareManagement, LocalAuthListManagement, SmartCharging, RemoteTrigger"}]}]
2024-03-25 19:29:50.319 INFO (MainThread) [ocpp] : send [2,"2b49cb95-261c-46d0-971b-afaeb8c4620b","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-03-25 19:29:50.783 INFO (MainThread) [ocpp] : receive message [3,"2b49cb95-261c-46d0-971b-afaeb8c4620b",{"configurationKey":[]}]
2024-03-25 19:29:50.784 ERROR (MainThread) [custom_components.ocpp] Unexpected exception in connection to '': 'list index out of range'`

closed 2024-06-03 by github-actions[bot]

Fayrewood, 2026-09-04:
> I've successfully managed to connect my charger, an EVEC VEC01 (which is using OCPP 1.6J), using v0.5.1 of the custom component, and various diagnostic values are shown correctly in the entities. Various entities also change correctly if I turn the charger on manually, and the charge control toggle shows as being turned on. The charger is connected via ethernet (it doesn't support OCPP over Wi-Fi). If I turn the charger on manually, and off using home assistant, no error is shown.

@b3uk  Did you ever get the EVEC VEC01 working OK?