lbbrhzn/ocpp #1119: upgrade to OCPP 0.5.2 bricks installation

Krobbelinks, 2024-03-25:
After upgrading OCPP to 0.5.2 I can not load my car any more using ev-box.

No sensors more available
charger not available

There is some cycling 
 For some brief moments the sensors are available again.

drc38, 2024-03-25:
What version did you upgrade from?

drc38, 2024-03-25:
@dreimer1986 have you had any issue with the upgrade to 0.5.2?

dreimer1986, 2024-03-26:
Hmm, nope, all fine here. Protocols are empty of any warnings or errors and all sensors are available. Services were not tested though. I just use the normal buttons and sensors for all my stuff.

What do you mean with you can't charge anymore? At least in my configuration I just get sensor data and do stuff with that. The car just charges when plugged in. (I added one of the cards as default card thoiugh)

Krobbelinks, 2024-03-27:
I upgraded from 0.5.1

lcevela, 2024-03-28:
I have the same problem. Version 0.5.1 worked fine, version 0.5.2 doesn't work. I tried reinstalling the integration but it still won't detect my charger. No error messages were returned. My charger is EVCH420RE1001VC20 and supports OCPP 1.6.

dreimer1986, 2024-03-28:
Some logs would be nice. This should do it:

```
logger:
  default: info
  logs:
    pyfritzhome: critical
    custom_components.ocpp: debug
    websocket: debug
```

lcevela, 2024-03-28:
I tried adding it to my docker-compose.yml but it returns errors. Here is the whole file with the logger part commented out:

version: '3.1'
services:
    homeassistant:
        image: homeassistant/home-assistant
        container_name: hass
        # network_mode: host
        environment:
            - PUID=1000
            - PGID=1000
            - TZ=Europe/Prague
        volumes:
            - /var/docker/hass/data:/config
        # privileged: true
        devices:
            - /dev/ttyUSB0:/dev/ttyUSB0
        ports:
            - 8123:8123
        environment:
            - APACHE_PORT=443
            - DISABLE_JEMALLOC=1
        restart: unless-stopped
       # logger:
       #    default: info
       #    logs:
       #     custom_components.ocpp: debug
       #     websocket: debug
    nodered:
         image: nodered/node-red:latest
         container_name: nodered
         # network_mode: host
         environment:
            - TZ=Europe/Prague
         restart: unless-stopped
         ports:
            - "1880:1880"
         volumes:
            - ./nodered:/data

lcevela, 2024-03-28:
I try to improve formating:
`version: '3.1'
services:
    homeassistant:
        image: homeassistant/home-assistant
        container_name: hass
        # network_mode: host
        environment:
            - PUID=1000
            - PGID=1000
            - TZ=Europe/Prague
        volumes:
            - /var/docker/hass/data:/config
        # privileged: true
        devices:
            - /dev/ttyUSB0:/dev/ttyUSB0
        ports:
            - 8123:8123
        environment:
            - APACHE_PORT=443
            - DISABLE_JEMALLOC=1
        restart: unless-stopped
       # logger:
       #    default: info
       #    logs:
       #     custom_components.ocpp: debug
       #     websocket: debug
    nodered:
         image: nodered/node-red:latest
         container_name: nodered
         # network_mode: host
         environment:
            - TZ=Europe/Prague
         restart: unless-stopped
         ports:
            - "1880:1880"
         volumes:
            - ./nodered:/data
`

dreimer1986, 2024-03-28:
Who said anything about modding the Docker container? These lines are for HA's configuration.yaml

lcevela, 2024-03-28:
I already found my mistake. Log entries contains following parts with 'ocpp' string:

2024-03-28 17:36:38.493 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

2024-03-28 17:36:40.243 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'tts', 'motioneye', 'input_select', 'hacs', 'map', 'file_upload', 'stt', 'homeassistant_alerts', 'mobile_app', 'zone', 'trace', 'goodwe', 'default_config', 'wake_word', 'input_boolean', 'ocpp', 'tag', 'hardware', 'blueprint', 'input_datetime', 'assist_pipeline', 'input_text', 'kodi', 'onboarding', 'logbook', 'application_credentials', 'input_number', 'conversation', 'diagnostics', 'script', 'repairs', 'ffmpeg', 'ipp', 'skodaconnect', 'unifi', 'persistent_notification', 'backup', 'media_source', 'homeassistant_sky_connect', 'zha', 'schedule', 'search', 'rpi_power', 'energy', 'lovelace', 'device_automation', 'timer', 'input_button', 'analytics', 'automation', 'stream', 'config', 'system_health', 'homeassistant_hardware', 'person', 'my', 'history', 'auth', 'image_upload', 'scene', 'counter', 'sun'}

2024-03-28 17:36:41.211 INFO (MainThread) [homeassistant.setup] Setting up ocpp
2024-03-28 17:36:41.211 INFO (MainThread) [custom_components.ocpp] config = {}
2024-03-28 17:36:42.047 INFO (MainThread) [homeassistant.components.sensor] Setting up ocpp.sensor

2024-03-28 17:36:42.047 INFO (MainThread) [homeassistant.components.sensor] Setting up ocpp.sensor
2024-03-28 17:36:42.061 INFO (SyncWorker_7) [homeassistant.loader] Loaded switch_as_x from homeassistant.components.switch_as_x
2024-03-28 17:36:42.061 INFO (MainThread) [homeassistant.setup] Setting up number
2024-03-28 17:36:42.061 INFO (MainThread) [homeassistant.setup] Setup of domain number took 0.0 seconds
2024-03-28 17:36:42.062 INFO (MainThread) [homeassistant.setup] Setting up button
2024-03-28 17:36:42.063 INFO (MainThread) [homeassistant.setup] Setup of domain button took 0.0 seconds
2024-03-28 17:36:42.064 INFO (MainThread) [homeassistant.setup] Setting up assist_pipeline
2024-03-28 17:36:42.074 INFO (MainThread) [homeassistant.components.number] Setting up ocpp.number
2024-03-28 17:36:42.076 INFO (MainThread) [homeassistant.components.button] Setting up ocpp.button
2024-03-28 17:36:42.088 INFO (MainThread) [homeassistant.setup] Setting up switch
2024-03-28 17:36:42.088 INFO (MainThread) [homeassistant.setup] Setup of domain switch took 0.0 seconds
2024-03-28 17:36:42.093 INFO (MainThread) [homeassistant.setup] Setup of domain input_boolean took 1.1 seconds
2024-03-28 17:36:42.094 INFO (MainThread) [homeassistant.setup] Setup of domain input_text took 0.9 seconds
2024-03-28 17:36:42.095 INFO (MainThread) [homeassistant.setup] Setup of domain counter took 0.8 seconds
2024-03-28 17:36:42.096 INFO (MainThread) [homeassistant.setup] Setup of domain application_credentials took 0.8 seconds
2024-03-28 17:36:42.099 INFO (MainThread) [homeassistant.components.switch] Setting up ocpp.switch

I tried to reimplement OCPP integration. Charger is detected and connected, but all sensors returns 'N/A' and I cant switch off Charger availability or Charge Control and I can't change Maximum current. I hope, it helps.

drc38, 2024-03-28:
@icevela the logs are not showing your charger trying to connect to HA. See https://github.com/lbbrhzn/ocpp/issues/1117#issuecomment-2018767615 for example

lcevela, 2024-03-28:
I found only this one:
2024-03-28 18:33:29.089 INFO (MainThread) [websockets.server] server listening on 0.0.0.0:9000

closed 2024-05-17 by lbbrhzn