lbbrhzn/ocpp #1124: IOCharger IOCH10-22T measureands problem

stefan73, 2024-03-29:
I have a IOCharger IOCH10-22T. These are sold in Germany as Entratek Power Dot Fix. This OCPP implementation basically works and I am now using 0.52. I can set the charge current for 3 phase charging from 6 to max and start/stop charging.
The problem are the measureands, which seem to get lost more and more with each update of this implementation. E.g. with 0.51 Current import, export and offered were available. With 0.52 those are now gone too. Now with 0.52 thre is only Energy Meter Start, Energy Session, Time Session, Transaction Id, Voltage left.
While charging only Voltage updates every 60s. The other value update every 900s. Restarting home assistant changes the measureands that I condigured in the charger web-interface. If I configure MeterValuesSampledData to report Current.Import,Energy.Active.Import.Register,Voltage,Current.Offered , this entry switches back to voltage only after the HA restart. The values for MeterValuesAlignedData remain. So what I see in the charger WebIF seems consistent with the observed reporting intervals.
If I change the measureands in the charger WebIF, the values update faster. 
I fail to make measureands persistent in the HA/Charger combination. And I fail to win back the values that got reported in prior versions.
Questions: Charger Firmware bug? - Can I somehow mitigate the situation?
Goal: Up to date (60s) values at least for Current Import and Energy Meter.

Here is a log after restarting HA:
<details>
2024-03-29 18:14:22.730 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration pyscript which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-03-29 18:14:22.732 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-03-29 18:14:22.733 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-03-29 18:14:22.734 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration solcast_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-03-29 18:14:32.026 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-03-29 18:14:32.541 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solcast_pv_forecast_prognose_heute (<class 'custom_components.solcast_solar.sensor.SolcastSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/oziee/ha-solcast-solar/issues
2024-03-29 18:14:34.800 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({% set voltage = states('sensor.wallbox_voltage') | float %} {% set current = states('sensor.wallbox_current_import') | float %}
{{ (voltage * current) | round(1, default=0) }}) renders=2>
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2037, in forgiving_float_filter
return float(value)
^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 573, in async_render
render_result = _render_with_context(self.template, compiled, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2364, in _render_with_context
return template.render(**kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/jinja2/environment.py", line 1301, in render
self.environment.handle_exception()
File "/usr/local/lib/python3.12/site-packages/jinja2/environment.py", line 936, in handle_exception
raise rewrite_traceback_stack(source=source)
File "<template>", line 1, in top-level template code
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2040, in forgiving_float_filter
raise_no_default("float", value)
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1668, in raise_no_default
raise ValueError(
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% set voltage = states('sensor.wallbox_voltage') | float %} {% set current = states('sensor.wallbox_current_import') | float %}
{{ (voltage * current) | round(1, default=0) }}' but no default was specified
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 699, in async_render_to_info
render_info._result = self.async_render(
^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 575, in async_render
raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% set voltage = states('sensor.wallbox_voltage') | float %} {% set current = states('sensor.wallbox_current_import') | float %}
{{ (voltage * current) | round(1, default=0) }}' but no default was specified
2024-03-29 18:14:34.810 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% set voltage = states('sensor.wallbox_voltage') | float %} {% set current = states('sensor.wallbox_current_import') | float %}
{{ (voltage * current) | round(1, default=0) }}' but no default was specified') while processing template 'Template<template=({% set voltage = states('sensor.wallbox_voltage') | float %} {% set current = states('sensor.wallbox_current_import') | float %}
{{ (voltage * current) | round(1, default=0) }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.wallbox_power'
2024-03-29 18:14:34.933 INFO (MainThread) [custom_components.pyscript.file.pv_excess_control.pv_excess_control] [switch.wallbox_charge_control (Prio 2)] Trigger Method started.
2024-03-29 18:14:34.934 INFO (MainThread) [custom_components.pyscript.file.pv_excess_control.pv_excess_control] [switch.wallbox_charge_control (Prio 2)] Registered appliance.
2024-03-29 18:14:34.934 INFO (MainThread) [custom_components.pyscript.file.pv_excess_control.pv_excess_control] Dict: {'automation.pv_ueberschuss_optimierer': {'instance': <custom_components.pyscript.eval.PvExcessControl object at 0x7fd5a898f080>, 'priority': 2}}
2024-03-29 18:14:34.935 WARNING (MainThread) [homeassistant.components.automation.pv_ueberschuss_optimierer] PV Ueberschuss Optimierer: Already running
2024-03-29 18:14:34.964 INFO (MainThread) [ocpp] 38CC7A28A780440BA6965171059494EF: receive message [2,"17117324741608752","Heartbeat",{}]
2024-03-29 18:14:34.977 INFO (MainThread) [ocpp] 38CC7A28A780440BA6965171059494EF: send [3,"17117324741608752",{"currentTime":"2024-03-29T17:14:34Z"}]
2024-03-29 18:14:35.368 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.wallbox_energy_reactive_import_interval not found.
2024-03-29 18:14:35.369 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.wallbox_energy_reactive_import_register not found.
2024-03-29 18:14:35.369 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.wallbox_power_reactive_export not found.
[... cut ...]
2024-03-29 18:14:38.717 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.wallbox_soc not found.
2024-03-29 18:14:38.718 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.wallbox_temperature not found.
2024-03-29 18:14:38.718 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.wallbox_energy_active_import_register not found.
2024-03-29 18:14:38.718 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.wallbox_power_factor not found.
</details>

stefan73, 2024-03-30:
Reinstalling does unfortunately not help.
The problem is not the reporting interval and how it is set up. The 60 seconds reporting interval during charging works - but only for voltage.

For clarification:
The charger supports the following values: Current.Import, Energy.Active.Import.Register ,Voltage, Current.Offered, Temperature, SoC. I can set those values for MeterValuesSampledData or MeterValuesAlignedData. 
Problem #1 When setting up values as MeterValuesSampledData in the charger firmware, the reporting inverval for the selected values increases to 60 seconds. BUT upon restarting HA, the MeterValuesSampledData setting in the charger is reset to voltage only.
Problem #2: With 0.52 of this code, the integration reports Current.Import, Current.Offered, Temperature and Energy.Active.Import.Register as not available. Those values were available with 0.51.

So the problems are: Loss of sensor / measureands with 0.52 and reset of MeterValuesSampledData entries with HA starting.

drc38, 2024-03-30:
See https://github.com/lbbrhzn/ocpp/issues/1096#issue-2159922663

The log should test each measurand for support by the charger.

drc38, 2024-03-30:
If you can no longer select measurands on setup try replacing line 464 in `api.py` with:
`all_measurands = MEASURANDS`

You will need to add  `MEASURANDS` to the import line as well.

closed 2024-06-07 by github-actions[bot]

stefan73, 2024-07-07:
Solution https://github.com/lbbrhzn/ocpp/issues/1150