lbbrhzn/ocpp #1126: Status intermittent between not available and available

StefanoGiu, 2024-04-09:
**Describe the bug**
Status intermittent between not available and available

**To Reproduce**
Steps to reproduce the behavior:
check the status of the integration

**Expected behavior**
Status should be stable and not become available every minute

**Desktop (please complete the following information):**
 - Version [e.g. 22] 0.5.2

StefanoGiu, 2024-04-30:
Any update? This custom component became unusable...

StefanoGiu, 2024-05-17:
Still happening.... any willingness or plan to fix this?

Aure51800, 2024-05-19:
Hi there, same here.
I can't find how to reproduce it. 
I know the last working version was 0.5.1
Entities become unavailable every 5s
What should I give you to debug ?

Aure51800, 2024-05-20:
Hi, after some testes, it happens every 40s, with errors:
Enregistreur: custom_components.ocpp
Source: custom_components/ocpp/api.py:223
intégration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [problèmes](https://github.com/lbbrhzn/ocpp/issues))
S'est produit pour la première fois: 21:46:38 (12 occurrences)
Dernier enregistrement: 21:56:33

Skipping websocket subprotocol validation

And 

Enregistreur: custom_components.ocpp
Source: custom_components/ocpp/api.py:1020
intégration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [problèmes](https://github.com/lbbrhzn/ocpp/issues))
S'est produit pour la première fois: 21:46:54 (10 occurrences)
Dernier enregistrement: 21:56:03

Unexpected exception in connection to 'TACW2244421S5041': 'list index out of range'
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 1020, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 489, in post_connect
    resp = await self.get_configuration(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 889, in get_configuration
    value = resp.configuration_key[0][om.value.value]
            ~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range

Aure51800, 2024-05-20:
And some more debugging logs

2024-05-20 22:49:45.180 INFO (MainThread) [ocpp] TACW2244421S5041: send [2,"03ab8853-0018-4024-bb5e-eec544199c7d","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
2024-05-20 22:49:45.679 INFO (MainThread) [ocpp] TACW2244421S5041: receive message [3, "03ab8853-0018-4024-bb5e-eec544199c7d", {"status": "NotSupported"}]
2024-05-20 22:49:45.679 DEBUG (MainThread) [custom_components.ocpp] 'TACW2244421S5041' measurands not configurable by OCPP
2024-05-20 22:49:45.680 INFO (MainThread) [ocpp] TACW2244421S5041: send [2,"417df093-3b29-4365-b06e-e859668851ba","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2024-05-20 22:49:46.179 INFO (MainThread) [ocpp] TACW2244421S5041: receive message [3, "417df093-3b29-4365-b06e-e859668851ba", {"configurationKey": [], "unknownKey": ["MeterValuesSampledData"]}]
2024-05-20 22:49:46.179 ERROR (MainThread) [custom_components.ocpp] Unexpected exception in connection to 'TACW2244421S5041': 'list index out of range'

Aure51800, 2024-05-22:
No news ?

ppnacho, 2024-05-30:
Same here. Some workaround? thanks

SAUhrich, 2024-06-01:
No new information here, just reporting the same issue. I have had a MaxpeedingRods charger connected without problems for about 7 months up through using 0,5,1, After upgrading to 0.5.2 it connects to the charger correctly for about 30 seconds, and then disconnects for about 60 seconds, Interestingly, when it is connected, I can read and control the charger,

I have tried each successive version after 0,5,1 with the same results, and cannot now go back ti 0,5,1, I tried manually copying the files from 0.5.1 to the ocpp folder, but, same results. I will note that when I updated ocpp, it happened at the same time that an HA core update occurred, so it's possible that some combination of the HA update caused the issues?

I have deleted my setup entities (central and charger) to stop getting the hundreds of error messages per day, but left the integration loaded to watch for a new update to ocpp. It wouldn't be right to complain about someone's work that they graciously shared for free, I'm just hopeful that a future update will resolve the issue. Like others, I have setup automations that control my charger (I'm 100% solar) and became dependent on the integration,

Aure51800, 2024-07-18:
Hi guys,
Good news, I have ABB TACW 22 with this problem, as there were no news here, I tried to update my wallbox, delivered with software 1.5.2 to 1.8.21 (3 major update to do). Now there is no more problem with loosing connection.
You can proceed do the upgrade with Charge Sync APP or TerraConfig ( both Need to register but free).
Please don't forget to screenshot every configuration pages, as I lost my config (modbus etc) from 1.6.6 to 1.7.1

enjoy!

StefanoGiu, 2024-07-19:
I upgraded to 1.8.21 but the problem persists. I have also checked that the wallbox is not accessible via Charger Sync App Legacy or TerraConfig in WiFi, but only in bluetooth, even if the wifi is configured properly and the router is nearby. The wallbox is still using wifi somehow because the OCPP server is updated (even if intermittently). Can you please suggest?

Aure51800, 2024-07-19:
Hi,
This is normal that you can only access your wall box on chargersync or terraconfig with Bluetooth.
When you choose your customized server, please stay in http://homeassistantadress:port
Give the same port on occp and disable configuration check on occp

StefanoGiu, 2024-07-19:
Actually my OCPP server is accessible from the url ws://IP:9000

After the upgrade of the firmware I made a reset of the wallbox, configured again first the Wifi and then the OCPP server and now the connection seems to be stable again.

PR #1292 (2024-09-09): Allow to disable autodetection of measurands
This PR is to address issues induced by ABB Terra AC firmware bug:
- #1275 
- probably #1263 
- probably #1126

As such bugs could be in other chargers, the solution is generic:

- Add config option to detect measurands automatically, set default to true

- Restrict checked measurands to the ones provided by users

I still need to add a quirk to make ABB chargers working again and update docs. So, it is WIP at this stage.

Todo list:

- [x] Notify ABB regarding the issue
- [x] Add a configuration option to disable automatic detection of the measuran [... cut]
closed 2024-09-09 by lbbrhzn