lbbrhzn/ocpp #1136: Confirm Wallbox Copper SB

Dixet, 2024-04-19:
Since Wallbox Pulsar is in the documentation and Wallbox Copper SB supports OCPP v1.6j I tried to switch to OCPP. 

I can conform Wallbox Copper SB to work perfectly with below sensors and controls. Some small differences with Wallbox Pulsar.

All sensors visible in attached screenshot, some confirmed working:

**Metrics**
Energy Active Import Register 
Energy Session (gives a different reading than Energy Active Import Register)
Power Active Import (sensor is available but always show 0)

Current Offered (maximum charging current available)
Time Session (elapsed time from start of charging session)

Current Import (instantaneous charging current - I use this instead of Power Active Import)

**Diagnostics**
Status Connector (shows the current state of available/preparing/charging/finishing/suspended etc)
Stop Reason (reason the charging session was stopped)
Charger Status (current status of the charger)

**Controls**
Charge Control
Availability
Maximum Current (sets maximum charging current available)
Reset

[image]

closed 2024-05-17 by lbbrhzn

brusuillis, 2024-11-19:
@Dixet , would you mind sharing your (purged) evcc.yaml and a screenshot from the app? I'm stuck on that...

Matssa56, 2025-02-20:
Hi! I tried to add OCPP under Home Assistant and use my Wallbox Copper SB and my measurands doesn't work. Indeed I can't pause the charging or get values.
Do you need to activate  “Improved charger control” (Profile -> Experimental functions) (myWallbox app)? What firmware do you use? On my end I'm on 6.6.20

Dixet, 2025-03-03:
@Matssa56 no other configuration in the app is necessarily. Only enabling OCPP as per the documentation with the correct home assistant address.