lbbrhzn/ocpp #1141: Question - Error messages

zbuh, 2024-04-23:
This is a question about error messages that are sent to persistent notifications.

There are situations in that we expect an error to occur. Like turn off charging, when margin is not active.
On personal automations, is easy to mitigate, but with other integrations or blueprints, it's not.
Would be nice to have the option to disable these notifications.

zbuh, 2024-04-23:
I saw the that workaround. It would be nice to have the option to enable/disable persistent notifications (keep the log only)
But I understand that this is a minor case.

Thanks

closed 2024-04-23 by zbuh

drc38, 2024-04-23:
The automation approach gives more flexibility if you only want to receive some notifications rather than simply turning them all off. And  each charger determines what is sent as an event. I use it to suppress SystemTime events and keep the rest for example.