lbbrhzn/ocpp #1148: Entratek Dot Pro 2 Error in Communication

LukasNitsche, 2024-04-29:
## Version of the custom_component

-->v0.5.2

## Describe the bug

I have the Entratek Dot Pro 2 connected to the OCPP Server. (DUOSIDA Mode3@11KW) Setting of the maximum current is not working. It only returns the error described below and does not change the charge rate. There are also other error messages, but unfortunately I have dismissed them and do not know how to access them again. If you can advise please do. 

I know the charger supports the setting of the rate, since I can do it through platforms like Clever PV. However, they might have implemented a special fix to make this charger work.

My assumption is that the charger does not comply 100 % with the OCPP instructions. However I do not have the technical knowledge on how to view the communication and debug it. If someone give advise, this would be highly appreciated, since I would love to use all the functionalities of the charger.
## Debug log

```text

Logger: custom_components.ocpp
Quelle: custom_components/ocpp/api.py:903
Integration: Open Charge Point Protocol (OCPP) (Dokumentation, Probleme)
Erstmals aufgetreten: 13:24:36 (17 Vorkommnisse)
Zuletzt protokolliert: 14:12:25

Get Configuration returned unknown key for: ChargingScheduleAllowedChargingRateUnit

Logger: homeassistant.components.websocket_api.http.connection
Quelle: components/websocket_api/commands.py:239
Integration: Home Assistant WebSocket API (Dokumentation, Probleme)
Erstmals aufgetreten: 13:24:36 (17 Vorkommnisse)
Zuletzt protokolliert: 14:12:25

[544321530816] argument of type 'NoneType' is not iterable
[543918468672] argument of type 'NoneType' is not iterable
[543918053952] argument of type 'NoneType' is not iterable
[543918686784] argument of type 'NoneType' is not iterable
[543918696384] argument of type 'NoneType' is not iterable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2543, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2580, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 971, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1043, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 112, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/ocpp/number.py", line 129, in async_set_native_value
    resp = await self.central_system.set_max_charge_rate_amps(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 298, in set_max_charge_rate_amps
    return await self.charge_points[cp_id].set_charge_rate(limit_amps=value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 678, in set_charge_rate
    if om.current.value in resp:
       ^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable

```

drc38, 2024-04-30:
You can try commenting out line 670 in `api.py`  and replacing it with `resp=om.current.value` but you may continue to run into issues. Best to raise the issue with the manufacturer.

LukasNitsche, 2024-04-30:
I will try this and I am already in communication with the (German) manufacturer / vendor. Do you have already an idea or feeling where the problem lies?

drc38, 2024-04-30:
`Get Configuration returned unknown key for: ChargingScheduleAllowedChargingRateUnit`

If you want more details enable debugging as per the readme

LukasNitsche, 2024-05-06:
Below please find the log for OCPP from a restart of homeassistant below. 

## Debug log

```text
2024-05-06 09:45:23.817 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-06 09:45:31.749 INFO (MainThread) [custom_components.ocpp] config = {}
2024-05-06 09:45:36.840 WARNING (MainThread) [custom_components.ocpp] Skipping websocket subprotocol validation
2024-05-06 09:45:36.841 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/0311304103123230841
2024-05-06 09:45:36.841 INFO (MainThread) [custom_components.ocpp] Charger 0311304103123230841 connected to 0.0.0.0:9000.
2024-05-06 09:45:39.400 INFO (MainThread) [ocpp] 0311304103123230841: send [2,"c42eaa60-890a-4827-a60c-906e569496d9","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-05-06 09:45:45.120 INFO (MainThread) [ocpp] 0311304103123230841: receive message [2, "0d6dae86-6638-8aa7-5c10-3baa1103884c", "BootNotification", {"chargePointModel":"DUOSIDA Mode3@11KW","chargePointSerialNumber":"0311304103123230841","chargePointVendor":"UCHEN","firmwareVersion":"V1.2@39f19d4+286,F0.gl,ws,ESP32","iccid":"","imsi":""}]
2024-05-06 09:45:45.128 DEBUG (MainThread) [custom_components.ocpp] Received boot notification for 0311304103123230841: {'charge_point_model': 'DUOSIDA Mode3@11KW', 'charge_point_serial_number': '0311304103123230841', 'charge_point_vendor': 'UCHEN', 'firmware_version': 'V1.2@39f19d4+286,F0.gl,ws,ESP32', 'iccid': '', 'imsi': ''}
2024-05-06 09:45:45.128 DEBUG (MainThread) [custom_components.ocpp] Updating device info charger: {'charge_point_model': 'DUOSIDA Mode3@11KW', 'charge_point_serial_number': '0311304103123230841', 'charge_point_vendor': 'UCHEN', 'firmware_version': 'V1.2@39f19d4+286,F0.gl,ws,ESP32', 'iccid': '', 'imsi': ''}
2024-05-06 09:45:45.154 INFO (MainThread) [ocpp] 0311304103123230841: send [3,"0d6dae86-6638-8aa7-5c10-3baa1103884c",{"currentTime":"2024-05-06T07:45:45Z","interval":3600,"status":"Accepted"}]
2024-05-06 09:45:45.254 INFO (MainThread) [ocpp] 0311304103123230841: receive message [3, "c42eaa60-890a-4827-a60c-906e569496d9", {"configurationKey":[{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"}]}]
2024-05-06 09:45:45.259 WARNING (MainThread) [custom_components.ocpp] Force Smart Charging feature profile
2024-05-06 09:45:45.260 DEBUG (MainThread) [custom_components.ocpp] Feature profiles returned: Profiles.CORE|SMART|RES|REM|AUTH
2024-05-06 09:45:45.264 INFO (MainThread) [ocpp] 0311304103123230841: send [2,"2c8db87a-a0da-49cf-a6be-03845fbd1c6a","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-05-06 09:45:45.422 INFO (MainThread) [ocpp] 0311304103123230841: receive message [2, "0d6dafbc-6638-8aa7-5c10-3baa1103884c", "StatusNotification", {"connectorId":1,"errorCode":"NoError","info":"","status":"Available","vendorId":"","vendorErrorCode":""}]
2024-05-06 09:45:45.444 INFO (MainThread) [ocpp] 0311304103123230841: send [3,"0d6dafbc-6638-8aa7-5c10-3baa1103884c",{}]
2024-05-06 09:45:45.554 INFO (MainThread) [ocpp] 0311304103123230841: receive message [3, "2c8db87a-a0da-49cf-a6be-03845fbd1c6a", {"configurationKey":[{"key":"NumberOfConnectors","readonly":true,"value":"1"}]}]
2024-05-06 09:45:45.556 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for NumberOfConnectors: 1
2024-05-06 09:45:45.558 INFO (MainThread) [ocpp] 0311304103123230841: send [2,"e8c30d76-16a7-4c34-bd2d-cdf8fc6f6b10","GetConfiguration",{"key":["HeartbeatInterval"]}]
2024-05-06 09:45:45.733 INFO (MainThread) [ocpp] 0311304103123230841: receive message [3, "e8c30d76-16a7-4c34-bd2d-cdf8fc6f6b10", {"configurationKey":[{"key":"HeartbeatInterval","readonly":false,"value":"3600"}]}]
2024-05-06 09:45:45.735 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for HeartbeatInterval: 3600
2024-05-06 09:45:45.736 DEBUG (MainThread) [custom_components.ocpp] '0311304103123230841' trying measurand 'Current.Export'
2024-05-06 09:45:45.743 INFO (MainThread) [ocpp] 0311304103123230841: send [2,"07dffc2e-7eff-42d5-b6a7-c33369dd9a8e","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export"}]
2024-05-06 09:45:45.854 INFO (MainThread) [ocpp] 0311304103123230841: receive message [3, "07dffc2e-7eff-42d5-b6a7-c33369dd9a8e", {"status":"Accepted"}]
2024-05-06 09:45:45.857 DEBUG (MainThread) [custom_components.ocpp] '0311304103123230841' adding measurand 'Current.Export'
2024-05-06 09:45:45.858 DEBUG (MainThread) [custom_components.ocpp] '0311304103123230841' trying measurand 'Current.Import'
2024-05-06 09:45:45.859 INFO (MainThread) [ocpp] 0311304103123230841: send [2,"2aa1b4e9-7f3b-44ec-95bf-8dc1a4494839","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
2024-05-06 09:45:46.036 INFO (MainThread) [ocpp] 0311304103123230841: receive message [3, "2aa1b4e9-7f3b-44ec-95bf-8dc1a4494839", {"status":"Accepted"}]
2024-05-06 09:45:46.038 DEBUG (MainThread) [custom_components.ocpp] '0311304103123230841' adding measurand 'Current.Import'
2024-05-06 09:45:46.038 DEBUG (MainThread) [custom_components.ocpp] '0311304103123230841' trying measurand 'Current.Offered'
[... cut ...]
2024-05-06 09:45:56.072 INFO (MainThread) [ocpp] 0311304103123230841: receive message [2, "0d6dd938-6638-8ab2-72e8-4cd18185b566", "StatusNotification", {"connectorId":1,"errorCode":"NoError","info":"","status":"Available","vendorId":"","vendorErrorCode":""}]
2024-05-06 09:45:56.083 INFO (MainThread) [ocpp] 0311304103123230841: send [3,"0d6dd938-6638-8ab2-72e8-4cd18185b566",{}]
2024-05-06 09:45:56.904 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to '0311304103123230841': ping=1.0 ms, pong=61.0 ms

```

drc38, 2024-05-06:
You can ignore the error message, looks like the charger has a few bugs in its ocpp implementation

LukasNitsche, 2024-05-06:
Yes, seems that way. I get many notifcations in HomeAssistant when the integration reloads (please see below). Setting the ocpp integrations log level to error or critical or notset does not help. Can I disable the messages another way since they are quite annoying?

[image]

closed 2024-05-14 by LukasNitsche