lbbrhzn/ocpp #1150: Some measureands not working in since 0.5 for WallboxPulsar Plus

marcofl, 2024-05-01:
**Describe the bug**
Since 0.5.x I can't get some measurands from my Wallbox Pulsar Plus (Firmware 6.4.41) like Power.Active.Import - it simply stays at 0 during charging.

I'm now back on 0.4.42 and it works again. I can't say whether it happend after the firmware upgrade or after the update from 0.4.x to 0.5.x

**To Reproduce**
Steps to reproduce the behavior:
1. Wallbox Pulsar Plus with stock firmware or current 6.4.41 connected to Home Assistant 2024.4.4 with OCPP 0.5.x
2. Power.Active.Import measurand seletcted
3. start charging
4. Entity in Home Assistant for Power.Active.Import stays at 0 and never gets updated.
5. Downgrade to OCPP 0.4.42 and Power.Active.Import works again during charing.

**Expected behavior**
Power.Active.Import gets update during charging

**Screenshots**
log from 0.4.42
```
2024-05-01 10:47:59.026 INFO (MainThread) [ocpp] WallboxPulsar: receive message [2,"1269728409","MeterValues",{"meterValue": [{"timestamp": "2024-05-01T08:47:58Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "3268.8"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "9699"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "14.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "14"} ]} ],"connectorId": 1,"transactionId": 1714552558}]
2024-05-01 10:47:59.026 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Import, phase_info: {'unit': 'A', 'L1': 14.4, 'L2': 0.0, 'L3': 0.0} value: 14.400000 unit :A
```

for 0.5.x Power.Active.Import is not in the receive message, but I see during "initialization" Power.Active.Import got accepted by the Wallbox, but it is never received during charging. I currently can't reproduce to generate logs as my car is full and not charging.

**Additional context**
I also tried to switch to  `all_measurands = MEASURANDS` in api.py, but that didn't fix it or gave me more insights in the debug log.

marcofl, 2024-05-01:
Update, during reproducing the logs with 0.5.2 it worked again.

logs from 0.5.2 during charging:
```
2024-05-01 11:39:04.764 INFO (MainThread) [ocpp] WallboxPulsar: receive message [2,"1014383070","MeterValues",{"meterValue": [{"timestamp": "2024-05-01T09:39:04Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "3297.6"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "9699"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "14.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "14"} ]} ],"connectorId": 1,"transactionId": 1714556302}]
```

This time it was no fresh install of 0.5.2 but an update from 0.4.42 without re-adding the integration in hass. maybe there is something wrong with the config? how can I check that?

lbbrhzn, 2024-05-17:
In v0.5.x, the ocpp integration tries try to autodetect the monitored variables by adding them one by one, skipping those that result in an error from the charger.  It appears the Wallbox does not send an error, even though it does not support the (large) list of measurands.   We should give the user the option to do manual configuration to work around this.

Roarrk, 2024-06-15:
Changing the monitored_variables and restarting home assistant solved the same issue for me, thanks @marcofl :)

stefan73, 2024-07-07:
Same here for a Entratek Power Dot Fix (which is an IoCharger IOCAH10-22T). With the default behaviour of 0.5x the measurands have reduced to voltage only. I tried all_measurands = MEASURANDS before and it did not fix it. 

Used ssh to change /config/.storage/core.config_entries as follows.
"monitored_variables": "Current.Import,Current.Offered,Energy.Active.Import.Interval,Energy.Active.Import.Register,Power.Active.Import,Temperature,Voltage,SoC",

Now the measureands work again. Homeassistant reports on restart that a reboot is required.
A reboot is required to apply MeterValuesSampledData=Current.Import,Current.Offered,Energy.Active.Import.Interval,Energy.Active.Import.Register,Power.Active.Import,Temperature,Voltage,SoC
Booted wallbox and homeassistant multiple times. Message does not vanish. Seems to me another issue with the wallbox firmware, which I can ignore.

P.S.: This fixes https://github.com/lbbrhzn/ocpp/issues/1124

Disincentivato, 2024-07-28:
Sorry, how can I:

 manually change the core.config_entries to:

          "monitored_variables": "Power.Active.Import,Power.Offered,Current.Import,Current.Offered"

I have the same problem, 0kW power active import and export.
Thanks.

lbbrhzn, 2024-09-09:
v0.5.12 supports manual configuration of the measurands

kltye, 2025-01-16:
I know this was opened (and then closed) some time back regarding this issue, but I'm not quite able to get a handle on what the issue is exactly, nor how I should fix it. I see that the operands are able to be selected manually, but where exactly can this be done? I see this UI in the docs:

[image]

But I'm not sure how to get to it. Do I have to remove the entity and re-add it?

I've tried to edit "monitored_variables" in core.config_entries, but that doesn't seem to do anything for me either.