lbbrhzn/ocpp #1152: Charge rate on 3-phase charger

confucius78, 2024-05-05:
**Describe the bug**
Setting max amp or using the ‘set charge rate’ service do not affect charge rate on 3-phase Alfen charger.

drc38, 2024-05-05:
Have a look at #1035, you'll need to enable debug to see what's going on. It'll be an issue with the charger.

confucius78, 2024-05-05:
Hold on...this was working fine with this specific integration until a few weeks ago. Does this mean there is new functionality in the integration that is not backwards compatible? Doesn't sound like a charger bug to me.

I'll have a detailed look at the other topic, but seems to me this has nothing to do with the charger.

drc38, 2024-05-06:
The new service hasn't been modified since its release in 5.2 (it did not exist prior), I suggest debugging to see what's going on.

confucius78, 2024-05-07:
So I have rolled back to 5.1 and it works again. As this issue was introduced with 5.2, I don’t see how this is a charger bug.

The mentioned other issue may still be a valid pointer, but I do not have the time or knowledge to become familiar enough with ocpp to make it work.

Please let me know if this will be addressed in code moving forward, if not I’ll see how I can manually load the integration and retain v5.1. I can support with testing a fix if it helps

drc38, 2024-05-07:
Here is the compare between the two versions:
https://github.com/lbbrhzn/ocpp/compare/v0.5.1...v0.5.2
Unless you can identify the issue in the code by way of debug logs there's not much anyone else can do to help.

closed 2024-07-14 by github-actions[bot]