lbbrhzn/ocpp #1153: BG SyncEV Rebooting Constantly

M450NC, 2024-05-06:
I have a BG SyncEV charger. I have set up the OCPP server URL on the charger and I am able to briefly control the charger via home assistant however the charger to be constantly reconnecting/rebooting and everything goes unavailable and then comes back.

[image]

I enabled debug logging and here are some of the latest logs in home-assistant.log:

```
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 327, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 958, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 361, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 459, in post_connect
    await self.get_supported_features()
  File "/config/custom_components/ocpp/api.py", line 571, in get_supported_features
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 331, in call
    raise asyncio.TimeoutError(
TimeoutError: Waited 30s for response on [2,"98f5355a-2e81-4326-bfdc-8fbe619242fc","GetConfiguration",{"key":["SupportedFeatureProfiles"]}].
2024-05-06 10:51:25.350 ERROR (DbWorker_0) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 327, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 958, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 361, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 459, in post_connect
    await self.get_supported_features()
  File "/config/custom_components/ocpp/api.py", line 571, in get_supported_features
    resp = await self.call(req)
[... 102 lines cut ...]
  File "/config/custom_components/hass_agent/__init__.py", line 147, in updated
    hass.async_create_task(handle_apis_changed(hass, entry, apis))
  File "/usr/src/homeassistant/homeassistant/core.py", line 816, in async_create_task
    self.verify_event_loop_thread("async_create_task")
  File "/usr/src/homeassistant/homeassistant/core.py", line 440, in verify_event_loop_thread
    frame.report(
  File "/usr/src/homeassistant/homeassistant/helpers/frame.py", line 162, in report
    _report_integration(what, integration_frame, level, error_if_integration)
  File "/usr/src/homeassistant/homeassistant/helpers/frame.py", line 203, in _report_integration
    raise RuntimeError(
RuntimeError: Detected that custom integration 'hass_agent' calls async_create_task from a thread at custom_components/hass_agent/__init__.py, line 147: hass.async_create_task(handle_apis_changed(hass, entry, apis)). Please report it to the author of the 'hass_agent' custom integration.

2024-05-06 10:52:18.252 INFO (MainThread) [custom_components.ocpp] Websocket Subprotocol matched: ocpp1.6
2024-05-06 10:52:18.252 INFO (MainThread) [custom_components.ocpp] Charger websocket path=/BS512S213
2024-05-06 10:52:18.252 INFO (MainThread) [custom_components.ocpp] Charger BS512S213 reconnected to 0.0.0.0:9000.
2024-05-06 10:52:18.252 DEBUG (MainThread) [custom_components.ocpp] Reconnect websocket to BS512S213
2024-05-06 10:52:19.280 INFO (MainThread) [ocpp] BS512S213: receive message [2,"3412","BootNotification",{"chargePointVendor":"BG Sync EV","chargePointModel":"EVT77G-01","chargePointSerialNumber":"BS512S213","chargeBoxSerialNumber":"3030012205205633","firmwareVersion":"AC7K-88071-V3.06"}]
2024-05-06 10:52:19.281 DEBUG (MainThread) [custom_components.ocpp] Received boot notification for BS512S213: {'charge_point_vendor': 'BG Sync EV', 'charge_point_model': 'EVT77G-01', 'charge_point_serial_number': 'BS512S213', 'charge_box_serial_number': '3030012205205633', 'firmware_version': 'AC7K-88071-V3.06'}
2024-05-06 10:52:19.281 DEBUG (MainThread) [custom_components.ocpp] Updating device info charger: {'charge_point_vendor': 'BG Sync EV', 'charge_point_model': 'EVT77G-01', 'charge_point_serial_number': 'BS512S213', 'charge_box_serial_number': '3030012205205633', 'firmware_version': 'AC7K-88071-V3.06'}
2024-05-06 10:52:19.309 INFO (MainThread) [ocpp] BS512S213: send [3,"3412",{"currentTime":"2024-05-06T09:52:19Z","interval":3600,"status":"Accepted"}]
2024-05-06 10:52:19.891 INFO (MainThread) [ocpp] BS512S213: receive message [2,"3413","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Unavailable"}]
2024-05-06 10:52:19.897 INFO (MainThread) [ocpp] BS512S213: send [3,"3413",{}]
```

Thanks in advance for any assistance, I'm fairly new to this so please let me know if you require anything else.

drc38, 2024-05-06:
The charger is not ocpp compliant, as it does report back any `SupportedFeatureProfiles`

drc38, 2024-05-06:
Look through #899 for various hacks that can be used on non-compliant units

ganterd, 2024-05-11:
I have the same charger, and had the same issue. I had to modify the code a bit to extend a timeout, and ignore the SupportedFeatures issue.

I'm not able to post a diff at the moment, but when I get a chance I'll do it.

hermanhobnob, 2024-05-18:
Oddly I've been charging with a BG SyncEV and this addon quite happily for a few months now. I've reboots a few times a day in the logs but they never seem to have caused an actual problem getting my 12:30am - 4:30am charge every day. However today something's gone wrong after updating HA and I had to edit the config to remove a hertz field which has apparently disappeared. It's late here and I'm basically confused and not sure if anything is working on not, but I just wanted to say until today I've been charging quite happily for a few months without any serious issues. The reboot notifications were annoying but that's about it. I'll check in again once I've worked out what (if anything) is going on.

kerdridge76, 2024-05-19:
> I have the same charger, and had the same issue. I had to modify the code a bit to extend a timeout, and ignore the SupportedFeatures issue.
> 
> I'm not able to post a diff at the moment, but when I get a chance I'll do it.

Hi, i have a A-Tronix wallbox and ocpp 0.5.1. works but Updates greater 0.5.1 are rebooting too. Sorry for my incredible english.

kerdridge76, 2024-05-19:
downgrade smaller 0.5.2 are not available at HACS

hermanhobnob, 2024-05-19:
Just l came back to say I'm running version 0.5.2 and my car charged fine last night so I don't seem to have a problem apart from reboot notifications every few hours which I've set up a automation to ignore. I am however now reluctant to upgrade 😁

Siumba, 2024-05-23:
I have the same issue on 0.5.2 or higher. Every few hours the charger seems to reboot. I downgraded to 0.5.1 (downloaded a release and copied the files over manually in the custom_components folder, and on 0.5.1 I haven't seen one reboot yet. Something seems to be off in the higher versions.

SAUhrich, 2024-06-09:
> I have the same charger, and had the same issue. I had to modify the code a bit to extend a timeout, and ignore the SupportedFeatures issue.
> 
> I'm not able to post a diff at the moment, but when I get a chance I'll do it.

Hi, I'm still trying to deal with this issue, if you can, please post specifically which files you modified and what parameters you changed? I've been trying for a couple of weeks to solve this. It seems like it is a timeout issue here as well, ocpp will talk to my inverter properly for around 1 min, then disconnect, and then around 30 seconds later it will reconnect, reporting that the charger rebooted.

ArtjomE, 2024-07-23:
Hi, 
I have a Vestel EVC04 Charger. Overall, it works well, but I receive the same error message "Ocpp integration: Charger 7000262321000253 rebooted" every minute while charging. When it's in the "not connected" state, the message appears sporadically every few hours.

OCCP Version v0.5.8
Vestel EVC04 Firmware: v3.150.0-1.0.144.0-v7.0.14

[image]

Plawasan, 2024-08-06:
Adding Growatt THOR_22AS-V2.2.15-20231130 to the list of constantly rebooting chargers.

```
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 1023, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 459, in post_connect
    await self.get_supported_features()
  File "/config/custom_components/ocpp/api.py", line 575, in get_supported_features
    feature_list = (resp.configuration_key[0][om.value.value]).split(",")
                    ~~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

Crown3r, 2024-08-11:
Same here! Does somebody have found a workaround or something?

PR #1283 (2024-08-20): Improve get feature error handling
should fix #1272 fix #1153
closed 2024-08-20 by lbbrhzn