lbbrhzn/ocpp #1155: TypeConstraintViolation: serial number length in the schema

ovomaltin, 2024-05-07:
**Describe the bug**
I am trying to integrate my Schneider EVLink wallbox with OCPP.
I am new to HomeAssistant and Python but did develop some OCPP software before.
My box is using OCCP 1.6. The websocket is working but the schema used in https://github.com/mobilityhouse/ocpp/blob/master/ocpp/v16/schemas/BootNotification.json prevent the bootnotification to be successful.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ocpp/messages.py", line 226, in validate_payload
    validator.validate(message.payload)
  File "/usr/local/lib/python3.12/site-packages/jsonschema/validators.py", line 451, in validate
    raise error
jsonschema.exceptions.ValidationError: '3N182720082F1S1B7551700016' is too long
Failed validating 'maxLength' in schema['properties']['chargePointSerialNumber']:
    {'maxLength': 25, 'type': 'string'}
On instance['chargePointSerialNumber']:
    '3N182720082F1S1B7551700016'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 187, in route_message
    await self._handle_call(msg)
  File "/config/custom_components/ocpp/api.py", line 1011, in _handle_call
    await super()._handle_call(msg)
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 215, in _handle_call
    validate_payload(msg, self._ocpp_version)
  File "/usr/local/lib/python3.12/site-packages/ocpp/messages.py", line 240, in validate_payload
    raise TypeConstraintViolationError(
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "'3N182720082F1S1B7551700016' is too long", 'ocpp_message': }
2024-05-07 08:54:44.838 INFO (MainThread) [ocpp] WB-001: send [4,"19","TypeConstraintViolation","Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. \u201csomestring\u201d: 12)",{"cause":"'3N182720082F1S1B7551700016' is too long","ocpp_message":"[2,\"19\",\"BootNotification\",{\"chargeBoxSerialNumber\":\"EVB1A22P4KI3N18322120130025084EBD\",\"chargePointModel\":\"EVlink Smart Wallbox\",\"chargePointSerialNumber\":\"3N182720082F1S1B7551700016\",\"chargePointVendor\":\"Schneider Electric\",\"firmwareVersion\":\"3.3.0.12\"}]"}]
2024-05-07 08:54:44.856 DEBUG (MainThread) [custom_components.ocpp] Connection closed to 'WB-001': received 1000 (OK); then sent 1000 (OK)
2024-05-07 08:54:44.857 INFO (MainThread) [custom_components.ocpp] Charger WB-001 disconnected from 0.0.0.0:9000.

**To Reproduce**
Not possible without a bpx

**Desktop (please complete the following information):**
Home assistant 2024.5.1 running in proxmox VM

If you need any additional info just let me know.
Thanks,
OvO

drc38, 2024-05-07:
Search the discussion and issues, pretty sure it's been covered already

ovomaltin, 2024-05-07:
@drc38 I searched with 'TypeConstraintViolation' within issues and discussion and did not find anything.
I can't change the serial number of my wallbox.
Serial number schema is limited to 25 characters and mine is 33.
So without having the constraint changed in the ocpp library, I do not see what I can do.

drc38, 2024-05-07:
Try searching on BootNotification or EVLink

sandersr, 2024-05-13:
Looks like the vendor did not implement the spec properly. The serial number string should be up to 25 characters long.

Have you tried reaching out to the vendor and raising a support case?

melstig, 2024-05-14:
I have the same problem with my Schneider charger, it has worked before but now on a newly installed Home Assistant under Proxmox LXC I am unable to connect the charger. The configuration choice "Skip OCPP schema validation" in the integration should to my knowledge bypass the mentioned sanity check for the length of the serial numbers but it is not working for Me.

Prevously as a native Docker HA instance I have also edited some files adding support for a longer SN in the BootNotification.json file but I cannot make it work in the new setup under Proxmox LXC.

        "chargeBoxSerialNumber": {
            "type": "string",
            "maxLength": 33
        },

ovomaltin, 2024-05-18:
> I have the same problem with my Schneider charger, it has worked before but now on a newly installed Home Assistant under Proxmox LXC I am unable to connect the charger. The configuration choice [...]
I did not have much time to test, but same here. The skip ocpp schema validation in the api.key file does not seem to work.
And probably some user authorization issues but I was not able to find and update the json file.
I will try to make another trial if I find some time, but the the modification of the api.py file should be sufficient in principle.

caitotheonlypotato, 2024-06-11:
> I have the same problem with my Schneider charger, it has worked before but now on a newly installed Home Assistant under Proxmox LXC I am unable to connect the charger. The configuration choice [...]

Same problem here, how can I manually make this change to accept the longer serial number.  I can't locate the file

[image]

caitotheonlypotato, 2024-06-12:
> Same problem here, how can I manually make this change to accept the longer serial number. I can't locate the file [...]

I got it sorted.  I'm running a VM on a Windows machine.  I didn't realise I'd have to use SSH and then further go into the docker container to find the relevant directory.  I modified the schema to accept a longer serial number in the bootnotification file

bardahlm, 2024-08-17:
Upgrade to newer firmware, I believe SE have fixed this issue. See the workaround described in #727. See also #213.

closed 2024-10-24 by github-actions[bot]