lbbrhzn/ocpp #1157: Warning: Set availability failed with response Rejected Project EV Apex 75

OneScuzzie2k, 2024-05-11:
Good Morning,

I have got this integration setup, but I don't seem to get a current draw of power to the car show, also I keep getting notifications on HA saying Warning: Set availability failed with response Rejected

The availability and charge control work, but nothing else seems to be correct or updated.
[image]

sandersr, 2024-05-14:
can you check the HA logs and grep for OCPP messages?
$ grep ocpp  /config/home-assistant.log 

Maybe the charger's firmware does not enable any measurands out of the box. If it doesn't, you might try the following:
go to developer tools  -> services
find: Open Charge Point Protocol (OCPP): Configure charger features
paste the following:

service: ocpp.configure
data:
  ocpp_key: MeterValuesSampledData
  value: Current.Import,Current.Offered,Energy.Active.Import.Interval,Energy.Active.Import.Register,Power.Active.Import,Power.Offered,Temperature,Voltage

If that works, you might want to create an automation which which re-applies it every time charger restarts

OneScuzzie2k, 2024-05-14:
I seem to be getting the correct Power active now anyway removed and re-added but still getting Warning: Set availability failed with response Rejected everytime I activate the charger.

When i carry out what you stated i get 
[image]

docomarc, 2024-05-19:
I have the same issue. I reinstalled the previous v0.5.5.

closed 2024-07-27 by github-actions[bot]