lbbrhzn/ocpp #1162: Integration could not load, after update to Homea Assistant Core 2024.5.3

juzel69, 2024-05-15:
After update to Version 2024.5.3 the integration could not be started. I removed the integration and reinstall it. But now it is not possible to start it.
[image]

Regards Juerg

lutfijw, 2024-05-15:
It is because of ha.UnitOfFrequency.HERTZ declaration in const.py file. HA seem to removed it so the integration can not look up the value. Suggest to remove this. I did it and it work good.

thomas6779, 2024-05-15:
> It is because of ha.UnitOfFrequency.HERTZ declaration in const.py file. HA seem to removed it so the integration can not look up the value. Suggest to remove this. I did it and it work good.

thanks. That works fine!

JakobEgPedersen, 2024-05-15:
> It is because of ha.UnitOfFrequency.HERTZ declaration in const.py file. HA seem to removed it so the integration can not look up the value. Suggest to remove this. I did it and it work good.

Thank you, this solved my issue. It is working again now.

closed 2024-05-16 by drc38