lbbrhzn/ocpp #1199: Detected blocking call to open inside the event loop by custom integration 'ocpp'

Roos-AID, 2024-06-09:
**Describe the bug**
Updated OCPP to 0.5.7 and to HA 2024.6.1 . 
Since then I get flood of warnings in log, 

**To Reproduce**
Update HA and OCPP to latest versions and check the HA log

**Expected behavior**
No flood of warning errors

Logger: homeassistant.util.loop
Bron: util/loop.py:84
Eerst voorgekomen: 09:46:49 (764 gebeurtenissen)
Laatst gelogd: 09:48:52

Detected blocking call to open inside the event loop by custom integration 'ocpp' at custom_components/ocpp/api.py, line 767: resp = await self.call(req) (offender: /usr/local/lib/python3.12/site-packages/ocpp/messages.py, line 164: with open(path, "r", encoding="utf-8-sig") as f:), please create a bug report at https://github.com/lbbrhzn/ocpp/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/config/custom_components/ocpp/switch.py", line 125, in async_turn_on self._state = await self.central_system.set_charger_state( File "/config/custom_components/ocpp/api.py", line 310, in set_charger_state resp = await self.charge_points[cp_id].start_transaction() File "/config/custom_components/ocpp/api.py", line 767, in start_transaction resp = await self.call(req)
Detected blocking call to open inside the event loop by custom integration 'ocpp' at custom_components/ocpp/api.py, line 1005: await self.hass.async_create_task(super()._handle_call(msg)) (offender: /usr/local/lib/python3.12/site-packages/ocpp/messages.py, line 164: with open(path, "r", encoding="utf-8-sig") as f:), please create a bug report at https://github.com/lbbrhzn/ocpp/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 238, in start await self.route_message(message) File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 261, in route_message await self._handle_call(msg) File "/config/custom_components/ocpp/api.py", line 1005, in _handle_call await self.hass.async_create_task(super()._handle_call(msg))
Detected blocking call to open inside the event loop by custom integration 'ocpp' at custom_components/ocpp/api.py, line 704: resp = await self.call(req) (offender: /usr/local/lib/python3.12/site-packages/ocpp/messages.py, line 164: with open(path, "r", encoding="utf-8-sig") as f:), please create a bug report at https://github.com/lbbrhzn/ocpp/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 121, in async_set_value await entity.async_set_native_value(native_value) File "/config/custom_components/ocpp/number.py", line 129, in async_set_native_value resp = await self.central_system.set_max_charge_rate_amps( File "/config/custom_components/ocpp/api.py", line 298, in set_max_charge_rate_amps return await self.charge_points[cp_id].set_charge_rate(limit_amps=value) File "/config/custom_components/ocpp/api.py", line 704, in set_charge_rate resp = await self.call(req)
Detected blocking call to open inside the event loop by custom integration 'ocpp' at custom_components/ocpp/api.py, line 787: resp = await self.call(req) (offender: /usr/local/lib/python3.12/site-packages/ocpp/messages.py, line 164: with open(path, "r", encoding="utf-8-sig") as f:), please create a bug report at https://github.com/lbbrhzn/ocpp/issues Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request resp = await request_handler(request) File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle resp = await handler(request) File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware return await handler(request) File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware response = await handler(request) File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle result = await handler(request, **request.match_info) File "/usr/src/homeassistant/homeassistant/components/websocket_api/http.py", line 54, in get return await WebSocketHandler(request.app[KEY_HASS], request).async_handle() File "/usr/src/homeassistant/homeassistant/components/websocket_api/http.py", line 438, in async_handle async_handle_str(command_msg_data) File "/usr/src/homeassistant/homeassistant/components/websocket_api/connection.py", line 229, in async_handle handler(sel [... cut ...] ce handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/config/custom_components/ocpp/switch.py", line 139, in async_turn_off resp = await self.central_system.set_charger_state( File "/config/custom_components/ocpp/api.py", line 312, in set_charger_state resp = await self.charge_points[cp_id].stop_transaction() File "/config/custom_components/ocpp/api.py", line 787, in stop_transaction resp = await self.call(req)

mariusves, 2024-06-09:
Can confirm this with OCPP 0.5.7 and HA 2024.6.1. Charger is an EVbox Elvi. Worked stable beforehand. OCPP loses connection to the charger (or vise versa) every couple of minutes.

drc38, 2024-06-09:
@lbbrhzn any ideas on how these calls can run outside the event loop or amended to run within it without causing errors?

bec-aus, 2024-06-12:
I have this too with an MG Smartcharger (which shows up as a Star Charge). It is updated to as newer firmware as I can find (V1.1.2.7.6b107.1_C_003) and has never really created a stable connection to HA.  I was trouble shooting it when I found this.

drc38, 2024-06-14:
This may need to be fixed upstream see https://github.com/mobilityhouse/ocpp/issues/647

drc38, 2024-08-13:
Still waiting on upstream merge

closed 2024-10-20 by github-actions[bot]