lbbrhzn/ocpp #1203: Topdon EVSE continuously reboots

bkcberry, 2024-06-12:
## Version of the custom_component

0.5.6

## Configuration

Not sure what to put here? I left most settings as default and checked skip schema validation

## Describe the bug

When pointed towards home assistant/ocpp, the evse will repeatedly connect for a few seconds before rebooting. I did see where someone on amazon who bought this same EVSE was apparently able to get it working with this same addin/repo, and issue #1023 claims that what looks to be the same EVSE from amazon with different branding works as well

## Debug log

```text

2024-06-12 13:41:06.223 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-06-12 13:41:07.331 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'localtuya', 'history', 'map', 'config', 'intent', 'scene', 'backup', 'input_boolean', 'hardware', 'input_select', 'blueprint', 'application_credentials', 'onboarding', 'automation', 'august', 'diagnostics', 'file_upload', 'zwave_js', 'stt', 'tasmota', 'input_button', 'trace', 'template', 'default_config', 'homeassistant_alerts', 'input_datetime', 'logbook', 'esphome', 'ocpp', 'wake_word', 'lovelace', 'group', 'system_health', 'conversation', 'schedule', 'counter', 'mobile_app', 'image_upload', 'person', 'ffmpeg', 'script', 'assist_pipeline', 'hacs', 'thread', 'media_source', 'zone', 'tts', 'owntracks', 'my', 'energy', 'sun', 'timer', 'input_number', 'input_text', 'search', 'stream', 'mqtt', 'switch_as_x', 'persistent_notification', 'device_automation', 'tag', 'analytics'}
2024-06-12 13:41:07.640 INFO (MainThread) [homeassistant.setup] Setting up ocpp
2024-06-12 13:41:07.640 INFO (MainThread) [custom_components.ocpp] config = {}
2024-06-12 13:41:07.640 INFO (MainThread) [homeassistant.setup] Setup of domain ocpp took 0.00 seconds
2024-06-12 13:41:08.714 INFO (MainThread) [homeassistant.components.sensor] Setting up ocpp.sensor
2024-06-12 13:41:08.725 INFO (MainThread) [homeassistant.components.switch] Setting up ocpp.switch
2024-06-12 13:41:08.726 INFO (MainThread) [homeassistant.components.number] Setting up ocpp.number
2024-06-12 13:41:08.727 INFO (MainThread) [homeassistant.components.button] Setting up ocpp.button
2024-06-12 14:00:25.589 INFO (MainThread) [homeassistant.components.sensor] Setting up ocpp.sensor
2024-06-12 14:00:25.590 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_reactive_export_register
2024-06-12 14:00:25.590 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_time_session
2024-06-12 14:00:25.591 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_reactive_import
2024-06-12 14:00:25.592 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_voltage
2024-06-12 14:00:25.592 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_reactive_import_register
2024-06-12 14:00:25.593 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_current_offered
2024-06-12 14:00:25.594 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_current_import
2024-06-12 14:00:25.595 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_meter_start
2024-06-12 14:00:25.595 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_reactive_import_interval
2024-06-12 14:00:25.596 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_active_export_interval
2024-06-12 14:00:25.596 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_soc
2024-06-12 14:00:25.597 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_active_export_register
2024-06-12 14:00:25.597 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_frequency
2024-06-12 14:00:25.598 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_active_import_interval
2024-06-12 14:00:25.598 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_factor
2024-06-12 14:00:25.599 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_reactive_export_interval
2024-06-12 14:00:25.600 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_rpm
2024-06-12 14:00:25.601 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_current_export
2024-06-12 14:00:25.601 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_reactive_export
2024-06-12 14:00:25.602 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_active_export
2024-06-12 14:00:25.602 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_transaction_id
2024-06-12 14:00:25.602 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_active_import_register
2024-06-12 14:00:25.603 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_active_import
2024-06-12 14:00:25.603 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_temperature
2024-06-12 14:00:25.604 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_offered
2024-06-12 14:00:25.604 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_session
2024-06-12 14:00:25.605 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_status
2024-06-12 14:00:25.605 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_status_connector
2024-06-12 14:00:25.607 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_heartbeat
2024-06-12 14:00:25.607 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_latency_ping
2024-06-12 14:00:25.608 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_latency_pong
2024-06-12 14:00:25.609 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_error_code
2024-06-12 14:00:25.609 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_error_code_connector
2024-06-12 14:00:25.610 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_stop_reason
2024-06-12 14:00:25.611 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_status_firmware
2024-06-12 14:00:25.611 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_reconnects
2024-06-12 14:00:25.612 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_id_tag
2024-06-12 14:00:25.613 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_id
2024-06-12 14:00:25.613 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_model
2024-06-12 14:00:25.614 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_vendor
[... cut ...]
2024-06-12 14:03:44.372 INFO (MainThread) [ocpp] 40252204456519: receive message [3,"1adf0660-d03f-48e5-bb04-d9cfd32d6ab0",{"status":"NotSupported"}]
2024-06-12 14:03:44.372 DEBUG (MainThread) [custom_components.ocpp] '40252204456519' trying measurand 'Frequency'
2024-06-12 14:03:44.373 INFO (MainThread) [ocpp] 40252204456519: send [2,"4db4c316-fae9-4bdc-97d9-8d41dec3d152","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Frequency"}]
2024-06-12 14:03:44.431 INFO (MainThread) [custom_components.ocpp] Charger 40252204456519 disconnected from 0.0.0.0:9000.

```

merkurmaniac, 2024-06-12:
I just bought one of these after reading the Home Assistant mention in the guy's review.  I did NOT get the 123456789 password to work, so I am at a loss.  Did you get that far ?

bkcberry, 2024-06-13:
Yeah it's 1-8 not 1-9. Make sure you write down all the info in it before you change anything (serial number, factory OCPP url, etc) because resetting the unit does not restore the original values

bkcberry, 2024-06-13:
Also FWIW, the problems the grizzl-e charger have looked similar so I tried the work around that's supposed to make those work but it made no difference

SAUhrich, 2024-06-13:
Just adding my experience with this charger and this issue, I bought and configured a MaxpeedingRods charger last August and configured it to work with this OCPP integration. It worked well, and provided me with the 3 key features I was looking for: HA control to start/stop the charge, current control during the charge, and tracking the power used daily.  I'm 100% solar at home and need to manage the big power loads timing. This worked fantastically until 0.5.2 when this behavior started where it will communicate for around 30sec, then lose connection, then reconnect about 60sec later, where the integration reports a charger reboot. The charger has not. however rebooted (unless it magically reboots without stopping the current charge cycle :).

Like I said, it worked fine until 0.5.2. I suspect it is some kind of Websockets timing issue or something similar, and I have tried many adjustments to the configuration options related to timing (WEBSOCKET_CLOSE_TIMEOUT, WEBSOCKET_PING_TRIES, WEBSOCKET_PING_INTERVAL, WEBSOCKET_PING_TIMEOUT, SLEEP_TIME, etc.) but no changes to those values seems to have any effect on the behavior. Sadly, I don't know more, I am hoping that a future update will bring a fix! I will certainly post back here though if I have any relevant info/experience.

bkcberry, 2024-06-13:
Thank you, i will try downloading 0.5.1 and see what happens. Is that still the version you're running? If so, is it still working correctly?

SAUhrich, 2024-06-13:
> Thank you, i will try downloading 0.5.1 and see what happens. Is that still the version you're running? If so, is it still working correctly?

Sadly, that did not work for me, I tried removing/installing about 3 or 4 versions back prior to 0.5.2 but saw the same behavior each time. I would be curious if that is the case with your setup too. if you go back to 0.5.1. I'm wondering if there is some other related code that updated around that time that is the real culprit here, like a python update, etc., but just not sure.  :(

bkcberry, 2024-06-13:
Same for me, 0.5.1 made no difference. Sad trombone. Just did a quick test on my lunch break, didn't have time to do much digging

SAUhrich, 2024-06-13:
Thanks bkcberry for the result info. Sometime in the next few days, I am going to try to see if I can tell if I can setup a virtual central server and connect the TapdonMaxpeedingRods charger to it to see if I can see any basic errors with communications. The ocpp addin is based on python code from mobilityhouse (https://github.com/mobilityhouse/ocpp). I am not a python expert or anything close to it (!) but I can will setup a test environment to a test "central" server. I will post anything relevant back here.

bkcberry, 2024-06-14:
I got it working! I downloaded and extracted 0.5.0 and needed to manually install the packages and their respective versions that are listed in requirements.txt

SAUhrich, 2024-06-14:
That is awesome to hear! I will do the same first thing in the morning and try to replicate your success! Thanks for posting, at least I know it’s possible :) btw, I did get a virtual machine setup with the base python test central server and got my charger to connect. It connects fine, although it gives an error about a couple of unknown messages from the charger, but it stays connected. I’m going to set that aside and try the 0.5.0 route in HA. Thx again

SAUhrich, 2024-06-14:
> I got it working! I downloaded and extracted 0.5.0 and needed to manually install the packages and their respective versions that are listed in requirements.txt for 0.5.0

This is a big ask, because I'm pretty ignorant with the way different installs of HA work, for example I can use the Terminal addon to get to my configuration, but things like pip, etc. are not available.

How did you get those packages set to the correct versions? 

I see the list of packages/versions in the requirements.txt. I suspect that one of those package updates is what is really behind the issue (jsonschema==4.19.2, or maybe websockets==12.0), but I don't know how to update/revert the versions of them.

Like I said, that is a big ask because of my ignorance, feel free to say "bruh, go get edumacated" :)

bkcberry, 2024-06-14:
No worries. I run home assistant in a venv (python virtual environment) on ubuntu, it's the instructions [on this page](https://www.home-assistant.io/installation/linux) towards the bottom for "Install Home Assistant Core." I have actually set up a 2nd instance of HA just for testing this addon, and since it requires old packages I may just leave it there with OCPP being the only thing it does. Anyway, follow the instructions on how to install the Core version of HA, but instead of 

`pip3 install homeassistant==2024.6.2` 

you would run 

`pip3 install homeassistant==2023.11.1 colorlog==6.7.0 ruff==0.1.4 ocpp==0.22.0 websockets==12.0 jsonschema==4.19.2 pre-commit pytest-homeassistant-custom-component==0.13.75`

I hope that helps, this is the only way I know of to get this old stuff installed. I couldn't figure out how to install this stuff in docker, and I have no idea how you'd do it if you are running one of the appliances or prebuilt virtual machines

SAUhrich, 2024-06-15:
Thank you so much for the info. I will have to wait a day or too to dig into this, but I think I understand. If I get any other info, or workarounds discovered, I will post back here.

onelight17, 2024-06-15:
Hi all, sorry if I am hijacking this thread but I am also trying to setup the TOPDON charger with this integration. Are there any special steps to get this working? I followed the standard guide and updated the TOPDON configuration to point to ws://<ocpp-central-server>:9000

Does plug and play need to be "No" or leave that as is? 

Any other configuration that needs to be updated? I know it's using the OCPP1.6-J protocol, didn't touch that. Left the "AuthPass" as is as well.

I get no connection on the home assistance logs though. I set the websockets.server to debug and it only shows the entry for starting the listener. 

For the ocpp logs I only see the entry for `[custom_components.ocpp] config = {}`

Full log if it's helpful:

`2024-06-15 23:07:11.746 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration ocpp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-06-15 23:07:12.969 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'media_source', 'tag', 'timer', 'input_button', 'person', 'history', 'image_upload', 'map', 'intent', 'met', 'schedule', 'ffmpeg', 'energy', 'onboarding', 'conversation', 'script', 'input_datetime', 'persistent_notification', 'input_text', 'mobile_app', 'counter', 'logbook', 'device_automation', 'shopping_list', 'diagnostics', 'lovelace', 'input_boolean', 'stream', 'homeassistant_alerts', 'system_health', 'ocpp', 'radio_browser', 'tts', 'search', 'application_credentials', 'my', 'file_upload', 'trace', 'thread', 'config', 'sun', 'zone', 'automation', 'input_number', 'hacs', 'analytics', 'wake_word', 'input_select', 'stt', 'blueprint', 'google_translate', 'cast', 'scene', 'backup', 'hardware', 'default_config', 'assist_pipeline'}
2024-06-15 23:07:13.682 INFO (MainThread) [homeassistant.setup] Setting up ocpp
2024-06-15 23:07:13.682 INFO (MainThread) [custom_components.ocpp] config = {}
2024-06-15 23:07:13.682 INFO (MainThread) [homeassistant.setup] Setup of domain ocpp took 0.00 seconds
2024-06-15 23:07:14.960 INFO (MainThread) [homeassistant.components.sensor] Setting up ocpp.sensor
2024-06-15 23:07:14.993 INFO (MainThread) [homeassistant.components.switch] Setting up ocpp.switch
2024-06-15 23:07:14.995 INFO (MainThread) [homeassistant.components.number] Setting up ocpp.number
2024-06-15 23:07:14.999 INFO (MainThread) [homeassistant.components.button] Setting up ocpp.button
2024-06-15 23:10:35.831 INFO (MainThread) [homeassistant.components.sensor] Setting up ocpp.sensor
2024-06-15 23:10:35.833 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_active_export_interval
2024-06-15 23:10:35.835 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_reactive_export_register
2024-06-15 23:10:35.838 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_transaction_id
2024-06-15 23:10:35.840 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_active_import
2024-06-15 23:10:35.841 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_reactive_export
2024-06-15 23:10:35.842 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_reactive_import
2024-06-15 23:10:35.844 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_meter_start
2024-06-15 23:10:35.845 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_frequency
2024-06-15 23:10:35.846 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_temperature
2024-06-15 23:10:35.847 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_reactive_import_interval
2024-06-15 23:10:35.850 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_power_active_export
2024-06-15 23:10:35.851 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_active_import_interval
2024-06-15 23:10:35.852 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_soc
2024-06-15 23:10:35.853 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_energy_active_export_register
2024-06-15 23:10:35.854 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_current_offered
2024-06-15 23:10:35.858 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_current_import
2024-06-15 23:10:35.859 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_time_session
2024-06-15 23:10:35.860 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_voltage
2024-06-15 23:10:35.861 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.ocpp entity: sensor.charger_current_export
[... cut ...]
2024-06-15 23:10:35.934 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.ocpp entity: number.charger_maximum_current
2024-06-15 23:10:35.936 INFO (MainThread) [homeassistant.components.button] Setting up ocpp.button
2024-06-15 23:10:35.937 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new button.ocpp entity: button.charger_reset
2024-06-15 23:10:35.938 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new button.ocpp entity: button.charger_unlock`

Thanks in advance!

onelight17, 2024-06-15:
Scratch that, I had a silly typo in the IP Address.

So now I actually just have the same issue described here where it loops in the connectivity. Will attempt the same downgrade of versions and see if it resolves. Thanks @bkcberry for posting your solutions!

merkurmaniac, 2024-06-16:
Does this mean that we need to switch back to all old version of home assistant and never upgrade, or is it just some small files that need to be reverted and held?

⁣Get BlueMail for Android ​

bkcberry, 2024-06-16:
I'm not sure, as I said I set up a test environment for this and I believe it was on 2023.12.1, but in the process of installing the packages that are required for v0.5.0 of this addon home assistant got rolled back to 2023.11.1. It may work with something later than that but it also may not. My plan is to just leave OCPP set up on my 2nd instance going forward but that is mostly because it's working now and I'm tired of messing with it

SAUhrich, 2024-06-16:
Just putting this here if the HA OCPP integration devs are reading this issue post. I found this from back in 2020. This is from the MobilityHouse git, which I think this integration is based on. There was an issue early on that presents specifically like this possible reoccurrence with the diconnects/reboot message every 40sec: 
https://github.com/mobilityhouse/ocpp/issues/79

onelight17, 2024-06-17:
For others who may stumble upon this and like me use the docker configuration. The steps are almost the same as the one posted above by @bkcberry 

Just pull the container for home assistant for version 2023.11.1 and then manually extract v0.5.0 of this custom_component "ocpp" and the run the same pip commands

`pip3 install homeassistant==2023.11.1 colorlog==6.7.0 ruff==0.1.4 ocpp==0.22.0 websockets==12.0 jsonschema==4.19.2 pre-commit pytest-homeassistant-custom-component==0.13.75`

All done inside the container. 

That worked for me and no longer getting the disconnect loop.

SAUhrich, 2024-06-27:
Just an FYI after the 0.5.8 release.
I'm still testing, but as far as I can tell so far, no joy on 0.5.8 resolving this issue. In fact it seems to make the connection drop more frequently and not ever really make the connection properly. 

In 0.5.7, it would accept and connect to the charger and actually allow you to send commands (begin or stop charge, change available current, etc) for about 10 seconds before going dead. Then after the 40 second timeout and reboot signal, you could get control for about 10 seconds again.

closed 2024-09-03 by github-actions[bot]