lbbrhzn/ocpp #1207: Connecting multiple EV chargers

VendisFronek, 2024-06-15:
**Describe the bug**
If I connect only one charger everything works. After connecting the second charger, the diagnostic data will remain, but all data from all sensors will disappear.

**Screenshots**
[image]

[image]

LOG report:

2024-06-13 07:11:40.863 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_export_register not found.
2024-06-13 07:11:40.864 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_rpm not found.
2024-06-13 07:11:40.865 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_import_interval not found.
2024-06-13 07:11:40.866 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_export_interval not found.
2024-06-13 07:11:40.866 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_power_factor not found.
2024-06-13 07:11:40.866 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_import_register not found.
2024-06-13 07:11:40.867 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_power_reactive_import not found.
2024-06-13 07:11:40.868 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_power_reactive_export not found.
2024-06-13 07:11:40.888 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'ocpp' at custom_components/ocpp/api.py, line 1005: await self.hass.async_create_task(super()._handle_call(msg)) (offender: /usr/local/lib/python3.12/site-packages/ocpp/messages.py, line 164: with open(path, "r", encoding="utf-8-sig") as f:), please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 238, in start
    await self.route_message(message)
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 261, in route_message
    await self._handle_call(msg)
  File "/config/custom_components/ocpp/api.py", line 1005, in _handle_call
    await self.hass.async_create_task(super()._handle_call(msg))

2024-06-13 07:11:40.890 INFO (MainThread) [ocpp] Wallbox_1: send [3,"1415483299",{}]
2024-06-13 07:11:40.899 INFO (MainThread) [ocpp] Wallbox_1: receive message [2,"1754391931","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Available","timestamp": "2024-06-13T05:11:40Z"}]
2024-06-13 07:11:40.900 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_export_register not found.
2024-06-13 07:11:40.901 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_rpm not found.
2024-06-13 07:11:40.902 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_import_interval not found.
2024-06-13 07:11:40.902 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_export_interval not found.
2024-06-13 07:11:40.902 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_power_factor not found.
2024-06-13 07:11:40.903 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_import_register not found.
2024-06-13 07:11:40.903 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_power_reactive_import not found.
2024-06-13 07:11:40.903 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_power_reactive_export not found.
2024-06-13 07:11:40.908 INFO (MainThread) [ocpp] Wallbox_1: send [3,"1754391931",{}]
2024-06-13 07:11:41.727 ERROR (MainThread) [custom_components.hacs] <QueueManager> No manifest.json file found 'custom_components/None/manifest.json'
2024-06-13 07:11:42.776 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open inside the event loop by custom integration 'ocpp' at custom_components/ocpp/api.py, line 571: resp = await self.call(req) (offender: /usr/local/lib/python3.12/site-packages/ocpp/messages.py, line 164: with open(path, "r", encoding="utf-8-sig") as f:), please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 672, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 639, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1988, in _run_once
[... 267 lines cut ...]
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 238, in start
    await self.route_message(message)
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 261, in route_message
    await self._handle_call(msg)
  File "/config/custom_components/ocpp/api.py", line 1005, in _handle_call
    await self.hass.async_create_task(super()._handle_call(msg))

2024-06-13 07:11:47.163 INFO (MainThread) [ocpp] Wallbox_1: send [3,"1738601311",{"currentTime":"2024-06-13T05:11:47Z","interval":3600,"status":"Accepted"}]
2024-06-13 07:11:47.166 INFO (MainThread) [ocpp] Wallbox_1: receive message [3,"69a8d277-a3c1-4444-8067-44903f0bff4d",{"status": "Accepted"}]
2024-06-13 07:11:48.175 INFO (MainThread) [ocpp] Wallbox_1: receive message [2,"604781483","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Available","timestamp": "2024-06-13T05:11:47Z"}]
2024-06-13 07:11:48.177 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_export_register not found.
2024-06-13 07:11:48.180 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_rpm not found.
2024-06-13 07:11:48.182 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_import_interval not found.
2024-06-13 07:11:48.183 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_energy_reactive_export_interval not found.
2024-06-13 07:11:48.183 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_power_factor not found.
[... cut ...]
2024-06-13 07:11:48.188 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_power_reactive_import not found.
2024-06-13 07:11:48.188 WARNING (MainThread) [homeassistant.helpers.entity_component] Forced update failed. Entity sensor.charger_1_power_reactive_export not found.
2024-06-13 07:11:48.202 INFO (MainThread) [ocpp] Wallbox_1: send [3,"604781483",{}]

**Desktop 
 - OS: Windows 10
 - Browser Chrome
 - Version  125.0.6422.113 (Oficiální sestavení) (64bitový)

VendisFronek, 2024-06-15:
Adding information:

EV charger:
[image]

sandersr, 2024-06-18:
Multiple chargers are not currently supported :(

puhycz, 2024-08-12:
I have same problem, plaese this multiple chargers feature will be available sometime in the future?

anon145, 2024-09-05:
same situation here, integration worked so well i bought another charger!
much work involved to add this capability?

lightheaded, 2024-10-20:
I am in process of installing chargers at home, and installed the first charger yesterday. Was happy to see that it mostly works. Today I plan on installing the second charger, but now discovered this topic 😅 

Has anyone looked into what it would take to get two chargers working? @lbbrhzn do you know if there are limitations for supporting multiple chargers or was just not considered when implementing as you probably just had one charger to work with?

lbbrhzn, 2024-10-20:
Hi Tom,There is a branch for supporting multiple chargers using a single server instance, but it still needs some work.Meanwhile, you can just install one server for each charger. Just use a different port for each one and it will work fine.Verstuurd vanaf mijn iPhone

npinguin, 2024-10-20:
Multiple chargers of the sale brand does not work… at least not for Wallbox even if you use différent ports

rbb-dev, 2024-11-24:
Hi.
It doesn't seems to be working for the set_charge_rate service, as both services are registered under opp.set_charge_rate

altserg, 2025-01-07:
see the solution #1366

drc38, 2025-01-31:
You can test https://github.com/lbbrhzn/ocpp/pull/1498

closed 2025-04-09 by github-actions[bot]