lbbrhzn/ocpp #1216: Entities not registering for measurands sent by Etrel Inch in v0.5.7

mazzy89, 2024-06-23:
## Version of the custom_component

v0.5.7

## Configuration

```yaml
Add your logs here.
```

```json
      {
        "data": {
          "host": "0.0.0.0",
          "port": 9000,
          "ssl": false,
          "ssl_certfile_path": "/config/fullchain.pem",
          "ssl_keyfile_path": "/config/privkey.pem",
          "csid": "central",
          "cpid": "charger",
          "max_current": 16,
          "monitored_variables": "Voltage.L3",
          "meter_interval": 60,
          "idle_interval": 900,
          "websocket_close_timeout": 10,
          "websocket_ping_tries": 2,
          "websocket_ping_interval": 20,
          "websocket_ping_timeout": 20,
          "skip_schema_validation": true,
          "force_smart_charging": false
        },
        "disabled_by": null,
        "domain": "ocpp",
        "entry_id": "b19eaa6852197733da1e9b7222af05b1",
        "minor_version": 1,
        "options": {},
        "pref_disable_new_entities": false,
        "pref_disable_polling": false,
        "source": "user",
        "title": "central",
        "unique_id": null,
        "version": 1
      }
```

## Describe the bug

The last versions after v0.5.1 messed up quite badly with the Etrel Inch charger and now many measurands are not reported anymore despite they are sent by the charger.

I have set it up the OCPP integration with the following measurands 

```txt
Current.Import,Energy.Active.Import.Register,Frequency,Power.Active.Import,Power.Factor,Temperature,Voltage
```

because these are the measurands that they are sent by the charger to the central as the log reports

```txt
[32m2024-06-23 21:30:25.858 INFO (MainThread) [ocpp] charger: receive message [2,"3","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-06-23T19:29:59.433Z","sampledValue":[{"value":"0","context":"Sample.Clock","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"0","context":"Sample.Clock","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"0","context":"Sample.Clock","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"733.255884","context":"Sample.Clock","measurand":"Energy.Active.Import.Register","unit":"kWh"},{"value":"50.02","context":"Sample.Clock","measurand":"Frequency"},{"value":"0","context":"Sample.Clock","measurand":"Power.Active.Import","phase":"L1","unit":"kW"},{"value":"0","context":"Sample.Clock","measurand":"Power.Active.Import","phase":"L2","unit":"kW"},{"value":"0","context":"Sample.Clock","measurand":"Power.Active.Import","phase":"L3","unit":"kW"},{"value":"0","context":"Sample.Clock","measurand":"Power.Active.Import","unit":"kW"},{"value":"-1","context":"Sample.Clock","measurand":"Power.Factor"},{"value":"27","context":"Sample.Clock","measurand":"Temperature","unit":"Celsius"},{"value":"238.21","context":"Sample.Clock","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"237.95","context":"Sample.Clock","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"237.46","context":"Sample.Clock","measurand":"Voltage","phase":"L3","unit":"V"}]}]}][0m
```

however many entities are not registered

[image]

## Debug log

```text

Common labels: {"container_name":"homeassistant","job":"systemd-journal","nodename":"homeassistant","syslog_identifier":"homeassistant","unit":"docker.service"}
[... 285 lines cut ...]
2024-06-23 21:22:27.067	2024-06-23 21:22:27.066 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=2.0 ms
2024-06-23 21:22:07.064	2024-06-23 21:22:07.063 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=2.0 ms
2024-06-23 21:21:47.058	2024-06-23 21:21:47.057 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=1.0 ms, pong=3.0 ms
2024-06-23 21:21:27.054	2024-06-23 21:21:27.052 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=2.0 ms
2024-06-23 21:21:07.050	2024-06-23 21:21:07.049 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=2.0 ms
2024-06-23 21:20:47.047	2024-06-23 21:20:47.045 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=2.0 ms, pong=2.0 ms
2024-06-23 21:20:27.043	2024-06-23 21:20:27.040 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=1.0 ms, pong=624.0 ms
2024-06-23 21:20:06.415	2024-06-23 21:20:06.413 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=830.0 ms
2024-06-23 21:20:00.159	2024-06-23 21:20:00.124 INFO (MainThread) [ocpp] charger: send [3,"7",{}]
2024-06-23 21:20:00.106	2024-06-23 21:20:00.067 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Voltage, phase_info: {'unit': 'V', 'L1': 238.49, 'L2': 240.06, 'L3': 238.47} value: 239.006667 unit :V
2024-06-23 21:20:00.086	2024-06-23 21:20:00.067 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Power.Active.Import, phase_info: {'unit': 'kW', 'L1': 0.0, 'L2': 0.0, 'L3': 0.0} value: 0.000000 unit :kW
2024-06-23 21:20:00.074	2024-06-23 21:20:00.066 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Import, phase_info: {'unit': 'A', 'L1': 0.0, 'L2': 0.0, 'L3': 0.0} value: 0.000000 unit :A
2024-06-23 21:20:00.038	2024-06-23 21:20:00.031 INFO (MainThread) [ocpp] charger: receive message [2,"7","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-06-23T19:19:59.431Z","sampledValue":[{"value":"0","context":"Sample.Clock","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"0","context":"Sample.Clock","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"0","context":"Sample.Clock","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"733.255884","context":"Sample.Clock","measurand":"Energy.Active.Import.Register","unit":"kWh"},{"value":"50","context":"Sample.Clock","measurand":"Frequency"},{"value":"0","context":"Sample.Clock","measurand":"Power.Active.Import","phase":"L1","unit":"kW"},{"value":"0","context":"Sample.Clock","measurand":"Power.Active.Import","phase":"L2","unit":"kW"},{"value":"0","context":"Sample.Clock","measurand":"Power.Active.Import","phase":"L3","unit":"kW"},{"value":"0","context":"Sample.Clock","measurand":"Power.Active.Import","unit":"kW"},{"value":"-0.338","context":"Sample.Clock","measurand":"Power.Factor"},{"value":"28","context":"Sample.Clock","measurand":"Temperature","unit":"Celsius"},{"value":"238.49","context":"Sample.Clock","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"240.06","context":"Sample.Clock","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"238.47","context":"Sample.Clock","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
2024-06-23 21:19:45.583	2024-06-23 21:19:45.581 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=1.0 ms, pong=4.0 ms
2024-06-23 21:19:25.577	2024-06-23 21:19:25.576 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=1.0 ms, pong=4.0 ms
2024-06-23 21:19:05.571	2024-06-23 21:19:05.570 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=12.0 ms
2024-06-23 21:18:45.557	2024-06-23 21:18:45.555 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=1.0 ms, pong=4.0 ms
2024-06-23 21:18:25.551	2024-06-23 21:18:25.549 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=1.0 ms, pong=2.0 ms
2024-06-23 21:18:05.546	2024-06-23 21:18:05.545 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=4.0 ms
2024-06-23 21:17:45.542	2024-06-23 21:17:45.540 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=1.0 ms, pong=3.0 ms
2024-06-23 21:17:25.537	2024-06-23 21:17:25.535 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=1.0 ms, pong=4.0 ms
2024-06-23 21:17:05.531	2024-06-23 21:17:05.529 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=4.0 ms
2024-06-23 21:16:45.526	2024-06-23 21:16:45.524 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=1.0 ms, pong=2.0 ms

```

mazzy89, 2024-06-23:
I confirm that downgrading to v0.5.1 works. Then the issue must be somewhere between v0.5.2 and v0.5.7. Looking carefully at the releases changelog the only culprit could be https://github.com/lbbrhzn/ocpp/releases/tag/v0.5.2 with this change https://github.com/lbbrhzn/ocpp/pull/1101

drc38, 2024-06-24:
Hi @mazzy89, please upload section of the log where each measurand is tested for acceptance by the charger.

mazzy89, 2024-06-24:
Thanks @drc38 let me test with the newer version

closed 2024-09-01 by github-actions[bot]