lbbrhzn/ocpp #1245: Unable to stop charging when writing 0 to number.charger_maximum_current

lodesmets, 2024-07-09:
## Version of the custom_component
V0.5.7

## Configuration

Setup through UI

## Describe the bug

I have a Wallbox Commander 2, Usually charge speed is set to 6 amps / phase. But when cooking, I stop charging by writing a 0 to number.charger_maximum_current. And afterwards, I write a 6 to number.charger_maximum_current. Often this works, but sometimes it fails. 

I have no idea why (because there are no errors). See logs below

I also tried using the service service: ocpp.set_charge_rate. But no difference

## Debug log

```
2024-07-09 19:49:58.012 INFO (MainThread) [ocpp] charger: receive message [2,"119362714","MeterValues",{"meterValue": [{"timestamp": "2024-07-09T17:49:57Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Export","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Export","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Export","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "5.6"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "5.5"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "5.5"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "6"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Export.Interval","unit": "Wh","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Export.Register","unit": "Wh","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Interval","unit": "Wh","value": "65.0"} ]} ],"connectorId": 1,"transactionId": 1720540578}]
2024-07-09 19:49:58.022 INFO (MainThread) [ocpp] charger: send [3,"119362714",{}]
2024-07-09 19:50:08.256 INFO (MainThread) [ocpp] charger: send [2,"6087ffd3-2e08-49b8-8127-4c7774f65594","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2024-07-09 19:50:08.313 INFO (MainThread) [ocpp] charger: receive message [3,"6087ffd3-2e08-49b8-8127-4c7774f65594",{"unknownKey": [],"configurationKey": [{"readonly": true,"key": "ChargingScheduleAllowedChargingRateUnit","value": "Current"} ]}]
2024-07-09 19:50:08.314 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current
2024-07-09 19:50:08.314 INFO (MainThread) [custom_components.ocpp] Charger supports setting the following units: Current
2024-07-09 19:50:08.314 INFO (MainThread) [custom_components.ocpp] If more than one unit supported default unit is Amps
2024-07-09 19:50:08.315 INFO (MainThread) [ocpp] charger: send [2,"87270e2a-d75f-4974-a2bd-569fc699cfc3","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2024-07-09 19:50:08.371 INFO (MainThread) [ocpp] charger: receive message [3,"87270e2a-d75f-4974-a2bd-569fc699cfc3",{"unknownKey": [],"configurationKey": [{"readonly": true,"key": "ChargeProfileMaxStackLevel","value": "1"} ]}]
2024-07-09 19:50:08.372 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 1
2024-07-09 19:50:08.373 INFO (MainThread) [ocpp] charger: send [2,"9cdb81cc-53a6-4963-933a-9bb2edf7352a","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":1,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6.0}]}}}]
2024-07-09 19:50:08.453 INFO (MainThread) [ocpp] charger: receive message [3,"9cdb81cc-53a6-4963-933a-9bb2edf7352a",{"status": "Rejected"}]
2024-07-09 19:50:08.453 DEBUG (MainThread) [custom_components.ocpp] ChargePointMaxProfile is not supported by this charger, trying TxDefaultProfile instead...
2024-07-09 19:50:08.454 INFO (MainThread) [ocpp] charger: send [2,"9567c6ae-5919-4ec8-98c8-9876684c381a","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":0,"chargingProfileKind":"Relative","chargingProfilePurpose":"TxDefaultProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6.0}]}}}]
2024-07-09 19:50:08.670 INFO (MainThread) [ocpp] charger: receive message [3,"9567c6ae-5919-4ec8-98c8-9876684c381a",{"status": "Accepted"}]
2024-07-09 19:50:16.400 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=17.0 ms
2024-07-09 19:50:30.695 INFO (MainThread) [ocpp] charger: send [2,"ac0f0630-96bd-4c9a-82ad-c7138be4679d","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2024-07-09 19:50:30.731 INFO (MainThread) [ocpp] charger: receive message [3,"ac0f0630-96bd-4c9a-82ad-c7138be4679d",{"unknownKey": [],"configurationKey": [{"readonly": true,"key": "ChargingScheduleAllowedChargingRateUnit","value": "Current"} ]}]
2024-07-09 19:50:30.732 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current
2024-07-09 19:50:30.732 INFO (MainThread) [custom_components.ocpp] Charger supports setting the following units: Current
2024-07-09 19:50:30.732 INFO (MainThread) [custom_components.ocpp] If more than one unit supported default unit is Amps
2024-07-09 19:50:30.733 INFO (MainThread) [ocpp] charger: send [2,"b32e6899-0687-456f-89ba-a009787367fd","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2024-07-09 19:50:30.760 INFO (MainThread) [ocpp] charger: receive message [3,"b32e6899-0687-456f-89ba-a009787367fd",{"unknownKey": [],"configurationKey": [{"readonly": true,"key": "ChargeProfileMaxStackLevel","value": "1"} ]}]
2024-07-09 19:50:30.761 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 1
2024-07-09 19:50:30.762 INFO (MainThread) [ocpp] charger: send [2,"ba125e50-eb9c-4a55-89d6-e89ad419f841","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":1,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":15.0}]}}}]
2024-07-09 19:50:30.815 INFO (MainThread) [ocpp] charger: receive message [3,"ba125e50-eb9c-4a55-89d6-e89ad419f841",{"status": "Rejected"}]
2024-07-09 19:50:30.815 DEBUG (MainThread) [custom_components.ocpp] ChargePointMaxProfile is not supported by this charger, trying TxDefaultProfile instead...
2024-07-09 19:50:30.816 INFO (MainThread) [ocpp] charger: send [2,"3e0b349c-85e0-4fea-ad17-8468bbca9491","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":0,"chargingProfileKind":"Relative","chargingProfilePurpose":"TxDefaultProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":15.0}]}}}]
2024-07-09 19:50:30.905 INFO (MainThread) [ocpp] charger: receive message [3,"3e0b349c-85e0-4fea-ad17-8468bbca9491",{"status": "Accepted"}]
2024-07-09 19:50:36.410 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=8.0 ms
2024-07-09 19:50:52.865 INFO (MainThread) [ocpp] charger: send [2,"4bc15dea-5b5f-4c7a-a4be-54633a2aeae4","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2024-07-09 19:50:52.895 INFO (MainThread) [ocpp] charger: receive message [3,"4bc15dea-5b5f-4c7a-a4be-54633a2aeae4",{"unknownKey": [],"configurationKey": [{"readonly": true,"key": "ChargingScheduleAllowedChargingRateUnit","value": "Current"} ]}]
2024-07-09 19:50:52.895 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current
2024-07-09 19:50:52.896 INFO (MainThread) [custom_components.ocpp] Charger supports setting the following units: Current
2024-07-09 19:50:52.896 INFO (MainThread) [custom_components.ocpp] If more than one unit supported default unit is Amps
[... cut ...] ,"transactionId": 1720540578}]
2024-07-09 19:50:57.911 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Export, phase_info: {'unit': 'A', 'L1': 0.0, 'L2': 0.0, 'L3': 0.0} value: 0.000000 unit :A
2024-07-09 19:50:57.911 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Import, phase_info: {'unit': 'A', 'L1': 5.6, 'L2': 5.6, 'L3': 5.5} value: 16.700000 unit :A
2024-07-09 19:50:57.918 INFO (MainThread) [ocpp] charger: send [3,"311142833",{}]

```

lodesmets, 2024-08-09:
Any ideas/ suggestions?

drc38, 2024-08-09:
Suggest you ask your charger manufacturer, as   the ocpp messages to set current to 0 are accepted.

lodesmets, 2024-08-21:
I'm pretty sure the charger supports it, because sometimes it works:

```
2024-08-21 21:36:09.267 INFO (MainThread) [ocpp] charger: send [2,"eb1ffbee-adbd-4db9-b942-969b90d80f2b","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":1,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6.0}]}}}]
2024-08-21 21:36:09.365 INFO (MainThread) [ocpp] charger: receive message [3,"eb1ffbee-adbd-4db9-b942-969b90d80f2b",{"status": "Rejected"}]
2024-08-21 21:36:09.366 DEBUG (MainThread) [custom_components.ocpp] ChargePointMaxProfile is not supported by this charger, trying TxDefaultProfile instead...
2024-08-21 21:36:09.367 INFO (MainThread) [ocpp] charger: send [2,"4a913ce3-071b-486e-9ad1-59df07c1c4d7","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":0,"chargingProfileKind":"Relative","chargingProfilePurpose":"TxDefaultProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6.0}]}}}]
2024-08-21 21:36:09.538 INFO (MainThread) [ocpp] charger: receive message [3,"4a913ce3-071b-486e-9ad1-59df07c1c4d7",{"status": "Accepted"}]
2024-08-21 21:36:20.339 INFO (MainThread) [ocpp] charger: receive message [2,"11102802","MeterValues",{"meterValue": [{"timestamp": "2024-08-21T19:36:19Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Export","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Export","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Export","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "5.6"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "1.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "6"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Export.Interval","unit": "Wh","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Export.Register","unit": "Wh","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Interval","unit": "Wh","value": "2.5"} ]} ],"connectorId": 1,"transactionId": 1724268742}]
2024-08-21 21:36:20.342 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Export, phase_info: {'unit': 'A', 'L1': 0.0, 'L2': 0.0, 'L3': 0.0} value: 0.000000 unit :A
2024-08-21 21:36:20.342 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Import, phase_info: {'unit': 'A', 'L1': 5.6, 'L2': 1.4, 'L3': 0.0} value: 7.000000 unit :A
2024-08-21 21:36:20.349 INFO (MainThread) [ocpp] charger: send [3,"11102802",{}]
2024-08-21 21:36:20.792 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=6.0 ms
2024-08-21 21:36:40.797 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=3.0 ms
2024-08-21 21:37:00.801 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=3.0 ms
2024-08-21 21:37:05.959 INFO (MainThread) [ocpp] charger: send [2,"771760ab-7880-458a-a2c9-395859d3b200","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2024-08-21 21:37:05.980 INFO (MainThread) [ocpp] charger: receive message [3,"771760ab-7880-458a-a2c9-395859d3b200",{"unknownKey": [],"configurationKey": [{"readonly": true,"key": "ChargingScheduleAllowedChargingRateUnit","value": "Current"} ]}]
2024-08-21 21:37:05.980 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargingScheduleAllowedChargingRateUnit: Current
2024-08-21 21:37:05.980 INFO (MainThread) [custom_components.ocpp] Charger supports setting the following units: Current
2024-08-21 21:37:05.980 INFO (MainThread) [custom_components.ocpp] If more than one unit supported default unit is Amps
2024-08-21 21:37:05.981 INFO (MainThread) [ocpp] charger: send [2,"023ad47a-a162-44b4-a1de-c77bdd5da7cf","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2024-08-21 21:37:06.009 INFO (MainThread) [ocpp] charger: receive message [3,"023ad47a-a162-44b4-a1de-c77bdd5da7cf",{"unknownKey": [],"configurationKey": [{"readonly": true,"key": "ChargeProfileMaxStackLevel","value": "1"} ]}]
2024-08-21 21:37:06.010 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 1
2024-08-21 21:37:06.011 INFO (MainThread) [ocpp] charger: send [2,"57a68fcf-817c-4d21-a3d3-fd1249ba5950","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":1,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0.0}]}}}]
[... cut ...]
2024-08-21 21:37:20.486 DEBUG (MainThread) [custom_components.ocpp] process_phases: metric: Current.Import, phase_info: {'unit': 'A', 'L1': 0.0, 'L2': 0.0, 'L3': 0.0} value: 0.000000 unit :A
2024-08-21 21:37:20.493 INFO (MainThread) [ocpp] charger: send [3,"1247518290",{}]
2024-08-21 21:37:20.806 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=3.0 ms
2024-08-21 21:37:40.810 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'charger': ping=0.0 ms, pong=3.0 ms

```