lbbrhzn/ocpp #1248: Legrand GreenUp blocked at 16 A

fredlamour, 2024-07-10:
**Describe the bug**
It is set to 32 A in the slider but the EV is mentionning 16 A as target current.

**To Reproduce**
Steps to reproduce the behavior:
1. Set current to 32A
2. Check on EV web interface and it is showing the order limit is 16 A
3. Scroll down the current under 16A and the EV is adjusting without being possible to go over 16A

**Expected behavior**
Whatever the current is set over 16A, the EV is getting the order to limit at 16A.
Whatever the currect is set under 16A, the EV is adjusting the current to the selected value.

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser : Chrome
 - Version : 126

Tallarean, 2024-08-01:
@fredlamour While I don't have a solution just yet, I think I have a hunch as to what is the problem:
Am I correct in assuming the charger is located in france? 
If so, the charger would be supplied with 230V. 
This in turn would mean that, to achieve the mentioned 7,4 kW that the charger seems to be rated at, 32A would have to be supplied over one single phase. Most cars do not accept this kind of amperage over one phase. Also, in some european countries, there are law prohibiting heavy unbalanced loads. So please check if your car actually has an onboard charging circuit that accepts 32A over one phase.

TLDR: What car are you trying to charge and have you succesfully charged that car at 32A without using the OCPP integration (for example using an app or similar)?

fredlamour, 2024-08-02:
Hi @Tallarean and thank you for your time spend to answer.
I am in France and the power supplied is 230V. As per the local law, this installation has be done by a professional to allow 32A on a dedicated circuit for EV charger. It is so possible to reach the 7.4 kW to charge.
Before going to OCPP, I was using the charger with the web interface and during it was working at 32A to have the 7,4 kW on charge. And before I had installed the communication kit to have the web server, I was using it with the embedded bluetooth system at 32A without issue too.
My car is Vinfast VF8 and is accepting 7.4 kW on the AC charger embedded and so I have been able to reach that power.
That limitation of 16A has only appeared when I switched to OCPP even the server is configured with 32A as limit, showing 32A on the HomeAssistant panel while logs are showing it is sending 16A (I would insist, the log is saying it is sending 16A and not receiving 16A).

If you need more details, feel free. the quickest I can solve that, the sooner I will be happy ...

Fredeeric

fredlamour, 2024-08-02:
[image]
this is from station history file, you can see I have charged at more than 7 kWh average and it was before OCPP

Tallarean, 2024-08-02:
Okay, that eliminates most possible issues with the car and charger combination. 

Let's try the easy stuff first: Could you post the part of the log from just before changing the power to until after the change?
You can get to that by navigation to Settings -> System -> Log
I suggest you open that, then change the power of the charger to a lower value, wait a moment for the change and then try to switch to 32A. Then update the log. With that we might get a better idea what happens.

For even more info on debugging, see this section: [Debugging](https://home-assistant-ocpp.readthedocs.io/en/latest/debugging.html)

fredlamour, 2024-08-03:
@Tallarean , I really appreciate your assistance and advices.

Here are the logs.
I have restarted first my HA and the charger then : 
1. without the car connected, I have moved the slider from 32 to 16 A and back to back with some time between each modification.
2. I have enable car charge on HA, setting is set to 32A
3. Connected the car. It start charging increasing to 16A (15.7 exactly)
4. I have reduced the current on HA from 32A to 16A, no change
5. I have tried to reduce again in different steps and it was changing the current the car was receiving (so under 16A it is responding correctly). By error I moved under 8A and that maked the charger into security. I have been obliged to stop, unplug, bypass the error on the charger and restarted to charge again still at 32A on HA but 16A limited on the charger.
[home-assistant_2024-08-03T16-25-25.776Z.log](https://github.com/user-attachments/files/16482260/home-assistant_2024-08-03T16-25-25.776Z.log)

Thanks again for your help.

Tallarean, 2024-08-07:
Okay, the way I see this is that setting the charger to 32A without the car charging seems to work fine, see from the log:
```
2024-08-03 18:15:01.050 INFO (MainThread) [ocpp] GreenUp: send [2,"0c019791-eb9e-4074-9ac1-0191b4edc58d","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":100,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":32.0}]}}}]
2024-08-03 18:15:01.082 INFO (MainThread) [ocpp] GreenUp: receive message [3,"0c019791-eb9e-4074-9ac1-0191b4edc58d",{"status":"Accepted"}]
2024-08-03 18:15:03.501 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'GreenUp': ping=0.0 ms, pong=112.0 ms
2024-08-03 18:15:23.508 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'GreenUp': ping=0.0 ms, pong=6.0 ms
```
After that there seems to be some time (I assume to plug in the car) where nothing special happens.
At 18:16:42 either you as the user or something else in the system seems to change the power limit back down to 17A:
```
2024-08-03 18:16:42.153 INFO (MainThread) [ocpp] GreenUp: send [2,"6fb88f3d-8d1d-4566-be65-fc0cb979e55d","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2024-08-03 18:16:42.216 INFO (MainThread) [ocpp] GreenUp: receive message [3,"6fb88f3d-8d1d-4566-be65-fc0cb979e55d",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"100"}]}]
2024-08-03 18:16:42.216 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for ChargeProfileMaxStackLevel: 100
2024-08-03 18:16:42.216 INFO (MainThread) [ocpp] GreenUp: send [2,"fee481d6-e615-41c6-93ca-1660b312e605","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":100,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":17.0}]}}}]
2024-08-03 18:16:42.285 INFO (MainThread) [ocpp] GreenUp: receive message [3,"fee481d6-e615-41c6-93ca-1660b312e605",{"status":"Accepted"}]
2024-08-03 18:16:44.247 DEBUG (MainThread) [custom_components.ocpp] Connection latency from 'central' to 'GreenUp': ping=0.0 ms, pong=9.0 ms
```
And after that, 6 seconds later, down to 11A.
(I'm going to leave that part of the log out. Just look at the log right after the above part)

7 seconds later (18:16:55) an unknown error is received from the charger. The error seems to stem from an incorrectly formated respone from the charger:
```
2024-08-03 18:16:55.669 INFO (MainThread) [ocpp] GreenUp: receive message [2,"28240181849166013077","StatusNotification",{"connectorId":0,"errorCode":"Error","status":"Faulted","timestamp":"2024-08-03T16:16:55Z"}]
2024-08-03 18:16:55.669 ERROR (MainThread) [ocpp] Error while handling request '<Call - unique_id=28240181849166013077, action=StatusNotification, payload={'connectorId': 0, 'errorCode': 'Error', 'status': 'Faulted', 'timestamp': '2024-08-03T16:16:55Z'}>'
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ocpp/messages.py", line 226, in validate_payload
    validator.validate(message.payload)
  File "/usr/local/lib/python3.12/site-packages/jsonschema/validators.py", line 451, in validate
    raise error
jsonschema.exceptions.ValidationError: 'Error' is not one of ['ConnectorLockFailure', 'EVCommunicationError', 'GroundFailure', 'HighTemperature', 'InternalError', 'LocalListConflict', 'NoError', 'OtherError', 'OverCurrentFailure', 'PowerMeterFailure', 'PowerSwitchFailure', 'ReaderFailure', 'ResetFailure', 'UnderVoltage', 'OverVoltage', 'WeakSignal']

Failed validating 'enum' in schema['properties']['errorCode']:
    {'type': 'string',
     'additionalProperties': False,
     'enum': ['ConnectorLockFailure',
              'EVCommunicationError',
              'GroundFailure',
              'HighTemperature',
              'InternalError',
              'LocalListConflict',
              'NoError',
              'OtherError',
              'OverCurrentFailure',
              'PowerMeterFailure',
              'PowerSwitchFailure',
              'ReaderFailure',
              'ResetFailure',
              'UnderVoltage',
              'OverVoltage',
              'WeakSignal']}

On instance['errorCode']:
    'Error'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 261, in route_message
    await self._handle_call(msg)
  File "/config/custom_components/ocpp/api.py", line 1008, in _handle_call
    await super()._handle_call(msg)
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 289, in _handle_call
    validate_payload(msg, self._ocpp_version)
  File "/usr/local/lib/python3.12/site-packages/ocpp/messages.py", line 244, in validate_payload
    raise FormatViolationError(
ocpp.exceptions.FormatViolationError: FormatViolationError: Payload for Action is syntactically incorrect or structure for Action,  {'cause': "Payload '{'connectorId': 0, 'errorCode': 'Error', 'status': 'Faulted', 'timestamp': '2024-08-03T16:16:55Z'}' for action 'StatusNotification' is not valid: 'Error' is not one of ['ConnectorLockFailure', 'EVCommunicationError', 'GroundFailure', 'HighTemperature', 'InternalError', 'LocalListConflict', 'NoError', 'OtherErro [... cut ...]
The charger seems to run into an OverCurrentFailure, specifically an overload on mode 3.
Based on the information on this website ([Energuide.be/charging-modes](https://www.energuide.be/en/questions-answers/mode-1-2-3-and-4-what-do-the-different-charging-modes-for-electric-cars-mean/14019/)) this seems to somehow be connected to smart charging and/or the car trying to change the accepted current?
I have to say, this is a bit outside of my expertise.

@fredlamour Do you use smart charging? When you set up the integration, did you toggle "smart charging" on?

fredlamour, 2024-09-11:
Thanks @Tallarean for the analysis of the logs.

First here are the details from the config files : 
`     {
        "created_at": "2024-08-27T16:19:30.905540+00:00",
        "data": {
          "host": "0.0.0.0",
          "port": 9000,
          "ssl": false,
          "ssl_certfile_path": "/config/fullchain.pem",
          "ssl_keyfile_path": "/config/privkey.pem",
          "csid": "Centrale",
          "cpid": "GreenUp",
          "max_current": 32,
          "monitored_variables": "Current.Import,Energy.Active.Import.Interval,Energy.Active.Import.Register,Power.Active.Import,Voltage",
          "meter_interval": 60,
          "idle_interval": 900,
          "websocket_close_timeout": 10,
          "websocket_ping_tries": 20,
          "websocket_ping_interval": 20,
          "websocket_ping_timeout": 20,
          "skip_schema_validation": true,
          "force_smart_charging": false
        },
        "disabled_by": null,
        "domain": "ocpp",
        "entry_id": "01J6AA57ES0YYV24KA6SJYVZJG",
        "minor_version": 1,
        "modified_at": "2024-08-27T16:24:46.439064+00:00",
        "options": {},
        "pref_disable_new_entities": false,
        "pref_disable_polling": false,
        "source": "user",
        "title": "Centrale",
        "unique_id": null,
        "version": 1
      }`

I hope this is what was requested.

To answer few of the questions, I would confirm that changing the current from the interface is working correctly all the time whatever the charge is in progress or not. I can select all the values from 1 A to 32 A (32A is the limit of the charger and the value also set in the plugin).
The issue is that whatever is selected the value selected over 16A, the plugin is not changing it with a maximum of 16A.

I hope that will help in the troubleshooting.

f3peter, 2024-09-20:
Hi, 
I have same problem, but different charger. My charger is an INJET Sonic 7,5kW 1phase 32Amps.
I have make a lot of tests: 

1.) I have used only OCPP with 32A initial settings, without charger application:
When I started the charging with the OCPP, then the maximum charging current is 16A. When I change the charging current via OCPP to 10A, then the charger accepted this, and the real charging current go to 10A. BUT when I change the charging current via OCPP to 20A or 25A or 32A, then the charger was accepted always, but the real charging current was maximized in 16A. 

2.) First set 32A in charger application, then set 32A in OCPP, then start the charging via OCPP: 
Absolutley same as in first. 

3.) Set the OCPP 25A, and set the charger application 32A, then start the charging via charger app: 
The charging method was starting with 25A charging current. When I set the current with OCPP below 25A, then the charger was accpted, and the charging current was changed. But when I set above 25A the current with OCPP, then it was accepted, but the real charging current was maximized in 25A. 
Here is a part of log: 
```
2024-09-18 22:07:22.256 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [3,"120072256",{}]
2024-09-18 22:08:22.230 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: receive message [2,"120082257","MeterValues",{"connectorId":1,"transactionId":1726685060,"meterValue":[{"timestamp":"2024-09-18T20:08:22.333Z","sampledValue":[{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"374000"},{"measurand":"Power.Active.Import","unit":"kW","value":"4.5"},{"measurand":"Temperature","unit":"Celsius","value":"33"},{"measurand":"Voltage","phase":"L1-N","unit":"V","value":"230.3"},{"measurand":"Current.Import","phase":"N","unit":"A","value":"19.8"}]}]}]
2024-09-18 22:08:22.240 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [3,"120082257",{}]
2024-09-18 22:09:22.167 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: receive message [2,"120092158","MeterValues",{"connectorId":1,"transactionId":1726685060,"meterValue":[{"timestamp":"2024-09-18T20:09:21.333Z","sampledValue":[{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"374070"},{"measurand":"Power.Active.Import","unit":"kW","value":"4.5"},{"measurand":"Temperature","unit":"Celsius","value":"33"},{"measurand":"Voltage","phase":"L1-N","unit":"V","value":"230.3"},{"measurand":"Current.Import","phase":"N","unit":"A","value":"19.7"}]}]}]
2024-09-18 22:09:22.176 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [3,"120092158",{}]
2024-09-18 22:09:45.439 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [2,"3c465361-e198-4ebc-a76d-ec0aabacba89","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2024-09-18 22:09:45.616 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: receive message [3,"3c465361-e198-4ebc-a76d-ec0aabacba89",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","value":"Current","readonly":true}]}]
2024-09-18 22:09:45.619 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [2,"021cc26a-4bcb-4399-8f50-fc1cfbdc8121","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2024-09-18 22:09:45.672 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: receive message [3,"021cc26a-4bcb-4399-8f50-fc1cfbdc8121",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","value":"100","readonly":true}]}]
2024-09-18 22:09:45.682 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [2,"388c68e4-bfb7-4890-bd58-f5bec6e98bab","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":100,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":10.0}]}}}]
2024-09-18 22:09:45.845 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: receive message [3,"388c68e4-bfb7-4890-bd58-f5bec6e98bab",{"status":"Accepted"}]
2024-09-18 22:10:04.205 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: receive message [2,"120100459","Heartbeat",{}]
2024-09-18 22:10:04.207 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [3,"120100459",{"currentTime":"2024-09-18T20:10:04Z"}]
2024-09-18 22:10:22.191 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: receive message [2,"120102160","MeterValues",{"connectorId":1,"transactionId":1726685060,"meterValue":[{"timestamp":"2024-09-18T20:10:21.333Z","sampledValue":[{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"374120"},{"measurand":"Power.Active.Import","unit":"kW","value":"2.2"},{"measurand":"Temperature","unit":"Celsius","value":"33"},{"measurand":"Voltage","phase":"L1-N","unit":"V","value":"231.7"},{"measurand":"Current.Import","phase":"N","unit":"A","value":"9.5"}]}]}]
2024-09-18 22:10:22.196 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [3,"120102160",{}]
2024-09-18 22:11:22.170 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: receive message [2,"120112161","MeterValues",{"connectorId":1,"transactionId":1726685060,"meterValue":[{"timestamp":"2024-09-18T20:11:21.333Z","sampledValue":[{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"374160"},{"measurand":"Power.Active.Import","unit":"kW","value":"2.1"},{"measurand":"Temperature","unit":"Celsius","value":"33"},{"measurand":"Voltage","phase":"L1-N","unit":"V","value":"231.6"},{"measurand":"Current.Import","phase":"N","unit":"A","value":"9.4"}]}]}]
2024-09-18 22:11:22.179 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [3,"120112161",{}]
[... cut ...] 3"},{"measurand":"Voltage","phase":"L1-N","unit":"V","value":"231.6"},{"measurand":"Current.Import","phase":"N","unit":"A","value":"15.7"}]}]}]
2024-09-18 22:18:22.278 INFO (MainThread) [ocpp]  7xxxxxxxxxxxxx9: send [3,"120182270",{}]
```

4.) Set the OCPP 32A, and set the charger application 32A, then start the charging via charger app: 
only this case can I changed the charging current form 6A to 32A with OCPP. 

My next step is to send an email to the charger support team describing this bug.

fredlamour, 2024-09-23:
Hi everybody,

We can consider the issue solved. The manufacturer LEGRAND has published a new firmware for the wallbox and the IP connection kit that has solved the issue described there.

This version is [KitCom.V01.17.28] (https://assets.legrand.com/pim/AUTRE/KitCom.V01.17.28.zip)

May I suggest to add the wallbox to the compatibility list of the plugin with the condition of firmware mentionned ?

Thanks all for the help is trying the solve my issue.

closed 2024-09-23 by fredlamour