lbbrhzn/ocpp #1265: Payload for Action too long (because measureands are queried that are not in the list)

Tallarean, 2024-07-31:
**Describe the bug**
OCPP integration generates error in home assistant log. Measurands are updated but are then invalidated as soon as the error occurs, leading to unavailable values. Charge can not be started or stopped.

**To Reproduce**
1. Install OCPP 0.5.8
2. Configure using GUI with measurands "Power.Active.Import,Power.Offered,Current.Import,Current.Offered"
3. Reboot.
4. Entities will update after a while, then become unavailable quickly after. Log shows error (see below)

**Expected behavior**
Values should not become "unavailable" after update.

**Additional context**
The Log:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/ocpp/messages.py", line 226, in validate_payload
    validator.validate(message.payload)
  File "/usr/local/lib/python3.12/site-packages/jsonschema/validators.py", line 435, in validate
    raise error
jsonschema.exceptions.ValidationError: 'Current.Export.L1,Current.Export.L2,Current.Export.L3,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N' is too long

Failed validating 'maxLength' in schema['properties']['configurationKey']['items']['properties']['value']:
    {'maxLength': 500, 'type': 'string'}

On instance['configurationKey'][0]['value']:
    'Current.Export.L1,Current.Export.L2,Current.Export.L3,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 1023, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 468, in post_connect
    chgr_measurands = await self.get_configuration(key)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 890, in get_configuration
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 422, in call
    validate_payload(response, self._ocpp_version)
  File "/usr/local/lib/python3.12/site-packages/ocpp/messages.py", line 240, in validate_payload
    raise TypeConstraintViolationError(
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12),  {'cause': "'Current.Export.L1,Current.Export.L2,Current.Export.L3,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N' is too long", 'ocpp_message': <CallResult - unique_id=33442100-6423-4b2d-888e-d36289e18866, action=GetConfiguration, payload={'unknownKey': [], 'configurationKey': [{'readonly': False, 'key': 'MeterValuesSampledData', 'value': 'Current.Export.L1,Current.Export.L2,Current.Export.L3,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N'}]}>}
```

**I also tried:**
As per  #1150  I have edited the "monitored_variables" in /config/.storage/core.config_entries to only include the above mentioned measurands. But as the log shows, far more measureands seem to be called for than listed in this variable. For example Current.Export.L3 even though there are no Export measureands in the list.
I wanted to try editing the max length as stated in #1155 but I can't find where to edit the code:
```    
"chargeBoxSerialNumber": {
        "type": "string",
        "maxLength": 33
    },
```
(of course, in my case, that would probably be "ocpp_message" not "chargeBoxSerialNumber")

Does anyone have any idea how to work around this for now? I tried installing an old version (0.4.42) in the hope that the automatic detection of supported measurands that was added with 0.5. was the culprit. But that install did not work as i hoped.

Tallarean, 2024-08-02:
Okay, after digging a bit further here are my new findings:
1. The error seems to occur due to the length limit of the GetConfigurationRequest (see here: [GitHub:ocpp/GetConfiguration](https://github.com/mobilityhouse/ocpp/blob/master/ocpp/v16/schemas/GetConfiguration.json) ). It is limited to 50 characters. The request made is 551 characters long. 
2. Changing this manually requires either changing the request made by this integration (as far as I understand, I might be wrong) or changing the max accepted length in the python module schema description. (See link above)

I am going to try to change this manually in the definition for now to see if that is a temporary fix.

drc38, 2024-08-02:
Have a look at https://github.com/djiesr/ocpp which makes various changes to support incompatible ocpp1.6 firmware

drc38, 2024-08-02:
You can comment out line 468, and define accepted_measurands with the measurands your charger supports.

Tallarean, 2024-08-07:
Thanks @drc38 that workaround does indeed work. 
I did have to uncomment another line of code where the additional measureands were appended, but with a few changes to the code it now works.
I do wonder if I am the only user that has this problem. To me, this looks like something everyone should encounter, after all it's not like I put in too many measurands. Even the "default" measurands are too many for the string lenght limit, set by the ocpp Python Module (if I understand this correctly).
(Maybe I am the only one who updates a running system. "Never change a running system" and all.)

malaakso, 2024-08-19:
As far as I can see, the error is due to the charger reporting a configured set of measurands that exceed the length limit of 500 characters in the spec. Maybe that is the default setting for the charger. This exception could simply be caught, I don't see a reason why that would cause any issue.

drc38, 2024-08-19:
If caught another error would be thrown as the measurands reported are also not correct. See https://github.com/mobilityhouse/ocpp/blob/master/ocpp/v16/enums.py eg `current.import` vs `current.import.L1`

malaakso, 2024-08-20:
> If caught another error would be thrown as the measurands reported are also not correct. See https://github.com/mobilityhouse/ocpp/blob/master/ocpp/v16/enums.py eg `current.import` vs `current.import.L1`

Aren't the erroneous measurands simply reconfigured [here](https://github.com/lbbrhzn/ocpp/blob/main/custom_components/ocpp/api.py#L490), so afterwards the reports are fine?

drc38, 2024-08-23:
> If caught another error would be thrown as the measurands reported are also not correct. See https://github.com/mobilityhouse/ocpp/blob/master/ocpp/v16/enums.py eg `current.import` vs `current.import.L1`

Ignore this, I see in the spec it is valid to append the phase to the measurand.

drc38, 2024-09-10:
v0.5.12 supports manual configuration of the measurands which will then avoid this.