lbbrhzn/ocpp #1275: ABB Terra AC-W22-T-0 Bootloop

pvprodk, 2024-08-16:
**Describe the bug**
ABB Terra AC-W22-T-0 (listed as supported) keeps booting with these notifications showing up.
Controls are avaliable for a few seconds, until it reboots again.

**Screenshots**
[image]

**Versions etc.**
OCPP v0.5.8
HA 2024.8.1
Charger firmware: V1.8.21 (Newest)

**Debug logging from HA (Not really sure if I'm including enough?)**

Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/websockets/legacy/protocol.py", line 963, in transfer_data
message = await self.read_message()
^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/websockets/legacy/protocol.py", line 1033, in read_message
frame = await self.read_data_frame(max_size=self.max_size)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/websockets/legacy/protocol.py", line 1108, in read_data_frame
frame = await self.read_frame(max_size)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/websockets/legacy/protocol.py", line 1165, in read_frame
frame = await Frame.read(
^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/websockets/legacy/framing.py", line 68, in read
data = await reader(2)
^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/asyncio/streams.py", line 752, in readexactly
await self._wait_for_data('readexactly')
File "/usr/local/lib/python3.12/asyncio/streams.py", line 545, in _wait_for_data
await self._waiter
File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1003, in _read_ready__data_received
data = self._sock.recv(self.max_size)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ConnectionResetError: [Errno 104] Connection reset by peer
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/config/custom_components/ocpp/api.py", line 460, in post_connect
resp = await self.get_configuration(ckey.number_of_connectors.value)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/ocpp/api.py", line 890, in get_configuration
resp = await self.call(req)
^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 404, in call
await self._send(call.to_json())
File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 457, in _send
await self._connection.send(message)
File "/usr/local/lib/python3.12/site-packages/websockets/legacy/protocol.py", line 635, in send
await self.ensure_open()
File "/usr/local/lib/python3.12/site-packages/websockets/legacy/protocol.py", line 939, in ensure_open
raise self.connection_closed_exc()
websockets.exceptions.ConnectionClosedError: no close frame received or sent
2024-08-16 22:02:18.384 INFO (MainThread) [websockets.server] connection closed
2024-08-16 22:02:18.385 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-08-16 22:02:18.392 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-08-16 22:02:25.568 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-08-16 22:02:28.396 INFO (MainThread) [websockets.server] connection closed
2024-08-16 22:02:28.396 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-08-16 22:02:35.571 INFO (MainThread) [websockets.server] connection closed
2024-08-16 22:02:35.572 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-08-16 22:02:35.579 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-08-16 22:02:45.584 INFO (MainThread) [websockets.server] connection closed
2024-08-16 22:02:45.585 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-08-16 22:02:45.592 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-08-16 22:02:55.594 INFO (MainThread) [websockets.server] connection closed
2024-08-16 22:02:55.595 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-08-16 22:02:55.602 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
[... 44 lines cut ...]
2024-08-16 22:03:05.853 DEBUG (MainThread) [websockets.server] < TEXT '[2, "8454120", "SecurityEventNotification", {"t...-08-16T20:03:04.000Z"}]' [139 bytes]
2024-08-16 22:03:05.853 INFO (MainThread) [ocpp] TACW2242222T6786: receive message [2, "8454120", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-16T20:03:04.000Z"}]
2024-08-16 22:03:05.854 INFO (MainThread) [custom_components.ocpp] Security event notification received: SettingSystemTime at 2024-08-16T20:03:04.000Z [techinfo: ocppBoot]
2024-08-16 22:03:05.856 INFO (MainThread) [ocpp] TACW2242222T6786: send [3,"8454120",{}]
2024-08-16 22:03:05.856 DEBUG (MainThread) [websockets.server] > TEXT '[3,"8454120",{}]' [16 bytes]
2024-08-16 22:03:06.190 DEBUG (MainThread) [websockets.server] < TEXT '[3, "fc4e6531-383f-42aa-9ec8-bd3c2abe80b7", {"c..."}], "unknownKey": []}]' [228 bytes]
2024-08-16 22:03:06.191 INFO (MainThread) [ocpp] TACW2242222T6786: receive message [3, "fc4e6531-383f-42aa-9ec8-bd3c2abe80b7", {"configurationKey": [{"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}], "unknownKey": []}]
2024-08-16 22:03:06.193 DEBUG (MainThread) [custom_components.ocpp] Feature profiles returned: Profiles.CORE|FW|SMART|REM|AUTH
2024-08-16 22:03:06.195 INFO (MainThread) [ocpp] TACW2242222T6786: send [2,"f9adbd68-2fc5-41a7-aece-651e2c50ef8e","GetConfiguration",{"key":["NumberOfConnectors"]}]
2024-08-16 22:03:06.195 DEBUG (MainThread) [websockets.server] > TEXT '[2,"f9adbd68-2fc5-41a7-aece-651e2c50ef8e","GetC..."NumberOfConnectors"]}]' [92 bytes]
2024-08-16 22:03:06.513 DEBUG (MainThread) [websockets.server] < TEXT '[3, "f9adbd68-2fc5-41a7-aece-651e2c50ef8e", {"c..."}], "unknownKey": []}]' [148 bytes]
2024-08-16 22:03:06.514 INFO (MainThread) [ocpp] TACW2242222T6786: receive message [3, "f9adbd68-2fc5-41a7-aece-651e2c50ef8e", {"configurationKey": [{"key": "NumberOfConnectors", "readonly": true, "value": "1"}], "unknownKey": []}]
2024-08-16 22:03:06.516 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for NumberOfConnectors: 1
2024-08-16 22:03:06.517 INFO (MainThread) [ocpp] TACW2242222T6786: send [2,"ae5684bf-eab3-464f-9783-dbe99a4354ef","GetConfiguration",{"key":["HeartbeatInterval"]}]
2024-08-16 22:03:06.518 DEBUG (MainThread) [websockets.server] > TEXT '[2,"ae5684bf-eab3-464f-9783-dbe99a4354ef","GetC...["HeartbeatInterval"]}]' [91 bytes]
2024-08-16 22:03:06.836 DEBUG (MainThread) [websockets.server] < TEXT '[3, "ae5684bf-eab3-464f-9783-dbe99a4354ef", {"c..."}], "unknownKey": []}]' [151 bytes]
2024-08-16 22:03:06.837 INFO (MainThread) [ocpp] TACW2242222T6786: receive message [3, "ae5684bf-eab3-464f-9783-dbe99a4354ef", {"configurationKey": [{"key": "HeartbeatInterval", "readonly": false, "value": "3600"}], "unknownKey": []}]
2024-08-16 22:03:06.838 DEBUG (MainThread) [custom_components.ocpp] Get Configuration for HeartbeatInterval: 3600
2024-08-16 22:03:06.840 INFO (MainThread) [ocpp] TACW2242222T6786: send [2,"7f23a2ba-ff51-4cf9-9f24-5f90a42fff13","GetConfiguration",{"key":["SupportedFeatureProfiles"]}]
2024-08-16 22:03:06.840 DEBUG (MainThread) [websockets.server] > TEXT '[2,"7f23a2ba-ff51-4cf9-9f24-5f90a42fff13","GetC...rtedFeatureProfiles"]}]' [98 bytes]
2024-08-16 22:03:07.204 DEBUG (MainThread) [websockets.server] < TEXT '[3, "7f23a2ba-ff51-4cf9-9f24-5f90a42fff13", {"c..."}], "unknownKey": []}]' [228 bytes]
2024-08-16 22:03:07.205 INFO (MainThread) [ocpp] TACW2242222T6786: receive message [3, "7f23a2ba-ff51-4cf9-9f24-5f90a42fff13", {"configurationKey": [{"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}], "unknownKey": []}]
2024-08-16 22:03:07.207 DEBUG (MainThread) [custom_components.ocpp] Feature profiles returned: Profiles.CORE|FW|SMART|REM|AUTH
2024-08-16 22:03:07.209 INFO (MainThread) [ocpp] TACW2242222T6786: send [2,"916e21a0-bd4b-4899-a024-7cb80b2d1a1e","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2024-08-16 22:03:07.209 DEBUG (MainThread) [websockets.server] > TEXT '[2,"916e21a0-bd4b-4899-a024-7cb80b2d1a1e","GetC...erValuesSampledData"]}]' [96 bytes]

cpaeby, 2024-08-18:
I also have a lot of notification with my ABB Terra. I use this automation to cancel them but I would also be interested about a bug fix...

`alias: supprimer les notification OCPP
description: >-
  Description trouvée sur https://github.com/lbbrhzn/ocpp/discussions/938 qui
  supprime les notification de OCPP.
trigger:
  - platform: persistent_notification
    update_type:
      - added
    notification_id: ""
condition:
  - condition: template
    value_template: "{{ trigger.notification.title | lower == \"ocpp integration\" }}"
action:
  - delay:
      hours: 0
      minutes: 10
      seconds: 0
      milliseconds: 0
  - data:
      notification_id: "{{ trigger.notification.notification_id }}"
    action: persistent_notification.dismiss
mode: parallel
max: 10
`

Eppigt, 2024-08-23:
I have the same problem with my Terra AC W11 and Firmware 1.8.21
I downgraded to OCPP 0.5.6  and it works again without reboot loop

kriisa10, 2024-08-26:
I am having the same issues on my ABB Terra AC-W22-T-0 (also running the latest firmware 1.8.21). 
I can also confirm that the issue seems to disappear when downgrading to OCPP v0.5.7

rinigus, 2024-08-27:
Stumbled into the same issue when testing HA OCPP integration. Have the latest firmware 1.8.21 as well.

What happens is that the charger gets into the bootloop after issuing 

```
send [2,"91b546e6-de46-4ed5-b48b-6a0d7b38b991","GetConfiguration",{"key":["MeterValuesSampledData"]}]
```

That bootloop, I think, is caused by initial configuration of HA OCPP. When adding my charger, as a new user, I have accepted all the defaults. On the first boot, I had logs like these:

```
2024-08-25 13:36:49.032 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "21162b4c-07a6-4626-aee9-d208e3c6921d", {"configurationKey": [{"key": "NumberOfConnectors", "readonly": true, "value": "1"}], "unknownKey": []}]
2024-08-25 13:36:49.037 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"31e347d0-ccf2-4b54-9167-d71afd67421a","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2024-08-25 13:36:49.401 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "31e347d0-ccf2-4b54-9167-d71afd67421a", {"configurationKey": [{"key": "MeterValuesSampledData", "readonly": false, "value": "Current.Import,Energy.Active.Import.Register,Power.Active.Import,Voltage"}], "unknownKey": []}]                                                                                                                                                                                                       
2024-08-25 13:36:49.427 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"6e01d0ca-b29f-4dbd-a31c-46a7541e004d","GetConfiguration",{"key":["HeartbeatInterval"]}]
2024-08-25 13:36:49.723 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "6e01d0ca-b29f-4dbd-a31c-46a7541e004d", {"configurationKey": [{"key": "HeartbeatInterval", "readonly": false, "value": "3600"}], "unknownKey": []}]
2024-08-25 13:36:49.729 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"16422111-4777-41b1-b9a3-948565e18ab7","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export"}]
2024-08-25 13:36:50.040 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "16422111-4777-41b1-b9a3-948565e18ab7", {"status": "Accepted"}]
2024-08-25 13:36:50.066 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"63e5cec8-9099-4576-8575-17f252da9083","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2024-08-25 13:36:50.365 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "63e5cec8-9099-4576-8575-17f252da9083", {"configurationKey": [{"key": "MeterValuesSampledData", "readonly": false, "value": "Current.Export"}], "unknownKey": []}]
2024-08-25 13:36:50.371 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"16aa5c35-fb72-4201-bbc4-95f9f74dbe98","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
2024-08-25 13:36:50.682 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "16aa5c35-fb72-4201-bbc4-95f9f74dbe98", {"status": "Accepted"}]
2024-08-25 13:36:50.686 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"096a0aa4-f2fb-4688-be2b-fc583fb862f1","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Export"}]
2024-08-25 13:36:52.000 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "096a0aa4-f2fb-4688-be2b-fc583fb862f1", {"status": "Accepted"}]
2024-08-25 13:36:52.004 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"625a0f92-720c-4a3b-a772-12f72027d7a2","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
2024-08-25 13:36:52.316 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "625a0f92-720c-4a3b-a772-12f72027d7a2", {"status": "Accepted"}]
2024-08-25 13:36:52.319 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"6f0efa3d-ba16-4930-993e-ff1900784688","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
2024-08-25 13:36:52.641 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "6f0efa3d-ba16-4930-993e-ff1900784688", {"status": "Accepted"}]
...
2024-08-25 13:37:03.681 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"6c80acfd-fe95-427a-8c7b-20b7e591f290","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Temperature"}]
2024-08-25 13:37:03.993 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "6c80acfd-fe95-427a-8c7b-20b7e591f290", {"status": "Accepted"}]
2024-08-25 13:37:03.998 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"d1364b9f-36e4-4292-a256-7756687a3c8e","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"SoC"}]
2024-08-25 13:37:04.310 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "d1364b9f-36e4-4292-a256-7756687a3c8e", {"status": "Accepted"}]
2024-08-25 13:37:04.314 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"b17d4bb7-fa62-42a4-a107-08bf68f20d7d","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
2024-08-25 13:37:04.626 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "b17d4bb7-fa62-42a4-a107-08bf68f20d7d", {"status": "Accepted"}]
2024-08-25 13:37:04.630 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"43931561-4e65-42fa-9a29-8ef97984bde6","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Temperature"}]
2024-08-25 13:37:34.634 INFO (MainThread) [ocpp] TACXXXXXXXXXX: send [2,"780b879b-d591-4cce-be1a-b507bfb3ee7d","GetConfiguration",{"key":["MeterValuesSampledData"]}]
2024-08-25 13:37:34.856 INFO (MainThread) [ocpp] TACXXXXXXXXXX: receive message [3, "780b879b-d591-4cce-be1a-b507bfb3ee7d", {"configurationKey": [{"key": "MeterValuesSampledData", "readonly": false, "value": "Voltage"}], "unknownKey": []}]
[... cut ...]
websockets.exceptions.ConnectionClosedError: no close frame received or sent                                                                                                                                                                                 
2024-08-27 18:40:38.344 INFO (MainThread) [websockets.server] connection closed
2024-08-27 18:40:38.346 DEBUG (MainThread) [websockets.server] = connection is CLOSED
```

Not that we can use `configure` service directly either, as it will ask for current configuration first.

rinigus, 2024-08-27:
What helped, was to force applicable settings somewhere in the beginning of the `try` block before measurands are probed:

```
            _LOGGER.info(f"'{self.id}' setting measurands to default values")
            req = call.ChangeConfiguration(key=key, value="Current.Import,Energy.Active.Import.Register,Power.Active.Import,Voltage")
            resp = await self.call(req)
```

and commenting out all that autoconfig of  measurands. Would have to test bit tomorrow as well.

drc38, 2024-08-27:
The best solution would be for ABB to correct the issue in their firmware, otherwise you will need to create your own fork of the repo with the above changes to workaround the issue.

rinigus, 2024-08-28:
Agreed, correction of ABB firmware would be the best solution. Would have to contact ABB and make them aware of it.

However, with such type of devices as chargers, we are expected to stumble on bugs in the firmware. In addition, forking this repo for ABB chargers would fragment the development instead of making this implementation as a one to use by many. Thus, I propose to handle these types of errors in this repository.

To fix this bug, I would work on PR that will:

- [ ] Add a configuration option to disable automatic detection of the measurands. This will be permanently enabled for ABB chargers and others that could have similar issues.
- [ ] Add a quirk to set `MeterValuesSampledData` to the one specified in configuration by users. It would have to be used by users who have got their chargers into the same state as mine (I would expect it to be few based on this issue, as well as issues #1263 and #1126). Later, users can disable it when the charger is working properly.
- [ ] Update documentation on how to configure ABB Terra AC charger.

Will try to work on this tonight.

drc38, 2024-08-28:
@rinigus that'd be a good solution, it was talked about shortly after the switch in logic to v0.5 but not implemented. I'd suggest adding an options flow to allow for reconfiguration of parameters after the initial setup.

rinigus, 2024-08-28:
@drc38 just saw your feedback while working on PR. It would be great if the options could be later changed as right now we need to remove and add integration for it. However, I am new to HA and it would be my first contribution to it - so let's start with what we have :) . Will learn by doing and maybe in time I could add options flow. But, I suspect that someone will beat me to it.

Current state is submitted as PR. Would have to add a quirk, docs, and do more testing. Will continue working on it tomorrow.

drc38, 2024-08-28:
Sounds good @rinigus. I take it ABB has been notified of the issue as it is a serious flaw in their firmware and as a reputable company you would expect them to issue a fix reasonably promptly.

rinigus, 2024-08-29:
@drc38 Re ABB contact: not yet, but will do it. Added it into TODO list for PR.

closed 2024-09-09 by lbbrhzn

sulazix, 2025-02-06:
@rinigus hey thanks for contacting ABB !  Do you know if the issue if fixed now ?

raulliive, 2025-02-10:
I reached out to ABB Charger team in Estonia few weeks ago.

They promised to forward issue report to the factory.

InnoGreenTech, 2025-03-27:
Hello i had the same problem, i have change the default port to resolve it.

[image]

PR #1290 (2024-08-31): add try except for getting measurands
fix #1275
PR #1292 (2024-09-09): Allow to disable autodetection of measurands
This PR is to address issues induced by ABB Terra AC firmware bug:
- #1275 
- probably #1263 
- probably #1126

As such bugs could be in other chargers, the solution is generic:

- Add config option to detect measurands automatically, set default to true

- Restrict checked measurands to the ones provided by users

I still need to add a quirk to make ABB chargers working again and update docs. So, it is WIP at this stage.

Todo list:

- [x] Notify ABB regarding the issue
- [x] Add a configuration option to disable automatic detection of the measuran [... cut]