lbbrhzn/ocpp #1303: Default heartbeat interval

rinigus, 2024-09-02:
**Is your feature request related to a problem? Please describe.**
For some reason, my ABB Terra AC had 3600 s set as a heartbeat interval. Which in practice resulted in charger being not connected to HA after HA restart (update, for example). 

**Describe the solution you'd like**
Would be great to have the heartbeat interval checked on start and set to some reasonable value if it is too large. It doesn't have to be configuration option (to keep them lower), but could be just set in the code to 5-10 minutes.

Adding as an option would improve user awareness. But maybe that can be done when options can be changed after install.

drc38, 2024-09-03:
My recollection is setting the heartbeat to a default value was removed at some point due to incompatibility with some chargers setting it.

With websockets the heartbeat is only used for time synchronisation.

rinigus, 2024-09-03:
> My recollection is setting the heartbeat to a default value was removed at some point due to incompatibility with some chargers when setting it.

OK, then maybe I should just use external automation and set it on boot using OCPP configure service.

For ABB, it seems to me, heartbeat is something that is used to reboot/reconnect the charger is the connection has been broken due to HA reboot.

rinigus, 2024-09-09:
For others who may have the similar issue, here is a fix:

Add automation that would be triggered by ABB Version Firmware change. That's happening on boot. In automation, set up delay of 5 minutes or so and after that set heartbeat to the value that you want. Example config below:

```yaml
alias: Set ABB charger heartbeat on boot
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.abb_charger_version_firmware
condition: []
action:
  - delay:
      hours: 0
      minutes: 5
      seconds: 0
      milliseconds: 0
  - action: ocpp.configure
    metadata: {}
    data:
      ocpp_key: HeartbeatInterval
      value: 300
mode: single

``` 

Closing it here.

closed 2024-09-09 by rinigus