lbbrhzn/ocpp #1306: OCPP is not shown in HA Integration

TMoersch, 2024-09-05:
Hi,
I wan't to add my Wallbox to the new energy chart and therefore I need to install OCPP in order to communicate with my Vestel EVC04 Wallbox. The repository is allready integrated in the HA integration list (checked), but when I search for OCPP; ocpp; open charger or whatsoever, the OCPP Integration does not pop up for installation?

Why? What am I doing wrong?

Thanks Thomas

g3r0nim00, 2024-09-05:
Hi,

I had the same issue this morning when I had to spin-up a test HA instance (for OCPP & Alfen charger troubleshooting)...
Clean install via OVA: 2024.9.0, HACS 2.0.1 & ocpp 0.5.9.
I had to re-download the ocpp package via the HACS interface & re-restart HA. 
After this 2nd attempt I was able to add the ocpp integration.

Hope this helps

Jeroen

lbbrhzn, 2024-09-09:
solved in v0.5.12

closed 2024-09-09 by lbbrhzn