lbbrhzn/ocpp #1308: ABB Tera AC w22-T-R-O - all entities are 'unavailable'

NiiiYaa, 2024-09-06:
Hello,

I have a charging unit ABB Tera AC w22-T-R-O.

I'm trying to connect it to my Home Assistant Via the HACS OCPP integration.

* HA version - Core: 2024.9.0 | Supervisor: 2024.08.0 | Operating System: 13.1
* Version of the custom_component OCPP- v0.5.9
* No YAML configuration was added/created.

After following the documentation steps/instructions, a new integration was added:
[image]
[image]

Accordingly, 2 device were added:
[image]

The 'central' device has no entities: 
[image]

The abb device has 52 entities, which all of those (expect one) are grayed-out (unavailable):
[image]

[image]

I cannot find any error logs related to this...

What do I do wrong? Can someone kindly please assist me

Thank you in advance,
N

rinigus, 2024-09-06:
I suspect it is related to  #1275. At least that was an issue with my charger. Fortunately, there is a fix and its submitted: #1292. We have to wait a bit for review process to get it sorted out. 

If you want to test it, you could use this branch: https://github.com/rinigus/ha-ocpp/tree/config-autodetect-measurands
Note that this branch has also updated documentation regarding ABB chargers. Please read it first.

Until the fix is applied, I would suggest to switch off the charger or connect it somewhere else. If it is the same issue as #1275, your charger would be bootloop'ing otherwise.

xlammer, 2024-09-06:
> Not sure if it's only ABB related, but after upgrading HA to 2024.9.0 mine w11 stopped working. <img alt="image" width="1402" src="https://private-user-images.githubusercontent.com/5887925/365091524-88fda8a8-8644-4e8b-9906-034081244088.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MTk2MjEsIm5iZiI6MTcyNTYxOTMyMSwicGF0aCI6Ii81ODg3OTI1LzM2NTA5MTUyNC04OGZkYThhOC04NjQ0LTRlOGItOTkwNi0wMzQwODEyNDQwODgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MDZUMTA0MjAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzBhYWJmYzdjZDZiZTAyN2Q0ZWU2ZTE1M2IyODNhNTMzZjI4OTUwYmJmMWM4ZDQwMWRkZmRkNWZiNzBkYTM2NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.z6GPeo-d6eHraQNTpjnfdzQDCrEDXfcLRtVA84HsgfU">

This seems to unrelated to HA update, WIFI connection issue.

rinigus, 2024-09-09:
Please test with the latest OCPP version (0.5.11) and follow instructions at https://home-assistant-ocpp.readthedocs.io/en/latest/supported-devices.html under ABB subsection. I hope your issue is fixed.

ksga, 2024-11-13:
> Please test with the latest OCPP version (0.5.11) and follow instructions at https://home-assistant-ocpp.readthedocs.io/en/latest/supported-devices.html under ABB subsection. I hope your issue is fixed.

The workarounds described in the link worked for me.

oh-c, 2024-11-15:
The workaround with disable automatic measureands works somewhat for me as well.. It does, however, often give me the "StartupOfTheDevice" notification. Today it stopped reporting back and I saw the following in my logs.

Firmware 1.8.21 as well

`2024-11-15 22:44:53.858 INFO (MainThread) [websockets.server] connection closed
2024-11-15 22:44:53.859 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-11-15 22:44:53.869 DEBUG (MainThread) [websockets.server] = connection is CONNECTING
2024-11-15 22:45:00.821 DEBUG (MainThread) [websockets.server] ! invalid handshake
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/websockets/legacy/server.py", line 288, in read_http_request
    path, headers = await read_request(self.reader)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/websockets/legacy/http.py", line 81, in read_request
    raise ValueError(f"invalid HTTP request line: {d(request_line)}") from None
ValueError: invalid HTTP request line: \x81\xfe\xe2abcd:PODCRS]QSV\CNCF,FMBF

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/websockets/legacy/server.py", line 171, in handler
    await self.handshake(
  File "/usr/local/lib/python3.12/site-packages/websockets/legacy/server.py", line 591, in handshake
    path, request_headers = await self.read_http_request()
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/websockets/legacy/server.py", line 292, in read_http_request
    raise InvalidMessage("did not receive a valid HTTP request") from exc
websockets.legacy.exceptions.InvalidMessage: did not receive a valid HTTP request
2024-11-15 22:45:00.824 DEBUG (MainThread) [websockets.server] > HTTP/1.1 400 Bad Request
2024-11-15 22:45:00.825 DEBUG (MainThread) [websockets.server] > Date: Fri, 15 Nov 2024 21:45:00 GMT
2024-11-15 22:45:00.825 DEBUG (MainThread) [websockets.server] > Server: Python/3.12 websockets/13.1
2024-11-15 22:45:00.825 DEBUG (MainThread) [websockets.server] > Content-Length: 157
2024-11-15 22:45:00.826 DEBUG (MainThread) [websockets.server] > Content-Type: text/plain
2024-11-15 22:45:00.826 DEBUG (MainThread) [websockets.server] > Connection: close
2024-11-15 22:45:00.826 DEBUG (MainThread) [websockets.server] > [body] (157 bytes)
2024-11-15 22:45:00.828 INFO (MainThread) [websockets.server] connection rejected (400 Bad Request)
2024-11-15 22:45:00.829 DEBUG (MainThread) [websockets.server] x closing TCP connection
2024-11-15 22:45:00.829 DEBUG (MainThread) [websockets.server] = connection is CLOSED
2024-11-15 22:45:00.831 INFO (MainThread) [websockets.server] connection closed
2024-11-15 22:45:00.858 DEBUG (MainThread) [websockets.server] = connection is CONNECTING`

closed 2025-01-22 by github-actions[bot]