lbbrhzn/ocpp #1318: Detected blocking call to open with args

Vestkrogen, 2024-09-09:
**Describe the bug**
Loosing connection to my wallbox pulsar plus i suspect this is the reason.
[image]

**To Reproduce**
Fault log:
Logger: homeassistant.util.loop
Kilde: util/loop.py:136
Første forekomst: 16.51.13 (1 forekomster)
Senest logget: 16.51.13

Detected blocking call to open with args ('/usr/local/lib/python3.12/site-packages/ocpp/v16/schemas/BootNotification.json', 'r') inside the event loop by custom integration 'ocpp' at custom_components/ocpp/api.py, line 1010: await super()._handle_call(msg) (offender: /usr/local/lib/python3.12/site-packages/ocpp/messages.py, line 164: with open(path, "r", encoding="utf-8-sig") as f:), please create a bug report at https://github.com/lbbrhzn/ocpp/issues For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 189, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete self.run_forever() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever self._run_once() File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once handle._run() File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run self._context.run(self._callback, *self._args) File "/usr/local/lib/python3.12/site-packages/ocpp/charge_point.py", line 238, in start await self.route_message(message) File 

**Expected behavior**
A clear and concise description of what you expected to happen.

lbbrhzn, 2024-09-09:
Which version of the integration are you using?

Vestkrogen, 2024-09-09:
I am using OCPP v0.5.10, and HA is as follows:
Core
2024.9.1
Supervisor
2024.08.0
Operating System
13.1
Brugerflade
20240906.0

Its running on a HA Yellow....

lbbrhzn, 2024-09-09:
try v0.5.11

Vestkrogen, 2024-09-09:
> try v0.5.11

Thx. I'll do that.

lbbrhzn, 2024-09-09:
unfortunately something went wrong when releasing v0.5.11, it does not work. Please try v0.5.12 instead.

drc38, 2024-09-10:
See #1199

Vestkrogen, 2024-09-10:
> See #1199

Thx - i hope that will fix my issue :)

Vestkrogen, 2024-09-10:
> unfortunately something went wrong when releasing v0.5.11, it does not work. Please try v0.5.12 instead.

Thank you very much, but i am still having connectivity issues. - Looks like it could be the problem described in #1199 as according to comment from drc38 below :)

drc38, 2024-09-10:
The warning and issue with blocking is unlikely to be related to your connectivity issues as the blocking code will still run. Enable debug and take a closer look through the logs.

closed 2024-11-17 by github-actions[bot]