lbbrhzn/ocpp #1320: Version 0.5.11 does not work at all

dreimer1986, 2024-09-09:
**Describe the bug**
Installing 0.5.11 causes the OCPP integration to be called not existing. 0.5.10 works fine.

**To Reproduce**
Steps to reproduce the behavior:
Just update and see

**Expected behavior**
A clear and concise description of what you expected to happen.

To work as good as 0.5.10 ^^

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser Chrome
 - Version no idea

**Smartphone (please complete the following information):**
 - Device: Pixel 7 Pro
 - OS: Android 15
 - Browser Chrome
 - Version no idea.

**Additional context**
Add any other context about the problem here.

pvprodk, 2024-09-09:
Same

[image]

lbbrhzn, 2024-09-09:
Sorry, i messed up in creating the release tag. Please try v0.5.12

lbbrhzn, 2024-09-09:
solved in v0.5.12

closed 2024-09-09 by lbbrhzn