lbbrhzn/ocpp #1323: Vestel EVC04 - Reconnect every 25s

pvillmann, 2024-09-13:
**Describe the bug**
Hi, i have connected my Vestel EVC04 mit Firmware v3.166.0 (newest) via OCPP to the custom Integration in HA.
Vestel configuration and Config of Integration is completly default.

My Problem is, the connection drops every 25s and all entity are set to "not available". After additional 25s the connection is established again. (Reconnect counter increase every time by 1)

**To Reproduce**
Steps to reproduce the behavior:
Install OCPP integration in default way
Active OCPP on Vestel WebUI and set server to ws://ha-ip:9000/
Save

The Homeassistant Logs in debug mode shows:

2024-09-13 12:49:07.494 ERROR (MainThread) [ocpp] Error while handling request '<Call - unique_id=1778a2c8-5ecf-4738-8154-0ac840983d76, action=MeterValues, payload={'connectorId': 1, 'transactionId': 1726213848, 'meterValue': [{'timestamp': '2024-09-13T08:14:52Z', 'sampledValue': [{'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh', 'value': '1388', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Current.Offered', 'unit': 'A', 'value': '', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Power.Offered', 'unit': 'W', 'value': '11040', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Voltage', 'unit': 'V', 'phase': 'L1', 'value': '225', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Voltage', 'unit': 'V', 'phase': 'L2', 'value': '233', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Voltage', 'unit': 'V', 'phase': 'L3', 'value': '234', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Current.Import', 'unit': 'A', 'phase': 'L1', 'value': '15.9', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Current.Import', 'unit': 'A', 'phase': 'L2', 'value': '0.0', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Current.Import', 'unit': 'A', 'phase': 'L3', 'value': '0.0', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Power.Active.Import', 'unit': 'W', 'phase': 'L1', 'value': '3564', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Power.Active.Import', 'unit': 'W', 'phase': 'L2', 'value': '0', 'context': 'Sample.Periodic', 'location': 'Outlet'}, {'measurand': 'Power.Active.Import', 'unit': 'W', 'phase': 'L3', 'value': '0', 'context': 'Sample.Periodic', 'location': 'Outlet'}]}]}>'

pieewiee, 2024-09-13:
looks like it is a problem with your Python pip package 

https://pypi.org/project/python-chargepoint/

based on the reuqirments.txt you need ocpp==1.0.0

drc38, 2024-09-13:
The error comes from the underlying ocpp package. You can raise the issue here https://github.com/mobilityhouse/ocpp

pvillmann, 2024-09-13:
Done 

https://github.com/mobilityhouse/ocpp/issues/661

stefa168, 2024-09-16:
Hi, I have the same issue with a SCAME Parre Wallbox.

pvillmann, 2024-09-24:
https://github.com/mobilityhouse/ocpp/issues/661#issuecomment-2370362820

Can someone support here what's the best way to get python debug log in homeassistant for ocpp package?
I'm no HA developer :(

pvillmann, 2024-09-25:
As described in mobilityhouse/ocpp#661 is not a ocpp package problem.

You need to add some checks before calling float(value)
This is server side, so in "lbbrhzn/ocpp".

PR #1336 (2024-09-26): fix empty value
fixes #1323 https://github.com/mobilityhouse/ocpp/issues/661
closed 2024-09-26 by lbbrhzn