lbbrhzn/ocpp #1341: Error changing "Maximum Current"

SimoneBWS, 2024-09-27:
## Version 
v0.5.12

## Configuration
Setup through UI

## Describe the bug
With a Zappi charging station, when I try to change the **Maximum Current** I get this error:
```text
int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
```

## Debug log

```text
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:245
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: September 25, 2024 at 18:52:04 (7 occurrences)
Last logged: 13:17:11

[140044712355600] Unexpected exception
[140044820171216] Unexpected exception
[140044675283072] Unexpected exception
[140044571489856] Unexpected exception
[140044510746880] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2761, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2804, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 996, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1068, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 120, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/ocpp/number.py", line 130, in async_set_native_value
    resp = await self.central_system.set_max_charge_rate_amps(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 302, in set_max_charge_rate_amps
    return await self.charge_points[cp_id].set_charge_rate(limit_amps=value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/api.py", line 720, in set_charge_rate
    stack_level = int(resp)
                  ^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
```

svhelge, 2024-10-20:
When trying to change the max current in Home Assistant I get the same error. The charger is a "Futurehome Charge".

closed 2024-12-28 by github-actions[bot]

SimoneBWS, 2025-02-13:
Is there any hope for this problem?