lbbrhzn/ocpp #1357: No option to turn off persistent notification

elad-bar, 2024-10-09:
**Describe the bug**
No option to turn off persistent notification

**To Reproduce**
Steps to reproduce the behavior:
1. Restart HA

**Expected behavior**
persistent notification works only for debug mode
https://www.home-assistant.io/integrations/logger/

Diagnostic file will be used for storing state for more advanced debugging

lightheaded, 2024-10-20:
I second this. I installed my charger yesterday and have over 60 notifications in 24 hours. That's not useful for me and drowns out the few useful persistent notifications I have:
[image]

Some Integrations use (somewhat to my surprise) the same id for the notifications, so I have been able to clear them using an automation in the past. However, here a different ID is used for every notification. This makes sense from what IDs usually are used for, but makes it harder to dismiss:
[image]

If you want, you can use a condition to by `trigger.notification.title`, as it's always "Ocpp integration" and then dismiss it by ID. 

But this is a workaround and it would be great to see an option for this.

closed 2025-02-26 by github-actions[bot]

lightheaded, 2025-02-26:
If anyone wants to take this up, there clearly seems a need for it. People are creating automations to dismiss these notifications, so it doesn't seem helpful for all.