lbbrhzn/ocpp #1366: Support for multiple connectors

helestrand, 2024-10-21:
I would like to request (once again?) the development of the feature for multiple connector support.

I believe this has been worked on before, but unfortunately, it does not seem to have been included in the integration for some reason. Below is a link to the closed feature:
[https://github.com/lbbrhzn/ocpp/pull/803](url)

I really appreciate the integration—great work! I hope the multiple connector feature can finally be included and make th integration complete.

lbbrhzn, 2024-10-21:
I've reopened the pull request... but there are some issues that need to be resolved before this can be merged to main.

helestrand, 2024-10-21:
Great, thank you. 
If someone could look into the issues and resolve them, many, including myself, would be grateful!

Steph6431, 2024-10-31:
I would also be very happy to control both Wallbpxes with this great integration. ;-)

markollindvall, 2024-12-08:
Dual socket model support would be really great. I have 2 dual socket chargers and I can only command connector 1 (left socket) with HA. Some entities have 1 and 2 attributes like connector state and connector error but real commands would be great. Any workaeounds for this?

Marianp90, 2024-12-11:
I would also like to use the integration for both of my Mennekes wall charging stations. Kindly let me know, if there is any beta to test multi device support.

markollindvall, 2024-12-11:
Multi charger device support is already working. You just need to use different port and server/client names for these chargers when adding to ocpp server.

Like:

Charger 1: port 9001 server1/charger1
Charger 2: port 9002 server2/charger2

If you would have 2 sockets in one charger only the left socket would be working trough ocpp and this thread is about that need.

lightheaded, 2024-12-14:
Thanks for the hint! I was able to connect both of my chargers by adding two integrations on separate ports, and it works as expected. So the support is there 👌

puhycz, 2025-01-13:
I add two chargers, one with port 9000, next one with port 9001 and it not working:

```
Logger: custom_components.ocpp
Zdroj: custom_components/ocpp/api.py:1023
integrace: Open Charge Point Protocol (OCPP) ([dokumentace](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [problémy](https://github.com/lbbrhzn/ocpp/issues))
První výskyt: 18:34:13 (27 výskyty)
Naposledy logováno: 18:38:34

Unexpected exception in connection to 'NB_Garaz': '01JHGBNCGT5GWCWDQEAQYRECTA'
Traceback (most recent call last):
  File "/config/custom_components/ocpp/api.py", line 1023, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/api.py", line 500, in post_connect
    self.hass.config_entries.async_update_entry(self.entry, data=updated_entry)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2177, in async_update_entry
    raise UnknownEntry(entry.entry_id)
homeassistant.config_entries.UnknownEntry: 01JHGBNCGT5GWCWDQEAQYRECTA
```
**Please how connect two chargers?**

Thanks, Jan

closed 2025-03-22 by github-actions[bot]