lbbrhzn/ocpp #1376: Integration won't load in HA 2024.11.0b0

Plawasan, 2024-10-30:
## Version of the custom_component

0.5.13

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

Integration won't load in HA 2024.11b0 with the error shown below

## Debug log

```text

Logger: homeassistant.setup
Source: setup.py:334
First occurred: 8:31:40 PM (1 occurrences)
Last logged: 8:31:40 PM

Setup failed for custom integration 'ocpp': Unable to import component: cannot import name 'Config' from 'homeassistant.core' (/usr/src/homeassistant/homeassistant/core.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1011, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1071, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/ocpp/__init__.py", line 6, in <module>
    from homeassistant.core import Config, HomeAssistant
ImportError: cannot import name 'Config' from 'homeassistant.core' (/usr/src/homeassistant/homeassistant/core.py). Did you mean: 'config'?

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1031, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1023, in async_get_component
    comp = self._get_component()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1071, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/ocpp/__init__.py", line 6, in <module>
    from homeassistant.core import Config, HomeAssistant
ImportError: cannot import name 'Config' from 'homeassistant.core' (/usr/src/homeassistant/homeassistant/core.py). Did you mean: 'config'?

```

drc38, 2024-11-01:
Feel free to subit a PR with the fix. There's a deprecation period in latest beta to switch over.

Plawasan, 2024-11-01:
I'd prefer someone with a bit more expertise/experience do that, I just got lucky the first thing I tried actually worked but I don't feel like messing with the main repo.

Plawasan, 2024-11-06:
https://developers.home-assistant.io/blog/2024/10/31/core-config-moved/ - here's how to properly address the change

lbbrhzn, 2024-11-06:
Thanks. Seems simple enough, i'll give it a try!

closed 2025-01-06 by Plawasan