lbbrhzn/ocpp #1403: "The unit of measurement of some entities has changed" warning messages

lechercheur123, 2024-11-24:
Hello

**Describe the bug**
Everytime I restart Home Assistant, I have 2 messages saying the unit of measurement of 2 entities have changed:
[image]
(Sorry it is in French. It is saying "The unit of sensor.charger2_power_reactive_import has changed" and "The unit of sensor.charger2_energy_reactive_export_register has changed"

If I click on one message, I get this:
[image]
In English it means:

> The unit has changed
> The unit of "charger2 Power Reactive Import" (sensor.charger2_power_reactive_import) has been changed to "kW" which cannot be converted to the previously stored unit, "var".
> If the historical statistics values ​​have the wrong unit, you can update the units of the old values. The values ​​will not be updated.
> Alternatively, you can choose to delete all historical statistics values ​​and start over.
> How do you want to fix this issue:
> Option 1: Update the unit of historical statistics values from "var" to "kW", without conversion
> Option 2: Delete old data

The same warning can be found in the logs:
```
2024-11-24 16:05:10.127 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.charger2_energy_reactive_import_register is changing, got multiple {'kWh', 'varh'}, generation of long term statistics will be suppressed unless the unit is stable and matches the unit of already compiled statistics (varh). Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2024-11-24 16:05:10.207 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.charger2_energy_reactive_export_register (kWh) cannot be converted to the unit of previously compiled statistics (varh). Generation of long term statistics will be suppressed unless the unit changes back to varh or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
2024-11-24 16:05:10.207 WARNING (Recorder) [homeassistant.components.sensor.recorder] The unit of sensor.charger2_power_reactive_import (kW) cannot be converted to the unit of previously compiled statistics (var). Generation of long term statistics will be suppressed unless the unit changes back to var or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
```

It is not a blocking issue, but it is annoying to have to open these messages everytime I have to reboot HA (after an update, after a configuration change...)

**To Reproduce**
Steps to reproduce the behavior:
1. Restart HA

**Expected behavior**
Nothing

**Screenshots**
See above

**Desktop:**
 - OS: Windows (HA is on a VM on a Proxmox hypervisor)
 - Browser Firefox
 - Version 132

**Additional context**
I think this issue might be linked to the charger. In fact, I have two chargers connected to my HA:
- charger, an Schneider EVlink Smart Wallbox 22kW, which is working fine with the OCPP integration. Its ID is "charger22kW"
- charger2, a eProWallbox Move from free2move, which I think is a crappy wallbox. Its ID is 1694AB070000244

I only have the issue with "charger2" (as seen on the screenshots)

To have 2 chargers on the same HA I used two instances of the OCPP integration listening to different ports, like this:
[image]

I guess this is the good way to do, no?

I will put the log file in attachment.
[home-assistant_ocpp_2024-11-24T16-52-05.376Z.log](https://github.com/user-attachments/files/17894201/home-assistant_ocpp_2024-11-24T16-52-05.376Z.log)

onlyfornetwork, 2024-11-27:
I have the same problem,
 
it also occurs when the column is turned on by the  integration 

Wallbox Legrand Green Up 059006

N3rdix, 2024-11-29:
The repair should disappear if you delete the old long-term statistics (if you don't need them from those entities).
I assume there was a reason to change the UOM of those entities, so this might be the only persistent option...

lechercheur123, 2024-11-29:
If I delete the warning saying that the unit has been changed from 'var' to 'kW', at the next reboot I have a warning saying the unit has been changed from 'kW' to 'var'...
So this isn't a long term fix

drc38, 2024-11-30:
You can try updating the logic here: 
https://github.com/lbbrhzn/ocpp/blob/a55913cec503110b6b2a5270bcd5f57e145ac19a/custom_components/ocpp/sensor.py#L152
with `SensorDeviceClass.REACTIVE_ENERGY`

N3rdix, 2024-11-30:
I created a PR to fix at least the power reactive sensors, maybe you can review if it'S fine from your perspective @drc38.
There is no equivalent for reactive energy in the device classes of HA as it seems...

drc38, 2024-11-30:
> I created a PR to fix at least the power reactive sensors, maybe you can review if it'S fine from your perspective @drc38. There is no equivalent for reactive energy in the device classes of HA as it seems...

Yeah, looks like I was looking at a fork that did not get merged by the devs. See my suggestion on energy in your PR. It should stop the warning by not giving it a device class.

PR #1417 (2024-12-01): Correct reactive power device class
fixes #1403 partially by giving a device class for reactive power sensors.

There is no device class for reactive energy so far, means this cannot be fixed and does not have a device class assigned on purpose: https://developers.home-assistant.io/docs/core/entity/sensor/
Once HA provides a device class for it it can be added here...
closed 2024-12-01 by drc38