lbbrhzn/ocpp #1425: configure notifications

DOliana, 2024-12-09:
**Is your feature request related to a problem? Please describe.**
I have a charger (Vestel) that seems to reboot frequently - it might also be due to network connectivity. TBH I don't really care if it reboots or not. The issue is, that I get multiple notifications in home assistant because it reboots, which leads to me ignoring notifications completely.

**Describe the solution you'd like**
make notifications configurable.
some ideas (focussed on reboots, since that's all I have been seeing):
- allow to turn notifications for specific events on/off
- allow to configure a timeframe to send notifications (e.g. only notify once every x hours)
- add system entities (e.g. reboot counter, time last reboot etc)

N3rdix, 2024-12-15:
I have a Vestel as well and the only reboot notification I get is when Home Assistant starts. I bet it is a bad WIFI connection of your wallbox which leads to disconnects.
If the solution to dismiss notifications is not fine for you, you might need to improve connectivity (e.g. using a LAN cable)

DOliana, 2024-12-18:
> PRs are welcome to improve the documentation or codebase
if you give me a pointer to the right place in the docs, I can add a PR. Right now I couldn't find a section for FAQs or similar in this repo.

DOliana, 2024-12-18:
> I have a Vestel as well and the only reboot notification I get is when Home Assistant starts. I bet it is a bad WIFI connection of your wallbox which leads to disconnects. If the solution to dismiss notifications is not fine for you, you might need to improve connectivity (e.g. using a LAN cable)

I am aware of the bad connectino - I have to hook up the cable. In the meantime I still think there might be others facing similar issues and an explanation could help.

drc38, 2024-12-18:
> > PRs are welcome to improve the documentation or codebase
> 
> if you give me a pointer to the right place in the docs, I can add a PR. Right now I couldn't find a section for FAQs or similar in this repo.

You could add it in the support section https://github.com/lbbrhzn/ocpp/blob/main/docs/support.md

DOliana, 2025-01-01:
> You could add it in the support section https://github.com/lbbrhzn/ocpp/blob/main/docs/support.md

To me it looked as if that should just redirect to the forums. Anyway, if you say that's the place I am fine with it :)
Let me start writing something and create a PR - that can be used to discuss the right place.

closed 2025-03-11 by github-actions[bot]