lbbrhzn/ocpp #1429: Supported Devices, Schneider Charge EVH5AxxN2xx confirmed working

BikeMikeAU, 2024-12-16:
Hi, could you please add this range confirmed as working in the documentation.

Link to the product: https://www.se.com/au/en/product-range/161147017-schneider-charge/#products
Product codes in this range:
EVH5A07N205  7kw with 5M attached cable
EVH5A07N207  7kw with 7M attached cable
EVH5A22N205  22kw with 5M attached cable
EVH5A22N207  22kw with 7M attached cable
EVH5A22N2S  22kW with socket

The charger first needs to be configured via the Schneider eSetup app (free). Latest firmware is automatically checked and installed by the app. Select "Other smart charging application" then enter Home Assistant URL eg. ws://192.168.1.5:9000 and a station ID.

Frank-Gr, 2025-01-14:
Does anybody know the login for the Schneider eSetup App? I am not an electrician an do not have a user ID.

BikeMikeAU, 2025-01-14:
The login is optional. Just click the "Skip" button to continue anonymously

Frank-Gr, 2025-01-15:
Thanks a lot! I didn't see the "skip" button :-)

flashg1, 2025-02-17:
Hi @BikeMikeAU,

Just wondering how well Schneider Charge EVH5AxxN2xx chargers are working with HA OCPP?  Do you have automations that can start/stop the charger and vary the current?

What is the minimum current you can set the charger to?  Can it be set to 0 Amp?  I charge on excess solar most of the time, and I need to vary the current between 0A and maximum current.

Appreciate any experience you can provide in using Schneider Charge.  Thanks.

BikeMikeAU, 2025-02-17:
Hi @flashg1,
Yes it's working perfectly for me. The minimum that works is 7A, but this is a limitation of the vehicle. The charger can be adjusted between 0 and 32A in 1A increments.

I have developed an automation that tracks my free solar and distributes it between controllable loads (hot water heater, pool pump, pool solar heater pump, EV charger) based on multiple conditions and priority sequences, so it is quite complex. I struggled to get this working with automations or yaml, so now I'm using python_script as a service called by automation every 2 minutes.

I have a "free power" helper that is power being exported plus power currently being used by the 4 loads. Then the script calculates which loads there is enough power for and which have priority, turns those on/adjust current, and turns others off.

flashg1, 2025-02-17:
Thank you for your detailed description and Python script.

"Yes it's working perfectly for me. The minimum that works is 7A, but this is a limitation of the vehicle. The charger can be adjusted between 0 and 32A in 1A increments."

Interesting that the MG have a 7A minimum charge current limitation.  I have a Tesla M3 and I can vary the charge current between 0A and 15A depending on available solar and household consumption using this [blueprint](https://github.com/flashg1/evSolarCharger).  It is not as sophisticated as yours since I only need to deal with heating the hot water and charging the car, with the car having the lowest priority.

Agreed that yaml is not very nice to code with.  I will probably change mine to Python script at some stage.

I am considering getting an OCPP charger, possibly a Schneider Charger.

flashg1, 2025-02-18:
FYI, you might want to read this on how best to set the current using OCPP,

https://github.com/taste66/ocpp/blob/patch-1/docs/Charge_automation.md

BikeMikeAU, 2025-02-18:
Ok, thanks for that. Yes I had considered that changing the max current might be impacting flash memory, but never saw that explicitly stated before now. And I'm potentially changing it hundreds of times per day, so I will try out that solution.

I think I was able to reduce it to 6A but found it still was drawing 6.5. Also I have noted based on EV charge current/voltage vs. the charger current/voltage, that charging is about 90-93% efficiency, but at current below 10A the efficiency drops a bit. Not really a concern if only using solar anyway.

BikeMikeAU, 2025-02-19:
I have updated the python script posted above to use ocpp.set_charge_rate with a custom profile. It works just like before but hopefully not wearing out any flash storage in the charger now!

BJReplay, 2025-02-19:
> FYI, you might want to read this on how best to set the current using OCPP,

Thanks, that was immensely useful.  I've swapped from _maximum_current to ocpp.set_charge_rate on my EN+ Caro, which I'll submit a PR as mostly working.

The only issue that I have is that there is an error on restart of HA, which I need to track down (reports that the charger didn't accept a command - but I can't recall which), but other than that, solar excess charging works well.

BikeMikeAU, 2025-02-19:
You will get an error "Failed with response: Rejected" if you do occp.set_charge_rate and the charger is off; this is expected. Probably your automation is triggering on restart.

flashg1, 2025-02-19:
Thanks @BJReplay  for being the guinea pig!  I have only been using it with the Tesla API for the past 2 years, so hopefully it should also work with the OCPP API.

I am traveling for the next 2 weeks, but will definitely check my mail when I come back.

closed 2025-04-28 by github-actions[bot]

andymac4182, 2025-09-09:
I have just connected a EVH5A22N2S and it detected most things but the metrics about actively charging are all 0. 

Any ideas on why this might be?