lbbrhzn/ocpp #1437: Stopped working after updating HA core to 2024.12.4

berdugorn, 2024-12-19:
latest version

a few days ago it stopped working, it said the charger is unavailable
i reinstalled the occp add-on and it didnt change anything
the error:
`Logger: homeassistant.util.package
Source: util/package.py:152
First occurred: 17:05:15 (6 occurrences)
Last logged: 17:05:38

Unable to install package ocpp>=1.0.0: error: Failed to fetch: `https://files.pythonhosted.org/packages/a5/66/5e427455af31dff76929f3f7117cb21a957b19eb6133eb0a50f7a3d1bb78/ocpp-1.0.0-py3-none-any.whl.metadata` Caused by: Could not connect, are you offline? Caused by: Request failed after 3 retries Caused by: error sending request for url (https://files.pythonhosted.org/packages/a5/66/5e427455af31dff76929f3f7117cb21a957b19eb6133eb0a50f7a3d1bb78/ocpp-1.0.0-py3-none-any.whl.metadata) Caused by: client error (Connect) Caused by: dns error: failed to lookup address information: Name does not resolve Caused by: failed to lookup address information: Name does not resolve`

Today i restored the HA VM to a week ago and it worked again. I then tried to update HA Core again to 2024.12.4 and after the reboot it showed the error again and it was unavailable.

Thank you in advance,
George

lechercheur123, 2025-01-07:
Hello,
I have the same issue, but with 2 integrations on HA 2025.1 (but I see it from 2024.12):
- ocpp (of course)
- sunspec (https://github.com/cjne/ha-sunspec)

```
2025-01-07 23:06:48.343 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'ocpp': Requirements for ocpp not found: ['ocpp>=2.0.0'].
2025-01-07 23:06:54.241 ERROR (SyncWorker_2) [homeassistant.util.package] Unable to install package pysunspec2==1.1.5: error: Failed to fetch: `https://files.pythonhosted.org/packages/5a/71/163ad2ce084f2f1d523b196a3b7c7a97fab30996d7d4a03fefac4339f8c4/pysunspec2-1.1.5-py3-none-any.whl.metadata`
  Caused by: Could not connect, are you offline?
  Caused by: Request failed after 3 retries
  Caused by: error sending request for url (https://files.pythonhosted.org/packages/5a/71/163ad2ce084f2f1d523b196a3b7c7a97fab30996d7d4a03fefac4339f8c4/pysunspec2-1.1.5-py3-none-any.whl.metadata)
  Caused by: client error (Connect)
  Caused by: dns error: failed to lookup address information: Name does not resolve
  Caused by: failed to lookup address information: Name does not resolve
2025-01-07 23:06:55.713 ERROR (SyncWorker_2) [homeassistant.util.package] Unable to install package pysunspec2==1.1.5: error: Failed to fetch: `https://files.pythonhosted.org/packages/5a/71/163ad2ce084f2f1d523b196a3b7c7a97fab30996d7d4a03fefac4339f8c4/pysunspec2-1.1.5-py3-none-any.whl.metadata`
  Caused by: Could not connect, are you offline?
  Caused by: Request failed after 3 retries
  Caused by: error sending request for url (https://files.pythonhosted.org/packages/5a/71/163ad2ce084f2f1d523b196a3b7c7a97fab30996d7d4a03fefac4339f8c4/pysunspec2-1.1.5-py3-none-any.whl.metadata)
  Caused by: client error (Connect)
  Caused by: dns error: failed to lookup address information: Name does not resolve
  Caused by: failed to lookup address information: Name does not resolve
```

So I think the issue is not limited to ocpp, but maybe it is a HA bug (or python).

I tried to investigate the bug with Wireshark, and I saw this:

[image]

You can see that my DNS resolver is answering correctly to the queries made by home assistant, but it seems to be stuck inside a loop. Below you will find some informations about the answer provided by my resolver:

[image]

The only troubling thing I see may be the very short TTL, but everything else seem fine to me.

So, I tried something to fix the low TTL, by adding a static entry to my DNS resolver for files.pythonhosted.org, thus also removing the CNAME part. Something like this (in my Mikrotik router):

[image]

And it made the loop to end:

[image]

I added the static entries between packets 698 and 699. You can see that the next answers (packets 701 and 702) are slightly different, and without the CNAME part. And the loop ends.

Here are the details of one of the answers:

[image]

So, I don't know what is causing this. This is clearly a bug, but maybe not directly from the ocpp integration. I will continue to investigate, but I don't find a lot of people having this issue. So it might be link to a specific configuration (IPv6 maybe ?).

See you later

berdugorn, 2025-01-08:
on the next HA core update the problem disappeared for me.. maybe it was a bug from HA.. who knows..

lechercheur123, 2025-01-08:
> on the next HA core update the problem disappeared for me.. maybe it was a bug from HA.. who knows..

Which update ? I still have the issue with 2025.1.1

lechercheur123, 2025-01-08:
I see that someone created an issue on HA core github: https://github.com/home-assistant/core/issues/135054

I will follow up there, and I think you can close the bug here.