lbbrhzn/ocpp #1444: Problem

danitampa, 2024-12-26:
Hi, sorry, I have often this report and today, in the logs, I sew the following text. I don't understand about programming, I'm a newbie. Thx for your help.

Danilo

[image]

.......
Registratore: homeassistant.util.loop
Fonte: util/loop.py:136
Prima occorrenza: 00:33:47 (1 occorrenze)
Ultimo accesso: 00:33:47

Detected blocking call to open with args ('/usr/local/lib/python3.13/site-packages/ocpp/v16/schemas/BootNotification.json', 'r') inside the event loop by custom integration 'ocpp' at custom_components/ocpp/chargepoint.py, line 516: await super()._handle_call(msg) (offender: /usr/local/lib/python3.13/site-packages/ocpp/messages.py, line 164: with open(path, "r", encoding="utf-8-sig") as f:), please create a bug report at https://github.com/lbbrhzn/ocpp/issues For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 708, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 679, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2027, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 238, in start await self.route_message(message) File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 261, in route_message await self._handle_call(msg) File "/config/custom_components/ocpp/chargepoint.py", line 516, in _handle_call await super()._handle_call(msg)
.......

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

lbbrhzn, 2024-12-26:
What is the version of the OCPP integration you are using?

There is an update to avoid blocking calls in v0.6.3

drc38, 2024-12-26:
This won't be resolved unless we go to ocpp>=2.0.0rc3 in the manifest. Had thought 2.0.0 would be out by now but there's no timeline for it unfortunately.

danitampa, 2024-12-26:
Good morning, the version is v0.6.3.

Can I take this opportunity to ask you why in your opinion I can't regulate
the charging power of my wallbox from HA? thanks a lot.

PR #1423 (2025-01-05): Update to ocpp 2.0.0 to fix blocking
Fixes #1444, fixes #1402, fixes #1371, fixes #1461
closed 2025-01-05 by lbbrhzn