lbbrhzn/ocpp #1449: Charger power reactive export/import have lost their state classes and units after update to v0.6.3

hummermann87, 2024-12-27:
**Describe the bug**
Charger power reactive export/import have lost their state classes and units after update to v0.6.3.

**To Reproduce**
Steps to reproduce the behavior:
N/A

**Expected behavior**
No errors

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Home Assistant, OS 14.1, Core 2024.12.5, Supervisor 2024.12.0

**Additional context**
Add any other context about the problem here.

N3rdix, 2024-12-28:
Good catch, I missed that when introducing the reactive power device_class. I'll fix it asap

BitsLeftOver, 2025-01-04:
Same problem.

N3rdix, 2025-01-04:
Should be fixed once #1451 is included in an upcomibg Release