lbbrhzn/ocpp #1450: Detected blocking call to open with args STATUS NOTIFICATION

alexcool83, 2024-12-27:
Registratore: homeassistant.util.loop
Fonte: util/loop.py:136
Prima occorrenza: 15:30:46 (1 occorrenze)
Ultimo accesso: 15:30:46

Detected blocking call to open with args ('/usr/local/lib/python3.13/site-packages/ocpp/v16/schemas/StatusNotification.json', 'r') inside the event loop by custom integration 'ocpp' at custom_components/ocpp/chargepoint.py, line 516: await super()._handle_call(msg) (offender: /usr/local/lib/python3.13/site-packages/ocpp/messages.py, line 164: with open(path, "r", encoding="utf-8-sig") as f:), please create a bug report at https://github.com/lbbrhzn/ocpp/issues For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main exit_code = runner.run(runtime_conf) File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return loop.run_until_complete(setup_and_run_hass(runtime_config)) File "/usr/local/lib/python3.13/asyncio/base_events.py", line 708, in run_until_complete self.run_forever() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 679, in run_forever self._run_once() File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2027, in _run_once handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run self._context.run(self._callback, *self._args) File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 238, in start await self.route_message(message) File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 261, in route_message await self._handle_call(msg) File "/config/custom_components/ocpp/chargepoint.py", line 516, in _handle_call await super()._handle_call(msg)

driesvandamme, 2024-12-28:
Having the same problem.

BitsLeftOver, 2025-01-04:
Same problem since 6.3. 6.4 no fix.

driesvandamme, 2025-01-04:
In custom_components/ocpp I changed the requirements to this:

ocpp @ git+https://github.com/mobilityhouse/ocpp.git@2.0.0-rc.3"

It seems to fix it until the PR gets merged

drc38, 2025-01-08:
Fixed in v0.7.0

closed 2025-01-08 by drc38