lbbrhzn/ocpp #1461: Integration won't load in 2025.1.0-b07

Plawasan, 2025-01-03:
**Describe the bug**
Integration fails to load in the latest HA Beta

**To Reproduce**
Update to 2025.1.0-b07
Restart

**Logs**
```
Logger: homeassistant.setup
Source: setup.py:334
First occurred: 9:43:27 AM (1 occurrences)
Last logged: 9:43:27 AM

Setup failed for custom integration 'ocpp': Unable to import component: cannot import name 'ConnectorStatusType' from 'ocpp.v201.enums' (/usr/local/lib/python3.13/site-packages/ocpp/v201/enums.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/ocpp/__init__.py", line 14, in <module>
    from .api import CentralSystem
  File "/config/custom_components/ocpp/api.py", line 18, in <module>
    from .ocppv201 import ChargePoint as ChargePointv201
  File "/config/custom_components/ocpp/ocppv201.py", line 19, in <module>
    from ocpp.v201.enums import (
    ...<17 lines>...
    )
ImportError: cannot import name 'ConnectorStatusType' from 'ocpp.v201.enums' (/usr/local/lib/python3.13/site-packages/ocpp/v201/enums.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/ocpp/__init__.py", line 14, in <module>
    from .api import CentralSystem
  File "/config/custom_components/ocpp/api.py", line 18, in <module>
    from .ocppv201 import ChargePoint as ChargePointv201
  File "/config/custom_components/ocpp/ocppv201.py", line 19, in <module>
    from ocpp.v201.enums import (
    ...<17 lines>...
    )
ImportError: cannot import name 'ConnectorStatusType' from 'ocpp.v201.enums' (/usr/local/lib/python3.13/site-packages/ocpp/v201/enums.py)
```

ohadbenita, 2025-01-03:
Same here for the release 2025.1.0

Plawasan, 2025-01-03:
Seems to be solved by v0.6.4

PR #1423 (2025-01-05): Update to ocpp 2.0.0 to fix blocking
Fixes #1444, fixes #1402, fixes #1371, fixes #1461
closed 2025-01-05 by lbbrhzn