lbbrhzn/ocpp #1474: Unit or measurement var cannot be converted to Kw

BitsLeftOver, 2025-01-07:
Install Occp 0.7.0.
debug log message:
****************
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:254
integration: Sensor (documentation, issues)
First occurred: 5:30:10 PM (2 occurrences)
Last logged: 5:30:10 PM

The unit of sensor.charger_power_reactive_export (var) cannot be converted to the unit of previously compiled statistics (kW). Generation of long term statistics will be suppressed unless the unit changes back to kW or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.charger_power_reactive_import (var) cannot be converted to the unit of previously compiled statistics (kW). Generation of long term statistics will be suppressed unless the unit changes back to kW or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
***************
Notifications show 2 x errors. 1 for import and 1 for export.
No issue prior to 0.6.3

Charger::
ZJ Beny Occp version
Firmware: 1.0.35

rbb-dev, 2025-01-07:
Reactive power cannot be kW.
Just press the fix button to delete the old values.

BitsLeftOver, 2025-01-07:
Many thanks for the direction.
Cheers.

rbb-dev, 2025-01-22:
@BitsLeftOver If everything is solved, can you please close this issue request :)

BitsLeftOver, 2025-01-22:
Brilliant work with this fix. Thank you.

closed 2025-01-22 by BitsLeftOver