lbbrhzn/ocpp #1483: WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration will stop working

arh-au, 2025-01-11:
**Describe the bug**
Warning appearing in Home Assistant logs during startup:

WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JETVDSDMS8Q5CGSNK763N169'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JETVDSDMS8Q5CGSNK763N169')} at custom_components/ocpp/button.py, line 57: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues

**To Reproduce**
Steps to reproduce the behavior:
1. Upgraded to HA 2025.1.2

**Expected behavior**
No warning to appear

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - RPI 3b+
 - occp v0.7.0
 - Core 2025.1.2
 - Supervisor 2024.12.3
 - Operating System 14.1
 - Frontend 20250109.0

jayzee225, 2025-01-28:
Same issue for me after updating to Core 2025.x
I went to HACS / OCPP / Download again and after an install and reboot of HA it worked again. 
Hope this will help.

caitotheonlypotato, 2025-01-28:
> Same issue for me after updating to Core 2025.x I went to HACS / OCPP / Download again and after an install and reboot of HA it worked again. Hope this will help.

Sadly for my installation, this did not work.  I've reinstalled several times also.  I can see all the sensors and switches as per normal, but the charge rate can't be adjusted

closed 2025-04-06 by github-actions[bot]