lbbrhzn/ocpp #1484: Charger not detected - Only charge current control available

Fannangir, 2025-01-12:
Tried to add the charger, I believe it is not detected correctly, Nexus EV charge.
I can control the charge current only, all the rest is unavailable.
Found only the following in the log:
2025-01-12 21:15:25.127 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JHDZ2G94X57D3PZJF646Z7A2'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JHDZ2G94X57D3PZJF646Z7A2')} at custom_components/ocpp/sensor.py, line 77: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
2025-01-12 21:15:25.629 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JHDZ2G94X57D3PZJF646Z7A2'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JHDZ2G94X57D3PZJF646Z7A2')} at custom_components/ocpp/switch.py, line 75: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
2025-01-12 21:15:25.651 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JHDZ2G94X57D3PZJF646Z7A2'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JHDZ2G94X57D3PZJF646Z7A2')} at custom_components/ocpp/number.py, line 69: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
2025-01-12 21:15:25.663 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JHDZ2G94X57D3PZJF646Z7A2'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JHDZ2G94X57D3PZJF646Z7A2')} at custom_components/ocpp/button.py, line 57: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues

Images:
[image]

[image]

Thanks.

altserg, 2025-01-13:
You need to enable ocpp lib logging to see if there is any communication between charger and system:
In configuration.yaml add
logger:
  logs:
    ocpp: info

dudan166, 2025-01-13:
Hello, the same problem with my TARY charger. Enable OCPP lib logging give me no results.
 Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JGVKBNYTDE81A1F9P94S7W6E'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JGVKBNYTDE81A1F9P94S7W6E')} at custom_components/ocpp/sensor.py, line 77: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JGVKBNYTDE81A1F9P94S7W6E'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JGVKBNYTDE81A1F9P94S7W6E')} at custom_components/ocpp/switch.py, line 75: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JGVKBNYTDE81A1F9P94S7W6E'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JGVKBNYTDE81A1F9P94S7W6E')} at custom_components/ocpp/number.py, line 69: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JGVKBNYTDE81A1F9P94S7W6E'), with device info: {'identifiers': {('ocpp', 'charger')}, 'via_device': ('ocpp', '01JGVKBNYTDE81A1F9P94S7W6E')} at custom_components/ocpp/button.py, line 57: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues

altserg, 2025-01-13:
Probably the charger does not connect to the system. Please check if you configured it properly according to the documentation

Fannangir, 2025-01-13:
Got my answer.
Charger does not support OCPP although available in its settings.
Thanks for your comment.

altserg, 2025-01-14:
I think you can close the issue then

closed 2025-01-14 by Fannangir