lbbrhzn/ocpp #1487: MeterValuesSampledDataMaxLength is not respected for ocppv1.6

altserg, 2025-01-14:
**Describe the bug**
Measurands autodetection in ocppv16.py assumes that there are no limitations set by charger MeterValuesSampledDataMaxLength key. However, Growatt Thor does has the limit of 5. When the measurands are configured in MeterValuesSampledData command, the integrations sends too many of them. The charger has to decline this request, but it succeeds, probably causing corruption of charger's internal data structures. After then results of some calls like NumberOfConnectors are returning random values causing further failures (see issue #1452 for example). Also, not all requested variables are reported back to the server

**To Reproduce**
Use the integration with Growatt Thor charger

**Expected behavior**
MeterValuesSampledDataMaxLength limit is respected, if exists (the key is optional in OCPP v1.6 protocol)

**Additional context**
Proposal to fix:
- check if corresponding sensor is enabled in HASS before trying to autodetect the measurand. This will give the user control to discover which configuration is usable

rbb-dev, 2025-02-13:
@altserg Hi.
The code doesn't strictly enforce the charger's measurand limit because it doesn't know which items to exclude to fit within that maximum. If your charger has a limit, i would recommend turning off auto-detection and manually selecting the specific measurands you need.

Please consider closing this issue if this is a sufficient answer for you :)

altserg, 2025-02-15:
I do not completely agree.
The code should at least give warning in the log if maximum limit is reached. Or use softer user controllable method that I proposed in the bug description. 
Without this some chargers may fail, and there is no clue why. See my other summitted bugs. I can hack the code locally, but many people cannot. 
Also, it's impossible to force every (cheap Chinese) company to adhere to the standard. The integration code should be robust assuming that the manufacturers implement the standard not properly. Look at ZHA and their quirk mechanism

rbb-dev, 2025-02-16:
Ask dr38 or lbbrhzn for advice, but I think the documentation for the workaround would be the best option.

drc38, 2025-02-16:
I'm in agreement with @rbb-dev, there is already an easy user fix/workaround for this issue.

altserg, 2025-02-18:
Closing per discussion

closed 2025-02-18 by altserg