lbbrhzn/ocpp #1489: Cannot reach the OCPP Backend

maryjam01, 2025-01-18:
No Connection available

Here the logs from Home Assistant:

Logger: homeassistant.helpers.frame
Quelle: helpers/frame.py:324
Erstmals aufgetreten: 11:01:47 (4 Vorkommnisse)
Zuletzt protokolliert: 11:01:47

Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JF4N69MAGFXS3YAQP93RC6KQ'), with device info: {'identifiers': {('ocpp', 'MaNiCharger')}, 'via_device': ('ocpp', '01JF4N69MAGFXS3YAQP93RC6KQ')} at custom_components/ocpp/sensor.py, line 77: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JF4N69MAGFXS3YAQP93RC6KQ'), with device info: {'identifiers': {('ocpp', 'MaNiCharger')}, 'via_device': ('ocpp', '01JF4N69MAGFXS3YAQP93RC6KQ')} at custom_components/ocpp/switch.py, line 75: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JF4N69MAGFXS3YAQP93RC6KQ'), with device info: {'identifiers': {('ocpp', 'MaNiCharger')}, 'via_device': ('ocpp', '01JF4N69MAGFXS3YAQP93RC6KQ')} at custom_components/ocpp/number.py, line 69: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues
Detected that custom integration 'ocpp' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('ocpp', '01JF4N69MAGFXS3YAQP93RC6KQ'), with device info: {'identifiers': {('ocpp', 'MaNiCharger')}, 'via_device': ('ocpp', '01JF4N69MAGFXS3YAQP93RC6KQ')} at custom_components/ocpp/button.py, line 57: async_add_devices(entities, False). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/lbbrhzn/ocpp/issues

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: homeassistant
Quelle: custom_components/ocpp/ocppv16.py:212
Integration: Open Charge Point Protocol (OCPP) (Dokumentation, Probleme)
Erstmals aufgetreten: 11:07:16 (1 Vorkommnisse)
Zuletzt protokolliert: 11:07:16

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/queues.py", line 186, in get
    await getter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 421, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        call.unique_id, self._response_timeout
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/ocpp/chargepoint.py", line 463, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 468, in _get_specific_response
    return await self._get_specific_response(unique_id, timeout_left)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/chargepoint.py", line 463, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 455, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/chargepoint.py", line 334, in post_connect
    await self.fetch_supported_features()
  File "/config/custom_components/ocpp/chargepoint.py", line 268, in fetch_supported_features
    self._attr_supported_features = await self.get_supported_features()
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 212, in get_supported_features
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 425, in call
    raise asyncio.TimeoutError(
    ...<2 lines>...
    )
TimeoutError: Waited 10s for response on [2,"9a299be3-d308-44fd-9467-7a13a0c963db","GetConfiguration",{"key":["SupportedFeatureProfiles"]}].

rbb-dev, 2025-01-18:
I'm not sure what charger you're using, but it's not responding or locking when it receives a GetConfiguration query for SupportedFeatureProfiles.
whats your charger doing when this happens? it might not be supported?

maryjam01, 2025-01-18:
Hi,

it is webasto live and before a few days it was working fine. This is the
configuration:

[image: image.png]

Greets
Marian

rbb-dev, 2025-01-18:
Last update to this branch was 2 weeks ago.
Did you do the update few days ago?
If so, download previous releases from https://github.com/lbbrhzn/ocpp/releases and see which one starts working for you.

rbb-dev, 2025-01-22:
@maryjam01 If everything is solved, can you please close this issue request :)

closed 2025-01-22 by maryjam01

maryjam01, 2025-01-23:
Hi,

in version 0.6.2 I have already the state that the integration doesn`t
load. Here is the error from the protokoll:

Logger: homeassistant.setup
Quelle: setup.py:334
Erstmals aufgetreten: 14:47:58 (1 Vorkommnisse)
Zuletzt protokolliert: 14:47:58

Setup failed for custom integration 'ocpp': Unable to import component:
cannot import name 'ConnectorStatusType' from 'ocpp.v201.enums'
(/usr/local/lib/python3.13/site-packages/ocpp/v201/enums.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in
async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59,
in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in
_get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in
protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in
import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in
_find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in
_call_with_frames_removed
  File "/config/custom_components/ocpp/__init__.py", line 14, in <module>
    from .api import CentralSystem
  File "/config/custom_components/ocpp/api.py", line 18, in <module>
    from .ocppv201 import ChargePoint as ChargePointv201
  File "/config/custom_components/ocpp/ocppv201.py", line 19, in <module>
    from ocpp.v201.enums import (
    ...<17 lines>...
    )
ImportError: cannot import name 'ConnectorStatusType' from
'ocpp.v201.enums'
(/usr/local/lib/python3.13/site-packages/ocpp/v201/enums.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in
_async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in
async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in
async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in
_get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in
protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in
import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in
_find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in
_call_with_frames_removed
  File "/config/custom_components/ocpp/__init__.py", line 14, in <module>
    from .api import CentralSystem
  File "/config/custom_components/ocpp/api.py", line 18, in <module>
    from .ocppv201 import ChargePoint as ChargePointv201
  File "/config/custom_components/ocpp/ocppv201.py", line 19, in <module>
    from ocpp.v201.enums import (
    ...<17 lines>...
    )
ImportError: cannot import name 'ConnectorStatusType' from
'ocpp.v201.enums'
(/usr/local/lib/python3.13/site-packages/ocpp/v201/enums.py)

Regards
Marian

maryjam01, 2025-01-23:
Sorry, and this warning:

Logger: homeassistant.util.loop
Quelle: util/loop.py:77
Erstmals aufgetreten: 14:47:58 (1 Vorkommnisse)
Zuletzt protokolliert: 14:47:58
Detected blocking call to import_module with args
('custom_components.ocpp',) in
/usr/src/homeassistant/homeassistant/loader.py, line 1075:
ComponentProtocol, importlib.import_module(self.pkg_path) inside the event
loop; This is causing stability issues. Please create a bug report at
https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For
developers, please see
https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module
Traceback (most recent call last): File "<frozen runpy>", line 198, in
_run_module_as_main File "<frozen runpy>", line 88, in _run_code File
"/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module>
sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__main__.py",
line 213, in main exit_code = runner.run(runtime_conf) File
"/usr/src/homeassistant/homeassistant/runner.py", line 154, in run return
loop.run_until_complete(setup_and_run_hass(runtime_config)) File
"/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in
run_until_complete self.run_forever() File
"/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in
run_forever self._run_once() File
"/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once
handle._run() File "/usr/local/lib/python3.13/asyncio/events.py", line 89,
in _run self._context.run(self._callback, *self._args) File
"/usr/src/homeassistant/homeassistant/setup.py", line 165, in
async_setup_component result = await _async_setup_component(hass, domain,
config) File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in
_async_setup_component component = await integration.async_get_component()
File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in
async_get_component comp = self._get_component() File
"/usr/src/homeassistant/homeassistant/loader.py", line 1075, in
_get_component ComponentProtocol, importlib.import_module(self.pkg_path)