lbbrhzn/ocpp #1491: OCPP - Warning - Stop transaction failed with response Rejected

RafAustralia, 2025-01-19:
Hi there - I am getting persistent notifications every hour or so that the OCPP has failed, with the : - Warning - Stop transaction failed with response Rejected

Has anyone come across that?

[image]

-->

I am running BENNYZJ OCPP V16 with firmware 1.0.0.34
Homeassistant version 2025.1.2

-->

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

Here are examples of ERRORS:
.........................................................................................................
Logger: custom_components.ocpp
Source: custom_components/ocpp/ocppv16.py:790
integration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [issues](https://github.com/lbbrhzn/ocpp/issues))
First occurred: 7:51:42 AM (1 occurrences)
Last logged: 7:51:42 AM

Stop transaction received for unknown transaction id=1737315090
.........................................................................................................
Logger: custom_components.ocpp
Source: custom_components/ocpp/ocppv16.py:412
integration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [issues](https://github.com/lbbrhzn/ocpp/issues))
First occurred: 5:31:32 AM (2 occurrences)
Last logged: 5:32:22 AM

Failed with response: Rejected
.........................................................................................................

WARNING:
.........................................................................................................
Logger: custom_components.ocpp
Source: custom_components/ocpp/ocppv16.py:629
integration: Open Charge Point Protocol (OCPP) ([documentation](https://github.com/lbbrhzn/ocpp/blob/main/README.md), [issues](https://github.com/lbbrhzn/ocpp/issues))
First occurred: 5:31:31 AM (161 occurrences)
Last logged: 7:51:42 AM

Unknown transaction detected with id=1737315090
.........................................................................................................

## Debug log

I have just enabled the debug log so it may not show any previous errors yet....

but this is from live Ha log:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/queues.py", line 186, in get
    await getter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 421, in call
    response = await self._get_specific_response(
[... 298 lines cut ...]
  File "/config/custom_components/ocpp/switch.py", line 139, in async_turn_off
    resp = await self.central_system.set_charger_state(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.cp_id, self.entity_description.off_action
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/ocpp/api.py", line 243, in set_charger_state
    resp = await self.charge_points[cp_id].stop_transaction()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 427, in stop_transaction
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 425, in call
    raise asyncio.TimeoutError(
    ...<2 lines>...
    ).

```

Could anyone comment of what you think is going on here?

Much appreciated.

Raf

rbb-dev, 2025-01-19:
hi @RafAustralia.

So far, all I can see in the logs is that the charger did not respond to the RemoteStopTransaction message for 10 seconds.
All other errors and exceptions in the logs are caused because of this timeout.
Maybe some logs are missing, but there are no "Rejected" responses from the charger.

In regards to your notification:
"Warning: Stop transaction failed with response Rejected"
Do you have automation that triggers a stop transaction every hour? If no transaction is ongoing, you will get Rejected back from the charger.

RafAustralia, 2025-01-19:
> hi [@RafAustralia](https://github.com/RafAustralia).
>
> So far, all I can see in the logs is that the charger did not respond to the RemoteStopTransaction message for 10 seconds. All other errors and [...]

Cheers @rbb-dev

I am in process of discovery - so with all of that - it has just started to happen persistently.
Yes we have automation controlling these... is there a way of stopping these notifications from happening other than hushing them? or stopping them all together?

Below are examples of what I think relates to rejection of calls...

EXAMPLE OF REJECTED CALLS FROM THE LOGS:

2025-01-19 17:10:07.095 INFO (MainThread) [homeassistant.components.automation.emhass_ev_charging_automation] EMHASS EV Charging Automation: Running automation actions
2025-01-19 17:10:07.096 INFO (MainThread) [homeassistant.components.automation.emhass_ev_charging_automation] EMHASS EV Charging Automation: Choose at step 1: default: Running automation actions
2025-01-19 17:10:07.096 INFO (MainThread) [homeassistant.components.automation.emhass_ev_charging_automation] EMHASS EV Charging Automation: Choose at step 1: default: Skipped disabled step device
2025-01-19 17:10:07.096 INFO (MainThread) [homeassistant.components.automation.emhass_ev_charging_automation] EMHASS EV Charging Automation: Choose at step 1: default: Executing step device automation
2025-01-19 17:10:07.097 INFO (MainThread) [ocpp] 240700203: send [2,"a1f32782-9b46-48c3-b0e3-18898794f584","RemoteStopTransaction",{"transactionId":1737195514}]
2025-01-19 17:10:07.239 INFO (MainThread) [ocpp] 240700203: receive message [3,"a1f32782-9b46-48c3-b0e3-18898794f584",{"status":"Rejected"}]
2025-01-19 17:10:07.242 WARNING (MainThread) [custom_components.ocpp] Failed with response: Rejected
2025-01-19 17:10:07.244 INFO (MainThread) [homeassistant.components.automation.emhass_ev_charging_automation] EMHASS EV Charging Automation: Executing step call service
2025-01-19 17:10:07.246 INFO (MainThread) [ocpp] 240700203: send [2,"b0450ba0-82f2-4378-9374-1425b0712948","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2025-01-19 17:10:07.405 INFO (MainThread) [ocpp] 240700203: receive message [3,"b0450ba0-82f2-4378-9374-1425b0712948",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"}]}]
2025-01-19 17:10:07.408 INFO (MainThread) [custom_components.ocpp] Charger supports setting the following units: Current,Power
2025-01-19 17:10:07.408 INFO (MainThread) [custom_components.ocpp] If more than one unit supported default unit is Amps
2025-01-19 17:10:07.409 INFO (MainThread) [ocpp] 240700203: send [2,"a45502ed-674b-4a6f-9308-70e18f719f92","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2025-01-19 17:10:07.492 INFO (MainThread) [ocpp] 240700203: receive message [3,"a45502ed-674b-4a6f-9308-70e18f719f92",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"10"}]}]
2025-01-19 17:10:07.498 INFO (MainThread) [ocpp] 240700203: send [2,"d0ef5695-6f28-46eb-a0f4-131fb5f3df84","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":8,"stackLevel":10,"chargingProfileKind":"Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0.0}]}}}]
2025-01-19 17:10:07.635 INFO (MainThread) [ocpp] 240700203: receive message [3,"d0ef5695-6f28-46eb-a0f4-131fb5f3df84",{"status":"Accepted"}]

.................................................................................................................................................................................

2025-01-19 17:02:33.463 INFO (MainThread) [ocpp] 240700203: send [3,"d7r2fu1g43ewcrw80fyvn382",{}]
2025-01-19 17:02:33.465 INFO (MainThread) [ocpp] 240700203: send [2,"2aa09b47-f1fd-4b47-a706-388cdbb4cb59","RemoteStopTransaction",{"transactionId":1737195514}]
2025-01-19 17:02:33.587 INFO (MainThread) [ocpp] 240700203: receive message [3,"2aa09b47-f1fd-4b47-a706-388cdbb4cb59",{"status":"Rejected"}]
2025-01-19 17:02:33.662 WARNING (MainThread) [custom_components.ocpp] Failed with response: Rejected
2025-01-19 17:02:33.665 INFO (MainThread) [homeassistant.components.automation.emhass_ev_charging_automation] EMHASS EV Charging Automation: Executing step call service
2025-01-19 17:02:33.666 INFO (MainThread) [ocpp] 240700203: send [2,"4c3201e4-6885-4b9f-b56b-3569a90bec9d","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2025-01-19 17:02:33.843 INFO (MainThread) [ocpp] 240700203: receive message [3,"4c3201e4-6885-4b9f-b56b-3569a90bec9d",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"}]}]
2025-01-19 17:02:33.843 INFO (MainThread) [custom_components.ocpp] Charger supports setting the following units: Current,Power
2025-01-19 17:02:33.843 INFO (MainThread) [custom_components.ocpp] If more than one unit supported default unit is Amps
2025-01-19 17:02:33.844 INFO (MainThread) [ocpp] 240700203: send [2,"a7fefcfa-ba55-4111-b0c1-23081cc9b439","GetConfiguration",{"key":["ChargeProfileMaxStackLevel"]}]
2025-01-19 17:02:33.936 INFO (MainThread) [ocpp] 240700203: receive message [3,"a7fefcfa-ba55-4111-b0c1-23081cc9b439",{"configurationKey":[{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"10"}]}]
[... cut ...] Relative","chargingProfilePurpose":"ChargePointMaxProfile","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0.0}]}}}]
2025-01-19 17:02:34.186 INFO (MainThread) [ocpp] 240700203: receive message [3,"caf6e5c8-2ba1-4936-81b8-4be3ec017745",{"status":"Accepted"}]

.................................................................................................................................................................................

Hope these help?

caitotheonlypotato, 2025-01-20:
I've been receiving the same over the last week or so.  From one day to the next no changes made to HA or the integration.  My installation still shows all of the correct values in sensors and charge control + Availability are controllable as is reset and unlock, however these identical errors are created if the max current is adjusted.  I can still adjust charge current on the fly during a charge if I use the standard web management interface of the EVSE.  This stopped working on v2024.12.1.
I've since tried the latest HA version, updating OCPP integration, removing the integration and repository and reinstalling.  The integration reinstalls perfectly and all works as shown except for Max Current.  I'm at a loss to proceed.

[image]

[image]

[image]

[image]

rbb-dev, 2025-01-20:
Mate, you have an issue with your automation setup.
You're trying to stop transaction while youre still pumping config to the charger.
2025-01-19 17:10:07.097 INFO (MainThread) [ocpp] 240700203: send [2,"a1f32782-9b46-48c3-b0e3-18898794f584","RemoteStopTransaction",{"transactionId":1737195514}]
2025-01-19 17:10:07.239 INFO (MainThread) [ocpp] 240700203: receive message [3,"a1f32782-9b46-48c3-b0e3-18898794f584",{"status":"Rejected"}]
2025-01-19 17:10:07.242 WARNING (MainThread) [custom_components.ocpp] Failed with response: Rejected
2025-01-19 17:10:07.244 INFO (MainThread) [homeassistant.components.automation.emhass_ev_charging_automation] EMHASS EV Charging Automation: Executing step call service
2025-01-19 17:10:07.246 INFO (MainThread) [ocpp] 240700203: send [2,"b0450ba0-82f2-4378-9374-1425b0712948","GetConfiguration",{"key":["ChargingScheduleAllowedChargingRateUnit"]}]
2025-01-19 17:10:07.405 INFO (MainThread) [ocpp] 240700203: receive message [3,"b0450ba0-82f2-4378-9374-1425b0712948",{"configurationKey":[{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"}]}]

thats not right :)
Automation definitely has issues.
you should change your automation to stop charging only when the connector state is Charging.

RafAustralia, 2025-01-20:
> Mate, you have an issue with your automation setup. You're trying to stop transaction while youre still pumping config to the charger. 2025-01-19 17:10:07.097 INFO (MainThread) [ocpp] 240700203: [...]

cheers @rbb-dev I will definitely look at that asap... much appreciate the reply.

rbb-dev, 2025-01-20:
Check this link:
https://github.com/lbbrhzn/ocpp/wiki/debugging
you might have logging enabled.
in HA you can quiet down anything, just have to use logger: 
https://www.home-assistant.io/integrations/logger/

rbb-dev, 2025-01-22:
@RafAustralia If you're ok now, please close this issue :)

RafAustralia, 2025-01-22:
Hey @rbb-dev 

Thank u for that

Me and @purcell-labs have altered the automation. 
We moved elements of actions around and I still continue to get rejected call notifications. Maybe 5-6 a day. 

Inverter works. Starts. Charges adjusts current. Stops. But messages still happen throughout the day and especially when ev is not around. 

Im monitoring situation. 

Raf

rbb-dev, 2025-01-22:
@RafAustralia i get it..
Still an issue with automation.
Need to check when this happens and with which automation.
Look at mine, I don't send stop unless connector state is Charging.

```python
alias: Charger DC - Stop Charging at 06:00AM
description: Charger DC - Stop Charging at 06:00AM
triggers:
  - trigger: time
    at: "06:00:00"
conditions:
  - condition: state
    entity_id: sensor.charger_dc_status_connector
    state: Charging
actions:
  - type: turn_off
    device_id: 1317fe2e42d9ef2626d5b12c2eb58188
    entity_id: b5ddbe1253875c6187bf8343b3e1759e
    domain: switch
mode: single

```

RafAustralia, 2025-01-22:
> [@RafAustralia](https://github.com/RafAustralia) i get it.. Still an issue with automation. Need to check when this happens and with which automation. Look at mine, I don't send stop unless connector state is Charging.
> 
> alias: Charger DC - Stop Charging at 06:00AM
> description: Charger DC - Stop Charging at 06:00AM
> triggers:
>   - trigger: time
>     at: "06:00:00"
> conditions:
>   - condition: state
>     entity_id: sensor.charger_dc_status_connector
>     state: Charging
> actions:
>   - type: turn_off
>     device_id: 1317fe2e42d9ef2626d5b12c2eb58188
>     entity_id: b5ddbe1253875c6187bf8343b3e1759e
>     domain: switch
> mode: single

Hey. Cheers @rbb-dev i will send u copy of my automation tomorrow when Im in front of my laptop
Too hard from the phone. 

We did fix automation components but. 
Or at least we think we have

Raf

RafAustralia, 2025-01-22:
> Hey. Cheers [@rbb-dev](https://github.com/rbb-dev) i will send u copy of my automation tomorrow when Im in front of my laptop Too hard from the phone.
>
> We did fix automation components but. Or at [...]

@rbb-dev Good Morning

@purcell-lab (cc Mark Purcell who has been helping me with this)

As promised the copy of automation code we have - would love your input

```
alias: EMHASS EV Charging Automation
description: ""
triggers:
  - trigger: state
    entity_id:
      - sensor.p_deferrable2
    enabled: true
    to: null
  - trigger: time_pattern
    minutes: /15
    enabled: false
  - trigger: state
    entity_id:
      - sensor.charger_status_connector
    to: null
conditions: []
actions:
  - choose:
      - conditions:
          - condition: numeric_state
            entity_id: sensor.p_deferrable2
            above: 200
        sequence:
          - device_id: b5a474be009c573aab9eb41d8c994c3a
            domain: select
            entity_id: ef9ac20600538ace9cc5b41233119941
            type: select_option
            option: immediate_charging
            enabled: false
          - type: turn_on
            device_id: a4629cb1037bf5edc0504389303d916f
            entity_id: 9ac8a80e1ddf8015b7b7271fd72aff77
            domain: switch
          - action: number.set_value
            metadata: {}
            data:
              value: >-
                {{(states('sensor.p_deferrable2')|float(0)/3/states('sensor.charger_voltage')|float(230))|int(0)}}
            target:
              entity_id: number.charger_maximum_current
    default:
      - device_id: b5a474be009c573aab9eb41d8c994c3a
        domain: select
        entity_id: ef9ac20600538ace9cc5b41233119941
        type: select_option
        option: delayed_charging
        enabled: false
      - if:
          - condition: state
            entity_id: sensor.charger_status_connector
            state: Charging
        then:
          - type: turn_off
            device_id: a4629cb1037bf5edc0504389303d916f
            entity_id: 9ac8a80e1ddf8015b7b7271fd72aff77
            domain: switch
mode: single
```

rbb-dev, 2025-01-25:
@RafAustralia sent you an email

closed 2025-04-04 by github-actions[bot]