lbbrhzn/ocpp #1499: Integration crashes on json value parsing

stefa168, 2025-01-24:
**Describe the bug**
I have recently configured my wallbox to connect to the integration, but shortly after it shows as available (and some of the basic informations about it are shown on the device section) it goes back to unavailable.
This cycle repeats forever, and from the logs it looks like the integration is crashing:

```log
Registratore: custom_components.ocpp
Fonte: custom_components/ocpp/chargepoint.py:531
Integrazione: Open Charge Point Protocol (OCPP) (documentazione, problemi)
Prima occorrenza: 13:28:28 (15 occorrenze)
Ultimo accesso: 13:36:34

Unexpected exception in connection to 'charger': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=b8bd4bea-4ab8-415b-8431-0fb527bc86e2, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Unexpected exception in connection to 'charger': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=54ec6265-0bb6-48eb-bbf1-5bbefc9aeb33, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Unexpected exception in connection to 'charger': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=b4228505-be99-4b79-90d2-241fdd89dcef, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Unexpected exception in connection to 'Scame%20Wallbox': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=736f01a4-48a5-4eb3-9323-10237e25d833, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Unexpected exception in connection to 'Scame%20Wallbox': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=4c0d0730-84c3-4ca7-9ac2-02c83998f5ce, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 239, in _validate_payload
    validator.validate(message.payload)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/jsonschema/validators.py", line 451, in validate
    raise error
jsonschema.exceptions.ValidationError: 1 is not of type 'string'

Failed validating 'type' in schema['properties']['configurationKey']['items']['properties']['value']:
    {'type': 'string', 'maxLength': 500}

On instance['configurationKey'][0]['value']:
    1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/chargepoint.py", line 531, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/chargepoint.py", line 335, in post_connect
    num_connectors: int = await self.get_number_of_connectors()
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 99, in get_number_of_connectors
    return await self.get_configuration(ckey.number_of_connectors.value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 525, in get_configuration
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 437, in call
    await validate_payload(response, self._ocpp_version)
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 179, in validate_payload
    await asyncio.get_event_loop().run_in_executor(
        None, _validate_payload, message, ocpp_version
    )
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 242, in _validate_payload
    raise TypeConstraintViolationError(
        details={"cause": e.message, "ocpp_message": message}
    )
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12),  {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=1e302d6c-faa4-44ff-a999-9eed9fd665d3, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}
```

The OCPP integration is v0.7.0
My wallbox is a SCAME SBC-0 with firmware 1.4.2.

If any other information is needed, I'll be happy to provide them!

rbb-dev, 2025-01-25:
@stefa168 
Try selecting "Skip OCPP schema validation" adding ocpp device to HA.
Your charger is the first one I've seen that returns NumberOfConnectors as an integer and not a string.

stefa168, 2025-01-25:
Thanks for the suggestion!

> Your charger is the first one I've seen that returns NumberOfConnectors as an integer and not a string.

Yeah I figured it wasn't exactly something expected 😄 

Anyway, I tried your suggestion, however the error still persists
```
Questo errore ha avuto origine da un'integrazione personalizzata.

Registratore: custom_components.ocpp
Fonte: custom_components/ocpp/chargepoint.py:531
Integrazione: Open Charge Point Protocol (OCPP) (documentazione, problemi)
Prima occorrenza: 24 gennaio 2025 alle ore 13:28:28 (39 occorrenze)
Ultimo accesso: 11:27:20

Unexpected exception in connection to 'Scame%20Wallbox': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=bcf5dcb5-1098-41c0-99cd-043802e13a0e, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Unexpected exception in connection to 'Scame%20Wallbox': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=87ad8461-f7e5-49f9-a2b0-585bb839821b, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Unexpected exception in connection to 'Scame%20Wallbox': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=1581527d-0936-4f4e-b3f1-8bc27f17a456, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Unexpected exception in connection to 'Scame%20Wallbox': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=d690676d-5ab8-4214-b4f0-f41ce443fb53, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Unexpected exception in connection to 'Scame%20Wallbox': 'TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12), {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=318afd73-e598-470a-bca2-0b492e780052, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}'
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 239, in _validate_payload
    validator.validate(message.payload)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/jsonschema/validators.py", line 451, in validate
    raise error
jsonschema.exceptions.ValidationError: 1 is not of type 'string'

Failed validating 'type' in schema['properties']['configurationKey']['items']['properties']['value']:
    {'type': 'string', 'maxLength': 500}

On instance['configurationKey'][0]['value']:
    1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/chargepoint.py", line 531, in run
    await asyncio.gather(*self.tasks)
  File "/config/custom_components/ocpp/chargepoint.py", line 335, in post_connect
    num_connectors: int = await self.get_number_of_connectors()
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 99, in get_number_of_connectors
    return await self.get_configuration(ckey.number_of_connectors.value)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 525, in get_configuration
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 437, in call
    await validate_payload(response, self._ocpp_version)
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 179, in validate_payload
    await asyncio.get_event_loop().run_in_executor(
        None, _validate_payload, message, ocpp_version
    )
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.13/site-packages/ocpp/messages.py", line 242, in _validate_payload
    raise TypeConstraintViolationError(
        details={"cause": e.message, "ocpp_message": message}
    )
ocpp.exceptions.TypeConstraintViolationError: TypeConstraintViolationError: Payload for Action is syntactically correct but at least one of the fields violates data type constraints (e.g. “somestring”: 12),  {'cause': "1 is not of type 'string'", 'ocpp_message': <CallResult - unique_id=1e302d6c-faa4-44ff-a999-9eed9fd665d3, action=GetConfiguration, payload={'configurationKey': [{'key': 'NumberOfConnectors', 'readonly': True, 'value': 1}]}>}
```

Is the value absolutely needed to be a string, or could we cast it into one if needed?

rbb-dev, 2025-01-25:
It looks like it is still validating.. Did you restart HA after deleting and re-adding CP?

In regards to casting...
This project cannot address this, as it involves a change to the OCPP Python library itself.
The exception is being generated by that library, not by the Home Assistant OCPP integration.

stefa168, 2025-01-25:
I tried again right now, I deleted the configuration, then restarted HA and then re-did the setup but the issue persisted.
I will open an issue in the Python Library Repo. Thanks for the support!

rbb-dev, 2025-01-26:
Mobilityhouse is right, its your charger, your best bet is to ask them to fix it or fork ocpp library and this component for it.
Give this to your charger support people, it clearly states that response must be string
https://ocpp-spec.org/schemas/v1.6/#tag/getconfigurationresponse

stefa168, 2025-01-26:
Yeah, I think you're right. I will try asking the manufacturer.

In the meantime, why doesn't the "Skip OCPP schema validation" actually skip the validation?
It might help at least in the meantime to mitigate the issue while I get a fix from the producer...

rbb-dev, 2025-01-26:
I'm not sure. Looking at the validation code in ocpp, it cannot skip variable type changes.
ocpp library and HA ocpp component both expect it to be a string, there is no casing for that variable.

closed 2025-04-05 by github-actions[bot]