lbbrhzn/ocpp #1503: BootNotification loop

scauwe, 2025-01-28:
**Describe the bug**
It seems that not all charger support processing massages prior to having the BootNotification being processed.

**To Reproduce**
1. Charger: Wallbox  CPB1 (CPB1-S-2-4)
2. Reboot the charger
3. The charger and the OCPP server get stuck in a loop: 
 * the charger keeps on sending BootNotification (answered by the OCPP server)
 * the OCPP keeps on sending GetConfiguration (which times out)
4. The charger does not want to connect

**Expected behavior**
The OCPP server should wait to send any messages until the BootNotification is processed. After this, the charger is accepting GetConfiguration 

**Additional context**
After a power outage, my charger did no longer connect.
I change line 333 in chargepoint.py:
- old code: await asyncio.sleep(2)
- new code: await asyncio.sleep(20)
As a result, the BootNotification is not processed, and the GetConfiguration is send 20 seconds after the web socket is opened by the client. The value of 20 could probably be lowered.
Ideally, this portion of the code should be called after processing the first OCPP message from the charger (BootNotification or other message).

delainus, 2025-01-29:
I have same problem after update wallbox commander 2

Firmware 6.7.17
Wall Box Chargers ⸱ CMX2-0-2-4
OCPP: 0.7.0

scauwe, 2025-01-29:
Firmware of my wallbox charger is 6.6.20. I do assume that our firmware is not the same (my charger claims to be on the latest version). I do have auto update disabled, but also the auto update of HA and the ocpp extension is disabled.
The only thing that 'changed' in my setup was a power outage, after which I noticed the loop.

delainus, 2025-01-31:
Wallbox webpage:
Release of 13th-January-2025
Commander 2 – Software version 6.7.17 Commander 2 is receiving all the improvements that other products received in previous software…

Release of 7th-October-2024
Copper SB – Software version 6.6.20 Fixed an issue where the charger could lose connectivity after being connected for some…

Then... now, Your problems are now my problems. :-)

Somewhere else, a man comments that his wallbox updated automatically, even though he had autoupdate turned off.

delainus, 2025-01-31:
Este error se originó a partir de una integración personalizada.

Registrador: homeassistant
Fuente: custom_components/ocpp/ocppv16.py:212
integración: Open Charge Point Protocol (OCPP) (documentación, problemas)
Ocurrió por primera vez: 09:13:37 (2 ocurrencias)
Última vez registrado: 09:15:53

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/queues.py", line 186, in get
    await getter
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 421, in call
    response = await self._get_specific_response(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        call.unique_id, self._response_timeout
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/ocpp/chargepoint.py", line 463, in _get_specific_response
    resp = await super()._get_specific_response(unique_id, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 455, in _get_specific_response
    response = await asyncio.wait_for(self._response_queue.get(), timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/ocpp/chargepoint.py", line 334, in post_connect
    await self.fetch_supported_features()
  File "/config/custom_components/ocpp/chargepoint.py", line 268, in fetch_supported_features
    self._attr_supported_features = await self.get_supported_features()
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/ocpp/ocppv16.py", line 212, in get_supported_features
    resp = await self.call(req)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/ocpp/charge_point.py", line 425, in call
    raise asyncio.TimeoutError(
    ...<2 lines>...
    )
TimeoutError: Waited 10s for response on [2,"25de1b2a-c4d9-404f-8530-1fedaef41d44","GetConfiguration",{"key":["SupportedFeatureProfiles"]}].

scauwe, 2025-01-31:
> Somewhere else, a man comments that his wallbox updated automatically, even though he had autoupdate turned off.

That can explain why my wallbox is up to date. I don't remember installing any updates in October or later, and can explain why it 'suddenlyl stopped working...

drc38, 2025-02-01:
You could also try commenting out line 600 of chargepoint.py 
`self.hass.async_create_task(self.post_connect())`

scauwe, 2025-02-01:
I think that commenting out that line will result in:
- some services not getting registered
- some other stuff that can be needed (eg getting number of connectors)
If not doing that, some features might be missing or the ocpp might not be fully operational.

drc38, 2025-02-01:
I doubt it, if a charger reboots the websocket is lost. It will then reconnect and call post_connect() if the connection sequence was not completed successfully. By calling post_connect() in line 600 there could be a race condition or similar going on.

delainus, 2025-02-09:
Today downgrade firmware wallbox commander 2     V6.7.17 to V6.7.15
Now, ocpp work great again.

Delorien84, 2025-02-14:
> Today downgrade firmware wallbox commander 2 V6.7.17 to V6.7.15 Now, ocpp work great again.

How do you downgrade firmware?

delainus, 2025-02-14:
> > Today downgrade firmware wallbox commander 2 V6.7.17 to V6.7.15 Now, ocpp work great again.
> 
> How do you downgrade firmware?

https://support.wallbox.com/na/knowledge-base/how-to-restore-the-firmware-of-your-charger/

Delorien84, 2025-02-14:
Ok, but then how do you upgrade to correct firmware? As I remember there is only option to upgrade to latest.

delainus, 2025-02-14:
> Ok, but then how do you upgrade to correct firmware? As I remember there is only option to upgrade to latest.

If factory firmware isn't the last, reset factory will be fine.

delainus, 2025-05-01:
wallbox commander 2

Firmware 6.7.25
Wall Box Chargers ⸱ CMX2-0-2-4
OCPP: 0.8.4

Work great